project-chip/connectedhomeip #25584: [BUG] [lighting-app] cannot build for ESP32-C3-devKitM-1

mmt-10, 2023-03-09:
### Reproduction steps

```bash
cd ${MATTER_SDK}
source scripts/activate.sh
cd examples/lighiting-app/esp32
idf.py set-target esp32c3
idf.py build
```

> 
> In file included from /home/mmt-10/matter/matter-master-20230309/examples/lighting-app/esp32/main/AppTask.cpp:23:
> /home/mmt-10/matter/matter-master-20230309/examples/lighting-app/esp32/main/DeviceWithDisplay.h:23:10: fatal error: Display.h: No such file or directory
>  #include "Display.h"
>           ^~~~~~~~~~~
> compilation terminated.

> /home/mmt-10/matter/matter-master-20230309/examples/lighting-app/esp32/main/Button.cpp:26:10: fatal error: ScreenManager.h: No such file or directory
>  #include "ScreenManager.h"
>           ^~~~~~~~~~~~~~~~~
> compilation terminated.

> /home/mmt-10/matter/matter-master-20230309/examples/lighting-app/esp32/main/include/WiFiWidget.h:21:10: fatal error: Display.h: No such file or directory
>  #include "Display.h"
>           ^~~~~~~~~~~
> compilation terminated.

> /home/mmt-10/matter/matter-master-20230309/examples/lighting-app/esp32/main/DeviceWithDisplay.h:23:10: fatal error: Display.h: No such file or directory
>  #include "Display.h"
>           ^~~~~~~~~~~
> compilation terminated.
> 

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit 1f2f8820073f96c842869f693248dc2acba17dd0 (HEAD -> master, origin/master, origin/HEAD)

### Platform

esp32

shubhamdp, 2023-03-09:
https://github.com/project-chip/connectedhomeip/pull/25550 had added display components, so may have broke the C3. Will fix this

mmt-10, 2023-03-09:
Thank you for your reply.

I revised temporary during fixing.
`Display.h`, `ScreenManager.h`, `ListScreen.h`, `StatusScreen.h` and any header file are under `#if CONFIG_HAVE_DISPLAY`.

closed 2023-03-09 by mmt-10

shubhamdp, 2023-03-09:
@mmt-10 If you have a patch you could put up a PR