project-chip/connectedhomeip #25602: [API-Epoch Key Generation]

sumaky, 2023-03-10:
### Feature description

...
Test cases as TC-G-2.2, 2.3, TC-SC-5.1,5.2 need to use the following  hard coded values while sending KeySetWrite command
EpochKey0: d0d1d2d3d4d5d6d7d8d9dadbdcdddedf
EpochStartTime0: 2220000
EpochKey0: d1d1d2d3d4d5d6d7d8d9dadbdcdddedf
EpochStartTime0: 2220001
EpochKey0: d2d1d2d3d4d5d6d7d8d9dadbdcdddedf
EpochStartTime0: 2220002
The specification mentions 
4.15.3. Epoch Keys
EpochKey = Crypto_DRBG(len = CRYPTO_SYMMETRIC_KEY_LENGTH_BITS)

Is there an API to generate the Epoch Keys or could we use any hard coded value to send the command?

### Platform

all

bzbarsky-apple, 2023-03-10:
@sumaky It can be any byte string of the right length. In an actual deployment, the keys would be randomly generated so that third parties do not know them, but in the tests it doesn't matter that much whether it's generated every time or hardcoded.

That said, we could make these variables that then allow the test runner to generate them and pass them into the test, if that's useful.

sumaky, 2023-03-13:
@bzbarsky-apple  thank you for the update. As the values can be hard coded we many not require any API.

closed 2023-03-13 by sumaky