project-chip/connectedhomeip #25603: Kindly address the error stating - CHIP Error 0x000000AC: Internal error[BUG]

Basavaraju4303, 2023-03-10:
### Reproduction steps

1. Install Ubuntu, open terminal.
2. Added dependencies 
 sudo apt-get update

sudo apt-get install git gcc g++ python pkg-config libssl-dev libdbus-1-dev libglib2.0-dev libavahi-client-dev ninja-build python3-venv python3-dev python3-pip unzip libgirepository1.0-dev libcairo2-dev libreadline-dev

3 clone repo and submodules

git clone https://github.com/project-chip/connectedhomeip.git --branch v1.0.0

cd connectedhomeip

./scripts/checkout_submodules.py --shallow --platform linux

4.Setup dev environment

./scripts/build/gn_bootstrap.sh

source scripts/activate.sh

5.Build new chip-tool
scripts/examples/ gn_build_example.sh 
examples/chip-tool out/chip-tool

6.Building all cluster-app

mkdir ${HOME}/tools
cd ${HOME}/tools
git clone https://github.com/espressif/esp-idf.git
cd esp-idf
git checkout v4.4
git submodule update --init
./install.sh
. ./export.sh

6.Now back to matter repository and activate the environment

cd /path/to/connectedhomeip
source scripts/activate.sh

cd examples/all-clusters-app/esp32
idf.py set-target esp32
idf.py build

7.after firmware is built now flash monitor
idf.py -p /dev/ttyUSB0 flash monitor
8.erase the flash
idf.py -p /dev/ttyUSB0 erase-flash
9. Commisioning the ESP32

./chip-tool pairing ble-wifi 12344321 "IoT Testing" NotTheRealPassword 20202021 3840.

10.After running it is stating an error as  CHIP Error 0x000000AC: Internal error
[esp log.txt](https://github.com/project-chip/connectedhomeip/files/10939542/esp.log.txt)
[log.txt](https://github.com/project-chip/connectedhomeip/files/10939543/log.txt)
[log.txt](https://github.com/project-chip/connectedhomeip/files/10939547/log.txt)

[esp log.txt](https://github.com/project-chip/connectedhomeip/files/10939558/esp.log.txt)
[log.txt](https://github.com/project-chip/connectedhomeip/files/10939561/log.txt)

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

project-chip/connectedhomeip

### Platform

esp32, python, raspi

bzbarsky-apple, 2023-03-10:
The log (which is clearly not the entire log, so it's hard to say that much useful about it) says:
```
[1678348935.453034][202210:202210] CHIP:TOO: Run command failure: ../../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:2063: CHIP Error 0x000000AC: Internal error
```
But if I actually do `git clone https://github.com/project-chip/connectedhomeip.git --branch v1.0.0` and look at line 2063 in that file, that is not a line that there is an error return on.    It's a blank line.

So @Basavaraju4303 are you using a different SHA than 561d23d0db215a99705ff0696e73853c8edf11b2 (which one?) or do you have local changes (which ones)?

closed 2023-03-29 by Basavaraju4303

chuckrui, 2023-04-03:
`chuckrui@ubuntu:~/matter$ ./chip-tool-debug pairing ble-thread 1234 hex:0e080000000000010000000300000f35060004001fffe0020811111111222222220708fd9d307e44b1e169051000112233445566778899aabbccddeeff030e4f70656e54687265616444656d6f010212340410445f2b5ca6f2a93a55ce570a70efeecb0c0402a0f7f8 20202021 3840
[1680495064.064719][8063:8063] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1680495064.065127][8063:8063] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1680495064.065262][8063:8063] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1680495064.065347][8063:8063] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1680495064.065631][8063:8063] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-eK1WHU)
[1680495064.066138][8063:8063] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1680495064.066185][8063:8063] CHIP:DL: NVS set: chip-counters/reboot-count = 7 (0x7)
[1680495064.066972][8063:8063] CHIP:DL: Failed to get Ethernet interface
[1680495064.067501][8063:8063] CHIP:DL: Failed to reset Ethernet statistic counts
[1680495064.067797][8063:8063] CHIP:DL: Got WiFi interface: wlan0
[1680495064.069511][8063:8063] CHIP:DL: Found the primary WiFi interface:wlan0
[1680495064.069576][8063:8063] CHIP:IN: UDP::Init bind&listen port=0
[1680495064.069660][8063:8063] CHIP:IN: UDP::Init bound to port=59560
[1680495064.069680][8063:8063] CHIP:IN: UDP::Init bind&listen port=0
[1680495064.069743][8063:8063] CHIP:IN: UDP::Init bound to port=52415
[1680495064.069761][8063:8063] CHIP:IN: BLEBase::Init - setting/overriding transport
[1680495064.069776][8063:8063] CHIP:IN: TransportMgr initialized
[1680495064.069805][8063:8063] CHIP:FP: Initializing FabricTable from persistent storage
[1680495064.069933][8063:8063] CHIP:TS: Last Known Good Time: 2022-10-21T05:54:44
[1680495064.070307][8063:8063] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0xFFB028E9FB4FFABA, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1680495064.072142][8063:8063] CHIP:ZCL: Using ZAP configuration...
[1680495064.076685][8063:8063] CHIP:DL: Avahi client registered
[1680495064.078000][8063:8063] CHIP:CTL: System State Initialized...
[1680495064.078104][8063:8063] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1680495064.078157][8063:8063] CHIP:CTL: Setting attestation nonce to random value
[1680495064.078207][8063:8063] CHIP:CTL: Setting CSR nonce to random value
[1680495064.078449][8063:8068] CHIP:DL: CHIP task running
[1680495064.078668][8063:8068] CHIP:DL: HandlePlatformSpecificBLEEvent 32784
[1680495064.078948][8063:8068] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1680495064.078986][8063:8068] CHIP:CTL: Setting attestation nonce to random value
[1680495064.079096][8063:8068] CHIP:CTL: Setting CSR nonce to random value
[1680495064.082040][8063:8068] CHIP:CTL: Generating NOC
[1680495064.083302][8063:8068] CHIP:FP: Validating NOC chain
[1680495064.085480][8063:8068] CHIP:FP: NOC chain validation successful
[1680495064.085679][8063:8068] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1680495064.085728][8063:8068] CHIP:TS: Last Known Good Time: 2022-10-21T05:54:44
[1680495064.085753][8063:8068] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1680495064.085776][8063:8068] CHIP:TS: Retaining current Last Known Good Time
[1680495064.089788][8063:8068] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1680495064.091489][8063:8068] CHIP:TS: Committing Last Known Good Time to storage: 2022-10-21T05:54:44
[1680495064.093055][8063:8068] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
[1680495064.102659][8063:8068] CHIP:IN: SecureSession[0xffff6c021720]: Allocated Type:1 LSID:60619
[1680495064.102723][8063:8068] CHIP:SC: Assigned local session key ID 60619
[1680495064.102830][8063:8068] CHIP:SC: Including MRP parameters in PBKDF param request
[1680495064.102934][8063:8068] CHIP:EM: <<< [E:52007i M:155915215] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
[1680495064.103016][8063:8068] CHIP:IN: (U) Sending msg 155915215 to IP address 'BLE'
[1680495064.103075][8063:8068] CHIP:IN: Message appended to BLE send queue
[1680495064.103107][8063:8068] CHIP:SC: Sent PBKDF param request
[1680495064.103160][8063:8068] CHIP:CTL: Setting thread operational dataset from parameters
[1680495064.103204][8063:8068] CHIP:CTL: Setting attempt thread scan from parameters
[1680495064.103233][8063:8068] CHIP:CTL: Setting attestation nonce to random value
[1680495064.103325][8063:8068] CHIP:CTL: Setting CSR nonce to random value
[1680495064.103405][8063:8068] CHIP:CTL: Commission called for node ID 0x00000000000004D2
[1680495064.103748][8063:8069] CHIP:DL: TRACE: Bluez mainloop starting Thread
[1680495064.103957][8063:8066] CHIP:DL: TRACE: Bus acquired for name C-1f7f
[1680495064.114194][8063:8068] CHIP:DL: PlatformBlueZInit init success
[1680495064.121273][8063:8066] CHIP:BLE: BLE removing known devices.
[1680495064.123217][8063:8066] CHIP:BLE: BLE initiating scan.
[1680495064.160915][8063:8066] CHIP:BLE: Device 5C:C3:36:47:39:E8 does not look like a CHIP device.
[1680495064.164717][8063:8066] CHIP:BLE: Device 51:81:BE:C6:50:1F does not look like a CHIP device.
[1680495064.167509][8063:8066] CHIP:BLE: Device 6D:3C:C0:3E:FC:8D does not look like a CHIP device.
[... cut ...]
[1680495068.978608][8063:8063] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1680495068.978629][8063:8063] CHIP:DL: Inet Layer shutdown
[1680495068.978645][8063:8063] CHIP:DL: BLE shutdown
[1680495068.979585][8063:8063] CHIP:DL: System Layer shutdown
[1680495068.979693][8063:8063] CHIP:TOO: Run command failure: ../../src/controller/CHIPDeviceController.cpp:1031: CHIP Error 0x000000AC: Internal error
`
Hi：
Such an error occurred when I used chip-tool to add devices to the matter network. Would you like to know where the problem occurred?

bzbarsky-apple, 2023-04-03:
@chuckrui That does not seem related to this issue.  Please file a separate issue and:

1. Say what SHA you are using.  This is very important for purposes of interpreting line numbers in error messages.
2. Attach, don't paste, your log.