project-chip/connectedhomeip #25607: [Documentation] Specs vs XML difference Flow Measurement Cluster

Apollon77, 2023-03-10:
### Documentation issues

The specs defined a default value of null fot the Measured­ Value, but the XML does not https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/flow-measurement-cluster.xml#L27

Which is right?

Formally there could also be defined a "min" value for MinMea­suredValue in the XML (=== 0) and a max value for MaxMeasuredValue (=== 65534) because as defined in the specs.

bzbarsky-apple, 2023-03-10:
The XML does not have a way to define a default value of null, to my knowledge.  @brdandu @tecimovic is there a way of doing it that I am missing?  See also https://github.com/project-chip/zap/issues/608

> Formally there could also be defined a "min" value for MinMea­suredValue in the XML (=== 0) and a max value for MaxMeasuredValue (=== 65534)

Those are already required by the type; arguably the spec should not list the redundant constraint....

andy31415, 2023-08-29:
Removed cert blocker during triage: we expect this needs fixing, however XML differences likely do not affect SDK ability to be certified (at least regarding these issues, given that certification  tests still pass).

Apollon77, 2026-10-22:
Addressed on master: MeasuredValue has no default, MinMeasuredValue max 65533, MaxMeasuredValue min 1 and max 65534, as in the spec. Closing.

closed 2026-10-22 by Apollon77