project-chip/connectedhomeip #25609: [Documentation] Specs Vs XML Difference IdentifyCluster - Query Feature missing in XML

Apollon77, 2023-03-10:
### Documentation issues

The Query Feature and also the IdentifyQuery Command which is described in the Application Specs 1.2 is missing in XML completely

bzbarsky-apple, 2023-03-10:
This feature and this command are not part of Matter.  They're defined in the cluster, because "shared data model" with Zigbee, but if you look at the device library they are required to not be supported in Matter devices.

closed 2023-03-10 by bzbarsky-apple

Vyrastas, 2023-07-28:
>  if you look at the device library they are required to not be supported in Matter devices.

@bzbarsky-apple  Sorry to bump an old issue, but I ran across the same thing as @Apollon77 and where exactly is the device library you mention?

I'm trying to understand how to tell that something isn't part of Matter (since for the IdentifyQuery command the spec doesn't make that distinction).  I see this https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/matter-devices.xml#L184 which says required for various device types so I'm a little confused.  Unless I'm looking at the wrong thing.

bzbarsky-apple, 2023-07-28:
> where exactly is the device library you mention?

If you go to https://csa-iot.org/developer-resource/specifications-download-request/  see "Matter 1.0 Device Library Specification" and "Matter 1.1 Device Library Specification", depending on which version you want.

You may also be able to find versions of this on the web if you just search for "matter device library specification" in your favority search engine.

> I'm trying to understand how to tell that something isn't part of Matter

The entrypoints to the spec in terms of device composition are _device types_.  Those will say what clusters are (mandatorily or optionally) included for the device type, as well as which features of those clusters are required/allowed/forbidden to be supported for that device type. 

In the specific case of "Matter 1.0 Device Library Specification" the relevant bit would be that for each device type that includes the Identify cluster (e.g. the "On/Off Light" device type, section 4.1), there is an Element Requirements bit (section 4.1.5 in this case) that says that for this device type for the Identify cluster the Query feature has conformance "!Matter".  Which means a Matter Identify cluster must not implement that feature.

In the 1.2 spec this is made a little saner and moved to the "Base Device Type Requirements for Matter" section so that it's not repeated in every place Identify appears.

The XML you point to is wrong; it was copy/pasted from somewhere, and needs to be fixed.  No one caught the issue, because absolutely nothing in Matter uses that XML file for anything in practice.  The ZAP UI does, but since the input cluster XML to ZAP does not have this command, in practice things "work".

The concrete thing you want in

Vyrastas, 2023-07-28:
Thanks @bzbarsky-apple I see it now.  And also in the spec repo... I had been looking at the master branch instead of the 1.0 tag.

bzbarsky-apple, 2023-07-28:
Master branch has a device library too.

Vyrastas, 2023-07-28:
Right, but the master branch doesn't have the Element Requirement you were referencing (like in the OnOffLight device type doc):

ID | Cluster | Element | Name | Constraint | Access | Conformance
-- | -- | -- | -- | -- | -- | --
0 | Identify | Feature | Query |   |   | !Matter

Appears to have been removed in master a month ago.

bzbarsky-apple, 2023-07-29:
I said above: 

> In the 1.2 spec this is made a little saner and moved to the "Base Device Type Requirements for Matter" section so that it's not repeated in every place Identify appears.

The Base Device Type lists requirements that apply for _all_ device types.