project-chip/connectedhomeip #25611: [Documentation] Specs vs XML differences and questions LevelControlCluster

Apollon77, 2023-03-10:
### Documentation issues

I discovered the following differences:
* Attribute CurrentLevel has a default value "null" according to specs, but "0" in XML. Which is right?
* Attribute minLevel Default value is formally dynamic based on feature support and might be not correct with being 0 in the XML
* Attribute onLevel has a default value of "Null" according to specs, XML do not specify any default value
* Attribute CurrentFrequency is in the end defined my min and max attributes. Why the default value is 0 then (in specs and XML)? (comparable for maxFrequency)

Apollon77, 2023-03-10:
> There's no way to specify the dynamic thing in the XML.

Yes thats right and also ok ... Interestingly some clusters then use the "maximum possible border min/max values" and some others simply do not specify min/max at all in this case ... ideally this shpuld also be consistent ... so using the "most outside border values" everywhere or not at all

bzbarsky-apple, 2023-03-10:
> ideally this shpuld also be consistent ... so using the "most outside border values" everywhere or not at all

In practice, min/max in the XML only matters for writable attributes, in terms of the observable behavior.  And even then more constraints can be done manually...  

> With the info I got from @turon that the XML is "the only truth"

That's the goal, yes.  We're pretty far from it, as you have noticed.....

Apollon77, 2023-03-10:
We planned to auto generate the Cluser code based on the XMLs for the TypeScript library, so we would also use it for "generating the types", shich means that also the developer that implements a device itself could make use of it because we make sure he uses correct values at all (not only when he sends a write request) :-)

But ok, knowing that we might need a more "use generated as basis and enhance" approach ...

> We're pretty far from it, as you have noticed.....

Glad I could help to come closer :-)

andy31415, 2023-08-29:
Removed cert blocker during triage: we expect this needs fixing, however XML differences likely do not affect SDK ability to be certified (at least regarding these issues, given that certification  tests still pass).

Apollon77, 2026-10-22:
Addressed on master: CurrentLevel and OnLevel are nullable without a default, MinLevel defaults to 1, and CurrentFrequency has no default. Closing.

closed 2026-10-22 by Apollon77