project-chip/connectedhomeip #25612: [Documentation] Specs vs XML Differences NetworkCommissioning Cluster

Apollon77, 2023-03-10:
### Documentation issues

I identified the following differences between Chip XML https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/network-commissioning-cluster.xml and the Specs 1.0
* Attribute maxNetworks has "min="1" according to specs, nothing set in XML
* Attribute networks defines a maximum length of 12 in XML which is nowhere mentioned in  the specs (there length is dynamic bound to attribute maxNetworks)
* Attribute Inter­faceEn­abled has a default value true in the specs, but not in XML

bzbarsky-apple, 2023-03-10:
> Attribute maxNetworks has "min="1" according to specs, nothing set in XML

There is no way to represent that in the XML that I know of.  @brdandu @tecimovic 

> Attribute Inter­faceEn­abled has a default value true in the specs, but not in XML

 "default" in the XML right now does not mean the same concept as "default" does in the spec. So there's no particular reason they should be aligned, and in some cases aligning them will be actively confusing.  This is one of those cases.  There is no actual "default" for InterfaceEnabled in implementation terms; it has to be dynamically computed.  We should be somehow flagging this as "don't allow specifying a default in the ZAP UI at all" or something.

andy31415, 2023-08-29:
Removed cert blocker during triage: we expect this needs fixing, however XML differences likely do not affect SDK ability to be certified (at least regarding these issues, given that certification  tests still pass).

Apollon77, 2026-10-22:
Addressed on master: MaxNetworks has min 1, InterfaceEnabled defaults to 1, and the Networks list length is not used. Closing.

closed 2026-10-22 by Apollon77