project-chip/connectedhomeip #25616: [Documentation] Specs vs XML differences NetworkCommissioning Cluster

Apollon77, 2023-03-10:
### Documentation issues

* Attribute supported Fabroce have a min/max as of 5/254 according to specs (Core specs 11.17.6), but this is not reflected in the XML
* Attribute curtrentFabricIntdex has a default=0 as per specs which is not reflected in the XML

bzbarsky-apple, 2023-03-10:
@Apollon77 Is this meant to be about the Operational Credentials cluster?

Apollon77, 2023-03-10:
Yes, sorry ... corrected

bzbarsky-apple, 2023-03-10:
> Attribute curtrentFabricIntdex has a default=0 as per specs which is not reflected in the XML

"default" in the XML right now does not mean the same concept as "default" does in the spec. So there's no particular reason they should be aligned, and in some cases aligning them will be actively confusing. This is one of those cases. There is no actual "default" for CurrentFabricIndex in implementation terms; it has to be dynamically computed. We should be somehow flagging this as "don't allow specifying a default in the ZAP UI at all" or something.

Apollon77, 2023-03-11:
Ok, but then also the default of 0 in the specs makes no sense ... or?!

bzbarsky-apple, 2023-03-13:
Well, what it means in the specs is that if for some reason you need a CurrentFabricIndex for some algorithm defined by the spec _and_ you are in a context where there is not one available (e.g. that algorithm is not associated with a message that delivered over an encrypted channel), _then_ you should just use 0 (i.e. "no fabric") for the value.

In practice, the preconditions for that "if" might never even be met, in that there might be no such algorithms.

closed 2023-03-14 by jmartinez-silabs