project-chip/connectedhomeip #25617: [Documentation] Specs vs XML difference PressureMeasurement Cluster

Apollon77, 2023-03-10:
### Documentation issues

I noticed the following differences between 1.0 Specs and XMLs
* measuredValue has a default=0 according to XML, but this is not reflected in the specs
* tolerance is flagged as reportable:true in XML which is not shown in Specs (and also makes no sense in my eyes)
* scaledValue is reportable as per XML but not in specs
* scale has min=-127 and max 127 in specs but not in XML

bzbarsky-apple, 2023-03-10:
> measuredValue has a default=0 according to XML, but this is not reflected in the specs

The spec does not list an explicit default.  Per "7.17.4. Default Column" that means the type-specific default is used, which is 0.

There's a good question as to what the XML should reflect here, and an even better question whether the XML should call its thing "default", since it's not quite what the spec means by "default"...  But yeah, for now we should probably just align this.

> scale has min=-127 and max 127 in specs but not in XML

That constraint is redundant, since the type is `int8`, no?

Apollon77, 2023-03-10:
> That constraint is redundant, since the type is int8, no?

Also true :-)

closed 2023-03-14 by jmartinez-silabs