project-chip/connectedhomeip #25637: need specify more integer type in android chip-tool UI for IM invoke and write

yunhanw-google, 2023-03-10:
https://github.com/joonhaengHeo/connectedhomeip/blob/499b480f1b7b6f7d12eff32d1b41cfaa23b0cb8e/examples/android/CHIPTool/app/src/main/java/com/google/chip/chiptool/clusterclient/WildcardFragment.kt#L402

We need specify unsigned short, unsignedInt, unsigned long in the below map since their encoding is different for different type.          
  "UnsignedInt" to object:TlvWriterInterface {
              override fun generate(writer : TlvWriter, value: String, tag: chip.tlv.Tag) {
                writer.put(tag, value.toUInt())
              }

joonhaengHeo, 2023-03-11:
@yunhanw-google 
In my opinion, Integer Type can be divided into two types.

Int : Int, Long, Short
UnsignedInt : UInt, ULong, UShort

Therefore, I think we only need two types.
All Int types are converted to Long types in TLVWriter.kt and are being processed.

joonhaengHeo, 2023-03-11:
https://github.com/project-chip/connectedhomeip/blob/e1381bf86142261601344adff5dc1c4304e9aaf2/examples/android/CHIPTool/app/src/main/java/com/google/chip/chiptool/clusterclient/WildcardFragment.kt#L405
`writer.put(tag, value.toULong())`

https://github.com/project-chip/connectedhomeip/blob/e1381bf86142261601344adff5dc1c4304e9aaf2/examples/android/CHIPTool/app/src/main/java/com/google/chip/chiptool/clusterclient/WildcardFragment.kt#L410
`writer.put(tag, value.toLong())`

PR #25646 (2023-03-20): [Android] Change TLV Parsing variable (Int -> Long, UInt -> ULong)
Fix #25637 
There are two Int variables in the TLV Format. 
However, in the case of Kotlin or Java, there are six types: Long, Int, Short, ULong, UInt, and UShort. 
To cover all of this, I changed the menu that supports Int and UInt to Long and ULong to support all Int Types.
closed 2023-03-20 by andy31415