project-chip/connectedhomeip #25642: [Documentation] XML vs Specs differences Group Key management cluster

Apollon77, 2023-03-11:
### Documentation issues

* The command "KeySetReadAllIndices" has no request parameters according to specs, but XML defines an ID as input. I would assume XML to be wrong when checking the meaning of the command
* The Feature defined in the specs is not contained in the XML
* The GroupKeySetStruct misses the GroupKey­Multicast­Policy enum as defined in the XML (ok this is a provisional feature, no ideal how to handle that?

bzbarsky-apple, 2023-03-13:
> I would assume XML to be wrong when checking the meaning of the command

The XML is wrong; the question is how we fix it without breaking compat at this point...  Need to think about this carefully.

> The Feature defined in the specs is not contained in the XML

Yep, not least because that feature is just not implemented in the SDK, so there's been no need for it....

> The GroupKeySetStruct misses the GroupKey­Multicast­Policy enum as defined in the XML (ok this is a provisional feature, no ideal how to handle that?

If it's provisional, it should generally not be included in the XML.

bzbarsky-apple, 2023-03-13:
I filed https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/6344 to track reserving this field id on the spec side.

Apollon77, 2023-03-14:
So we will add this field to the specs even that it makes no real sense?

bzbarsky-apple, 2023-03-14:
> Ahh you just make sure that if ever fields get added they do not use this field ID?

Yes, exactly.

ReneJosefsen, 2023-06-02:
> The GroupKeySetStruct misses the GroupKey­Multicast­Policy enum as defined in the XML (ok this is a provisional feature, no ideal how to handle that?

Can you elaborate on this, looking at the spec, the defined GroupKeySetStruct type does include GroupKeyMulticastPolicy (field 8) and the type of that field is GroupKeyMulticastPolicyEnum?

bzbarsky-apple, 2023-06-02:
@ReneJosefsen The _spec_ has that, but the XML in the SDK does not.

PR #28413 (2023-07-31): Add missing enums in Group Key Management cluster
- Missing 2 enums in Group Key Management cluster XML (including feature bit)
- Not adding the GroupKeyMulticastPolicy as there is no support at all in SDK and adding it as mandatory would break 1.0 clients. This needs to be fixed in spec to be O, P before it's touched.

Fixes #25642
closed 2023-07-31 by mergify[bot]