project-chip/connectedhomeip #25659: [BUG] Android: Failed to resolve service 'xxxxx-yyyyy' : 101

caipiblack, 2023-03-13:
### Reproduction steps

1. Commission a device on android
2. Checks the logs to see if find operational step fails

If find operational step fails, there are many reasons, the reason in this issue is "Failed to resolve service 'xxxxx-yyyyy' : 101"

The "101" is maybe important here.

### Bug prevalence

1/20

### GitHub hash of the SDK that was being used

v1.0.0

### Platform

android

### Platform Version(s)

Android 10, MIUI global 12.5.3 Stable

### Anything else?

otbr-agent is integrated into a OpenWRT gateway.

The Android phone and a computer are connected to the LAN of the OpenWRT gateway

The computer is able to discover the matter device using avahi browse command: 

```
caipiblack@caipiblack-PC-Portable:/$ avahi-browse _matter._tcp -r | grep 5BB2BF7451309CA6-0000000000000009
+ wlp0s20f3 IPv6 5BB2BF7451309CA6-0000000000000009             _matter._tcp         local
+ wlp0s20f3 IPv4 5BB2BF7451309CA6-0000000000000009             _matter._tcp         local
+ enp1s0 IPv6 5BB2BF7451309CA6-0000000000000009             _matter._tcp         local
+ enp1s0 IPv4 5BB2BF7451309CA6-0000000000000009             _matter._tcp         local
= enp1s0 IPv6 5BB2BF7451309CA6-0000000000000009             _matter._tcp         local
= enp1s0 IPv4 5BB2BF7451309CA6-0000000000000009             _matter._tcp         local
= wlp0s20f3 IPv6 5BB2BF7451309CA6-0000000000000009             _matter._tcp         local
= wlp0s20f3 IPv4 5BB2BF7451309CA6-0000000000000009             _matter._tcp         local
```

But from the android phone, I get these error:

```
D CTL     : Commissioning stage next step: 'ThreadNetworkEnable' -> 'FindOperational'
D CTL     : Performing next commissioning step 'FindOperational'
D CSM     : FindOrEstablishSession: PeerId = [1:0000000000000009]
D CSM     : FindOrEstablishSession: No existing OperationalSessionSetup instance found
I ReactNativeJS: 'credentialInitialized ', true, 'firstCommissioningEnded', true, ' circularProgressValueText ', undefined, ' haveError ', false, ' from ', 'oncommissioningstatusupdate', ' code ', 0, ' codeDescription ', 'SUCCESS', ' data ', '{"nodeId":"9","stage":"ThreadNetworkEnable"}'
D CTL     : OperationalSessionSetup[1:0000000000000009]: State change 1 --> 2
D DIS     : Resolving 5BB2BF7451309CA6:0000000000000009 ...
D NsdManagerServiceResolver: resolve: Starting service resolution for '5BB2BF7451309CA6-0000000000000009' type '_matter._tcp'
D DMG     : ICR moving to [AwaitingDe]
D DIS     : Checking node lookup status after 202 ms
D DL      : Received SendWriteRequest
D chip.BluetoothManager: EveEnergy04B16.onCharacteristicWrite: 18ee2ef5-263d-4559-959f-4f9c429f9d11 -> 0
W NsdManagerServiceResolver: Failed to resolve service '5BB2BF7451309CA6-0000000000000009': 101
D chip.BluetoothManager: EveEnergy04B16.onCharacteristicChanged: 18ee2ef5-263d-4559-959f-4f9c429f9d12
D chip.BluetoothManager: EveEnergy04B16.onCharacteristicChanged permissions: 0
D DL      : Received SendWriteRequest
D chip.BluetoothManager: EveEnergy04B16.onCharacteristicWrite: 18ee2ef5-263d-4559-959f-4f9c429f9d11 -> 0
D chip.BluetoothManager: EveEnergy04B16.onCharacteristicChanged: 18ee2ef5-263d-4559-959f-4f9c429f9d12
D chip.BluetoothManager: EveEnergy04B16.onCharacteristicChanged permissions: 0
D DL      : Received SendWriteRequest
D chip.BluetoothManager: EveEnergy04B16.onCharacteristicWrite: 18ee2ef5-263d-4559-959f-4f9c429f9d11 -> 0
D chip.BluetoothManager: EveEnergy04B16.onCharacteristicChanged: 18ee2ef5-263d-4559-959f-4f9c429f9d12
D chip.BluetoothManager: EveEnergy04B16.onCharacteristicChanged permissions: 0
D DL      : Received SendWriteRequest
D chip.BluetoothManager: EveEnergy04B16.onCharacteristicWrite: 18ee2ef5-263d-4559-959f-4f9c429f9d11 -> 0
D chip.BluetoothManager: EveEnergy04B16.onCharacteristicChanged: 18ee2ef5-263d-4559-959f-4f9c429f9d12
D chip.BluetoothManager: EveEnergy04B16.onCharacteristicChanged permissions: 0
D DL      : Received SendWriteRequest
D chip.BluetoothManager: EveEnergy04B16.onCharacteristicWrite: 18ee2ef5-263d-4559-959f-4f9c429f9d11 -> 0
D BluetoothGatt: onClientConnectionState() - status=19 clientIf=9 device=DE:95:C8:56:A8:6C
I chip.BluetoothManager: EveEnergy04B16.onConnectionStateChange status = 19, newState=0
D AndroidBleManager: onConnectionStateChange Disconnected
D IN      : Clearing BLE pending packets.
I chip.BluetoothManager|onConnectionStateChange: STATE_DISCONNECTED
D DIS     : Checking node lookup status after 45014 ms
E DIS     : OperationalSessionSetup[1:0000000000000009]: operational discovery failed: ../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:96: CHIP Error 0x00000032: Timeout
D CTL     : Device connection failed. Error ../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:96: CHIP Error 0x00000032: Timeout
```

The important things are in this line:

```
NsdManagerServiceResolver: Failed to resolve service '5BB2BF7451309CA6-0000000000000009': 101
```

The code that generate this error is in the Matter stack here:

[image]

What is very strange is that "101" is the value of `errorCode` but when I read the doc of NsdManager, the errorCode values are < 10.

https://developer.android.com/reference/android/net/nsd/NsdManager

Did someone know what is "101" ?

zhenqinrui, 2023-03-22:
please use RxDnssd to replease android framework nsdManager

nicelyjust, 2023-05-19:
I am facing the same issue, Do you have a reliable solution?

nicelyjust, 2023-05-19:
> RxDnssd

Have you replaced the library now? How is the stability?Have you replaced the library now? How is the stability? I'm hesitant to take action right now?

caipiblack, 2024-06-13:
This issue was mostly to understand where the "101" error code comes from,

Maybe nobody has the response, so I'm going to close.

closed 2024-06-13 by caipiblack