project-chip/connectedhomeip #25678: [Documentation] XML vs Specs differences General Diagnostic Cluster

Apollon77, 2023-03-14:
### Documentation issues

* Attribute Name "BOOT_REASONS" in XML do not match to specs and also just contain one entry, so name is confusing ...  :-)
* Attribute Network­Interfaces has a maxLength of 254 in XML, but 8 in the specs, which is right?
* The three faults arrays in attributes have maxlength in specs but not in XML, is spec valid?

bzbarsky-apple, 2023-03-14:
> Attribute Name "BOOT_REASONS" in XML do not match to specs and also just contain one entry, so name is confusing ... :-)

The attribute name in the XML is `BootReason`.  `BOOT_REASONS` is the "define", which does not reflect anything from the spec, and is a holdover that is not used for anything in practice.

> Attribute Network­Interfaces has a maxLength of 254 in XML, but 8 in the specs, which is right?

8 is right, but either way the "length" attribute in the XML is not used for anything for lists.  Note also that the XML has no way to express a min length for a list, and so on.  It's not really set up to be a 1-1 reflection of the spec.

> The three faults arrays in attributes have maxlength in specs but not in XML, is spec valid?

Spec is correct.  XML is "correct" too, in the sense that it's semantically the same as if it specified a "length", per above.

Apollon77, 2023-03-14:
Ok, then we should remove the 254 length as we did on otzers because it is confusing ... also syncing the default (if not used) to the name (like on the others) would make sese, right?

PS: Does it make sense what I do here with showing up the differences - also if "known" that length is unused ad such or should I just report the "real" issues? Please advice

bzbarsky-apple, 2023-03-14:
>  then we should remove the 254 length as we did on otzers because it is confusing 

Yes, just haven't gotten to it yet.

> also syncing the default (if not used) to the name (like on the others) would make sese, right?

s/default/define/ and yes.

> Does it make sense what I do here with showing up the differences

Yes, and it's very much appreciated!

Apollon77, 2026-10-22:
Addressed on master: NetworkInterfaces and the three fault lists now have the spec lengths. Only the unused BOOT_REASONS define name remains. Closing.

closed 2026-10-22 by Apollon77