project-chip/connectedhomeip #25680: How can my Matter-over-Thread device(nrf52840dongle) connect to Apple TV?

392625227, 2023-03-14:
In this question, I used the Apple TV 7 128G (with thread support) as the border router, not the Raspberry Pi . Apple TV is connected to the iPhone and working. Without the Apple TV, when I tried to add my NRF52840dongle to my 'apple home' network with the iPhone, the iPhone said "border routing required". When Apple TV working, it becomes a prompt "Unable to pair". 

How can my Matter-over-Thread device connect to Apple TV? My ultimate goal, of course, is that Matter-over-Thread devices end up like Apple home-kits.

I burned examples/lighting-app on nrf52840dongle, tried to test this lighting example with the home-kit app connection on the iphone, but the iPhone shows "Unable to add accessories, pairing failed", nrf52840dongle's uart log file is as follows:

renxn@Jason_Ren:~$ sudo minicom -D /dev/ttyACM0 -C log -c on

Welcome to minicom 2.7.1

OPTIONS: I18n
Compiled on Dec 23 2019, 02:06:26.
Port /dev/ttyACM0, 12:31:07

Press CTRL-A Z for help on special keys

I: Init CHIP stack
I: 279184 [DL]BLE address: D0:E1:82:09:73:9B
I: SoftDevice Controller build revision:
I: 6d 90 41 2a 38 e8 ad 17 |m.A*8...
I: 29 a5 03 38 39 27 d7 85 |)..89'..
I: 1f 85 d8 e1 |....
I: 279196 [DL]OpenThread started: OK
I: 279197 [DL]Setting OpenThread device type to MINIMAL END DEVICE
I: 279198 [SVR]Server initializing...
I: 279199 [TS]Last Known Good Time: 2023-03-09T08:43:41
I: 279202 [DMG]AccessControl: initializing
I: 279202 [DMG]Examples::AccessControlDelegate::Init
I: 279203 [DMG]AccessControl: setting
I: 279204 [DMG]DefaultAclStorage: initializing
I: 279204 [DMG]DefaultAclStorage: 0 entries loaded
E: 279205 [IN]SO_REUSEPORT failed: 109
I: 279213 [ZCL]Using ZAP configuration...
I: 279231 [DMG]AccessControlCluster: initializing
I: 279232 [ZCL]Initiating Admin Commissioning cluster.
I: 279232 [ZCL]Door Lock server initialized
I: 279233 [ZCL]Initialize PCC Plugin Server Cluster.
E: 279234 [ZCL]DeviceInfoProvider is not registered
E: 279235 [ZCL]Failed to read calendar type with error: 0x86
I: 279236 [ZCL]Initialize PCC Server Cluster [EP:1]
I: 279237 [ZCL]Constant Pressure Not Supported
I: 279238 [ZCL]PCC Server: Constant Proportional Pressure Not Supported
I: 279239 [ZCL]PCC Server: Constant Flow Not Supported
I: 279239 [ZCL]PCC Server: Constant Temperature Not Supported
I: 279240 [ZCL]PCC Server: Constant Speed Not Supported
I: 279241 [ZCL]PCC Server: ControlMode attribute not available
I: 279242 [ZCL]PCC Server: PumpStatus attribute not available
I: 279246 [DIS]Updating services using commissioning mode 1
I: 279248 [DIS]Advertise commission parameter vendorID=65521 productID=32769 discriminator=3840/15 cm=1
E: 279250 [DIS]Failed to advertise commissionable node: 3
E: 279250 [DIS]Failed to finalize service update: 1c
I: 279251 [DIS]Updating services using commissioning mode 1
I: 279253 [DIS]Advertise commission parameter vendorID=65521 productID=32769 discriminator=3840/15 cm=1
E: 279254 [DIS]Failed to advertise commissionable node: 3
E: 279255 [DIS]Failed to finalize service update: 1c
I: 279256 [DIS]Delaying proxy of operational discovery: missing delegate
I: 279257 [IN]CASE Server enabling CASE session setups
I: 279258 [SVR]Joining Multicast groups
I: 279259 [SVR]Server Listening...
I: 279259 [DL]Device Configuration:
I: 279260 [DL] Serial Number: 11223344556677889900
I: 279261 [DL] Vendor Id: 65521 (0xFFF1)
I: 279262 [DL] Product Id: 32769 (0x8001)
I: 279263 [DL] Hardware Version: 0
I: 279264 [DL] Setup Pin Code (0 for UNKNOWN/ERROR): 20202021
I: 279266 [DL] Setup Discriminator (0xFFFF for UNKNOWN/ERROR): 3840 (0xF00)
I: 279268 [DL] Manufacturing Date: (not set)
I: 279269 [DL] Device Type: 65535 (0xFFFF)
I: 279271 [SVR]SetupQRCode: [MT:-24J042C00KA0648G00]
I: 279272 [SVR]Copy/paste the below URL in a browser to see the QR Code:
[... 157 lines cut ...]
I: 363931 [DIS]Advertise commission parameter vendorID=65521 productID=32769 discriminator=3840/15 cm=1
I: 363933 [DL]advertising srp service: 022D169D4ED61DD1._matterc._udp
E: 363934 [ZCL]OpCreds: Got FailSafeTimerExpired
E: 363934 [ZCL]OpCreds: Proceeding to FailSafeCleanup on fail-safe expiry!
E: 363935 [FP]Reverting pending fabric data for fabric 0x1
E: 363937 [FP]Warning: metadata not found during delete of fabric 0x1
I: 363967 [FP]Fabric (0x1) deleted.
I: 363968 [ZCL][OnFabricRemoved] Handling a fabric removal from the door lock server [endpointId=1,fabricIndex=1]
E: 363969 [ZCL]Failed to read DoorLock attribute: attribute=0x11, status=0x86
E: 363970 [ZCL][OnFabricRemoved] Unable to cleanup fabric from users - internal error [endpointId=1,fabricIndex=1]
E: 363971 [ZCL]Unable to handle fabric removal from the Door Lock Server instance [endpointId=1,fabricIndex=1]
I: 364010 [ZCL]UserLabel: Last Fabric index 0x1 was removed
I: 364011 [ZCL]OpCreds: Fabric index 0x1 was removed
I: 364142 [TS]Pending Last Known Good Time: 2023-03-14T04:35:57
I: 364144 [TS]Previous Last Known Good Time: 2023-03-09T08:43:41
I: 364145 [TS]Reverted Last Known Good Time to previous value
E: 364147 [FP]Warning: metadata not found during delete of fabric 0x1
E: 364151 [ZCL]OpCreds: failed to delete fabric at index 1: d8
I: 364153 [NP]Reverting Thread operational dataset
E: 364163 [DL]Long dispatch time: 243 ms, for event type 32782
I: 364164 [FS]Fail-safe cleanly disarmed
I: 364167 [DL]BLE GAP connection terminated (reason 0x16)
I: 364168 [DL]Current number of connections: 0/2
I: 364171 [DL]CHIPoBLE advertising started
I: 394171 [DL]CHIPoBLE advertising mode changed to slow

bzbarsky-apple, 2023-03-14:
What seems to be happening is the CASE handshake fails after the device is on the Thread network.  This part from the log:
```
I: 312528 [SC]Sent Sigma2 msg
I: 312753 [EM]>>> [E:31116r M:81400263 (Ack:2660917)] (U) Msg RX from 0:461BEBD62DE29576 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
E: 312755 [SC]Received error (protocol code 2) during pairing process: 54
```
where 0x54 is CHIP_ERROR_INVALID_CASE_PARAMETER.  

@392625227 What SHA is the Thread device version of Matter built from?  Can you commission that device using command-line chip-tool (if you pass it the right Thread credentials)?

392625227, 2023-03-31:
After a few weeks of studying, I finally figured out some problems.

The first question, webUI is a web tool used to manage thread devices on otbr (open thread board router), through the join button can see the thread PAN ID of the matter device, such as Apple TV box has established a thread network, assuming you know PSK, you can join this network. Unfortunately, I don't know how to get Apple TV Box's thread network PSK.
The second problem, compiled with matter/example/lighting-app/nrfconnect, burned on nrf52840dongles, can already be well controlled by Apple HomeKit. The original problem was that the connection process only needed to ensure that the matter device, Apple TV box, iPhone, all three must be in close range, keep the wireless communication signal good, and connect well to the Internet. When adding a new matter device to your iPhone, enter the QR code, 20202021, 34970112332, all three ways ok.

closed 2023-03-31 by 392625227