project-chip/connectedhomeip #25681: [Feature] Provide Darwin SDK API to retrieve and clean storage fabric data

kpark-apple, 2023-03-14:
### Feature description

Application might want to retrieve storage data to decide which fabric to use and also to send NOC to add fabric through another fabric.
Also, application may want to clean up obsolete fabrics in the storage.

### Platform

darwin

bzbarsky-apple, 2023-03-14:
@kpark-apple This is going to need a somewhat clearer description of exactly what APIs you are looking for.  Specifically:

1. What "storage data" do you want to retrieve and in what form?  How should the data to be retrieved be identified?
2. What does "obsolete fabrics" mean?  What are the criteria there?  Where did they come from?

kpark-apple, 2023-03-14:
1. What "storage data"
  * Fabric root CA cert in DER format per either fabric ID or fabric index (for the latter, we'll also need an API to retrieve list of available fabric indices as well).
  * Node ID of the NOC for a fabric per either fabric ID or fabric index.
2. Obsolete fabrics:
  * I saw fabric records for a fabric index which isn't listed in g/fidx. As they consume storage, it would be nice to have an API to remove all such obsolete records.

bzbarsky-apple, 2023-03-14:
> Fabric root CA cert in DER format per either fabric ID

There can be multiple fabrics with the same fabric id but different root certificates, in general, no?  How do you expect this API to work?

> (for the latter, we'll also need an API to retrieve list of available fabric indices as well).

Having an API for "here are the configured fabrics" would be pretty reasonable, and seems like it would address the "storage data" requirement, right?

> Node ID of the NOC for a fabric per either fabric ID or fabric index.

Again, "fabric id" does not identify a fabric uniquely.  

> I saw fabric records for a fabric index which isn't listed in g/fidx. As they consume storage, it would be nice to have an API to remove all such obsolete records.

This indicates data corruption on the part of the storage.  Why would the API for dealing with this live in Matter.framework (and how would you expect it to be implemented?) as opposed to being the responsibility of the (broken) storage implementation?

bzbarsky-apple, 2023-03-14:
> but instead at least provide an API to read the content of g/fidx

An API that provides a list of configured fabrics would give you that, right?

kpark-apple, 2023-03-14:
> An API that provides a list of configured fabrics would give you that, right? That is, if for each fabric it includes the fabric index in the information it returns.

That sounds good.

closed 2023-03-17 by bzbarsky-apple