project-chip/connectedhomeip #25684: [BUG] [controller] Support for TemperatureSetpointHold is missing on client side.

turon, 2023-03-14:
### Reproduction steps

See appclusters spec 4.3.7. Attributes (HVAC.Thermostat.TemperatureSetpointHold = 0x24).
Notice [device-side implementation](https://github.com/project-chip/connectedhomeip/blob/8a35b081840079da64eef4d2cb09c68d416758d2/src/app/zap-templates/zcl/data-model/chip/thermostat-cluster.xml#L88).
Notice missing [client-side implementation](https://github.com/project-chip/connectedhomeip/blob/8a35b081840079da64eef4d2cb09c68d416758d2/src/controller/data_model/controller-clusters.matter#L3185) (no support for `"code": 36`).

### Bug prevalence

always

### GitHub hash of the SDK that was being used

master

### Platform

core

bzbarsky-apple, 2023-03-14:
@turon This is supported fine in chip-tool, Darwin framework, darwin-framework-tool.

It's possible that the java codegen and maybe the python codegen actually pay attention to the enabled state of things in controller-clusters.zap.  We should fix them so they stop paying attention, like all the other clients.

turon, 2023-03-14:
@bzbarsky-apple Wouldn't the proper fix be to set the enable correctly in the XML?  Or from your comment, should I take it that there are so many fields with `enabled` set incorrectly that it is effectively a dead field?

bzbarsky-apple, 2023-03-15:
> Wouldn't the proper fix be to set the enable correctly in the XML?

"enable" is not set in XML.  It's set in the .zap file.

Past that, define "correctly"?  If the intent for the client is "expose everything", examining "enabled" just adds a point of failure, and we had repeated bugs with people forgetting to enable things in controller-clusters.zap when adding something new to the XML.

So yes, it's basically a dead field in practice, as far as chip-tool and the Darwin framework and darwin-framework-tool are concerned.

If Java and Python want to "expose everything" they should stop examining it too.   If they want to have an allow-list of things they expose, then their maintainers need to watch for all changes to XML and update the .zap file as needed.

> I don't see TemperatureSetpointHold in the client [XML](https://github.com/project-chip/connectedhomeip/blob/8a35b081840079da64eef4d2cb09c68d416758d2/src/controller/data_model/controller-clusters.zap#L12639) either.

That's not XML.  That's a .zap file.  And yes, it's not listed there, probably because no one touched its state on that cluster in that .zap file after the attribute was added to the XML.

andy31415, 2023-04-27:
I believe this is fixed now.

closed 2023-04-27 by andy31415