project-chip/connectedhomeip #25688: [Feature] Update build scripts to OpenSSL 3

mweinelt, 2023-03-14:
### Feature description

The build scripts should be updated to openssl 3, because support for 1.1 is going to end in 2023/09.

### Platform

all

### Anything else?

> Version 1.1.1 will be supported until 2023-09-11 (LTS). Extended support for 1.1.1 to gain access to security fixes beyond that date for that version is available.

https://www.openssl.org/policies/releasestrat.html

mweinelt, 2023-09-11:
Public support for OpenSSL 1.1 has now ended. This issue affects distros with a reasonable security posture.

Please consider getting on top of this problem.

felschr, 2023-10-24:
Pinging @robszewczyk @tcarmelveilleux

This should probably have the `security` label.

agners, 2023-12-21:
From what I can tell this is mostly a question of bumping the chip-tools container from Ubuntu 20.04 to 22.04?

@woody-apple since you mentioned you intended to update the chip-build container in https://github.com/project-chip/connectedhomeip/pull/23444/files#r1012308811, do you have any insights? Did you hit a roadblock there?