project-chip/connectedhomeip #25696: [BUG] ESP32 Unable to connect to ASUS ZenWiFi 8 Router

s-mcclain, 2023-03-15:
### Reproduction steps

Using chip-tool as Admin on macOS. Building ESP32C3 using lighting-app, I am unable to commission the device into the WiFi network. It constantly loops between Connecting -> Connecting_Failed -> NotConnected -> Connecting.

I have supplied the device with the correct credentials.

`I (71616) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 0
E (71626) chip[DL]: Long dispatch time: 417 ms, for event type 7
I (71696) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14
I (71696) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (71796) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (71886) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (71896) chip[EM]: Received message of type 0x8 with protocolId (0, 1) and MessageCounter:167262557 on exchange 14136r
I (71906) chip[ZCL]: OpCreds: Received an AddNOC command
I (71906) chip[FP]: Validating NOC chain
I (72606) chip[FP]: NOC chain validation successful
I (72606) chip[FP]: Added new fabric at index: 0x1
I (72606) chip[FP]: Assigned compressed fabric ID: 0xFA620EFF58F3EC13, node ID: 0x00FC26B56E2979B0
I (72616) chip[TS]: Last Known Good Time: 2022-08-16T09:46:40
I (72626) chip[TS]: New proposed Last Known Good Time: 2023-03-15T19:26:42
I (72636) chip[TS]: Updating pending Last Known Good Time to 2023-03-15T19:26:42
I (72646) chip[ZCL]: OpCreds: ACL entry created for Fabric index 0x1 CASE Admin Subject 0xFFFFFFFD0011002A
I (72656) chip[DIS]: Advertise operational node FA620EFF58F3EC13-00FC26B56E2979B0
I (72666) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'.
I (72676) chip[DIS]: mDNS service published: _matter._tcp
I (72676) chip[ZCL]: OpCreds: successfully created fabric index 0x1 via AddNOC
I (72686) chip[IN]: Prepared secure message 0x3fcb83d8 to 0xFFFFFFFB00000000 (1)  of type 0x9 and protocolId (0, 1) on exchange 14136r with MessageCounter:69429872.
I (72706) chip[IN]: Sending encrypted msg 0x3fcb83d8 with MessageCounter:69429872 to 0xFFFFFFFB00000000 (1) at monotonic time: 0000000000011BFB msec
I (72716) NimBLE: GATT procedure initiated: indicate;
I (72716) NimBLE: att_handle=14

I (72726) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 0
E (72736) chip[DL]: Long dispatch time: 849 ms, for event type 7
I (72816) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14
I (72816) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (72826) chip[EM]: Received message of type 0x8 with protocolId (0, 1) and MessageCounter:167262558 on exchange 14137r
I (72846) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0030', EndPoint ID: '0x00', Attribute ID: '0x0000'
I (72846) app-devicecallbacks: Unhandled cluster ID: 48
I (72856) app-devicecallbacks: Current free heap: 103052

I (72856) chip[IN]: Prepared secure message 0x3fcb83d8 to 0xFFFFFFFB00000000 (1)  of type 0x9 and protocolId (0, 1) on exchange 14137r with MessageCounter:69429873.
I (72886) chip[IN]: Sending encrypted msg 0x3fcb83d8 with MessageCounter:69429873 to 0xFFFFFFFB00000000 (1) at monotonic time: 0000000000011CAE msec
I (72896) NimBLE: GATT procedure initiated: indicate;
I (72896) NimBLE: att_handle=14

I (72906) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 0
I (72956) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14
I (72956) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (72976) chip[EM]: Received message of type 0x8 with protocolId (0, 1) and MessageCounter:167262559 on exchange 14138r
I (72986) chip[NP]: ESP NetworkCommissioningDelegate: SSID: Pretty Fly for a WiFi
I (72986) chip[DL]: WiFi station mode change: Enabled -> Disabled
I (73016) chip[DL]: Changing ESP WiFi mode: STA -> STA+AP
I (73016) wifi:mode : sta (84:f7:03:61:10:c0) + softAP (3a:53:5a:98:b4:a5)
I (73016) wifi:Total power save buffer number: 16
I (73036) wifi:Init max length of beacon: 752/752
I (73036) phy: chan:1,max_power:80
I (73036) phy: chan:2,max_power:80
I (73046) phy: chan:3,max_power:80
I (73046) phy: chan:4,max_power:80
I (73056) phy: chan:5,max_power:80
I (73056) phy: chan:6,max_power:80
I (73056) phy: chan:7,max_power:80
I (73066) phy: chan:8,max_power:80
I (73066) phy: chan:9,max_power:80
I (73066) phy: chan:10,max_power:80
I (73076) phy: chan:11,max_power:80
I (73076) phy: chan:12,max_power:80
I (73076) phy: chan:13,max_power:80
I (73086) phy: chan:14,max_power:80
I (73096) wifi:Init max length of beacon: 752/752
[... 335 lines cut ...]
I (106836) phy: chan:8,max_power:80
I (106836) phy: chan:9,max_power:80
I (106846) phy: chan:10,max_power:80
I (106846) phy: chan:11,max_power:80
I (106846) phy: chan:12,max_power:80
I (106856) phy: chan:13,max_power:80
I (106866) phy: chan:14,max_power:80
W (106866) wifi:Haven't to connect to a suitable AP now!
I (106876) chip[DL]: Done driving station state, nothing else to do...`

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

561d23d

### Platform

esp32

### Platform Version(s)

1.0.0

bzbarsky-apple, 2023-03-16:
@s-mcclain Just to make sure: this is a 2.4GHz WiFi network, right?

@shubhamdp @dhrishi

dhrishi, 2023-03-16:
@s-mcclain ESP32 support 2.4GHz. Please confirm that the Wi-Fi network is not 5GHz. Else the Wi-Fi connection will fail.

s-mcclain, 2023-03-16:
@bzbarsky-apple / @dhrishi yes that's correct. The network which I've provided is 2.4 GHz.

The WiFi network SSIDs are also separated based on the frequency (i.e. There exists `WiFi A` for 2.4G, `WiFi B` for 5G, and `WiFi C` for 6G.

s-mcclain, 2023-03-16:
Another note... Seems like if I use a different ESP32C3 then it appears to be connecting (for now at least). Same commands and everything.

From glancing through the WiFi Router logs, there is nothing which indicates it has blocked the device from associating into the network. But even after waiting several days, it still has not been able to join into the network successfully.

dhrishi, 2024-10-25:
As it worked with a different ESP32C3, it should not be related to the software. May be it is some issue with the devkit.

closed 2024-10-25 by dhrishi