project-chip/connectedhomeip #25707: Getting stuck on ble comissioning

ivishaltejwani, 2023-03-16:
### Reproduction steps

I (32018) NimBLE: GAP procedure initiated: advertise; 
I (32028) NimBLE: disc_mode=2
I (32038) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (32048) NimBLE: 

this is where it gets stuck , I can see in apple home as matter accessory but cannot add it
same with google home it sometimes appear as matter device found but never gets added to the app

no changes in the example code

I have not used homepod or google nest mini to add , are they necessary to add in apple home?

I have tried idf 5.0.1 and idf 4.4.4 same issue with both , esp matter example as well as connectedhomeip example
ESP32-D0WD (revision 1)
ESP32-D0WD (revision 3)

both of the chips are tested

os- macos

### Bug prevalence

always

### GitHub hash of the SDK that was being used

latest checkout

### Platform

esp32

### Platform Version(s)

ESP32-D0WD (revision 1)

bzbarsky-apple, 2023-03-16:
@ivishaltejwani Can you commission using command-line chip-tool?  If not, can you please attach, not paste, the logs from both that and the esp32 to this issue?  If you _can_ commission using chip-tool, then this sounds like an issue with Apple Home and Google Home that should be reported to them directly.

ivishaltejwani, 2023-03-16:
> @ivishaltejwani Can you commission using command-line chip-tool? If not, can you please attach, not paste, the logs from both that and the esp32 to this issue? If you _can_ commission using [...]

not familiar with chip tool

ivishaltejwani, 2023-03-17:
> @ivishaltejwani Can you commission using command-line chip-tool? If not, can you please attach, not paste, the logs from both that and the esp32 to this issue? If you _can_ commission using [...]
it did comission once but unable to after that have tried alot 
idf 4.4.4 , esp matter example - light -switch

[image]

ivishaltejwani, 2023-03-18:
Tested on 

ESP32-D0WD (revision 3)

working properly - correcting as previously mentioned, no issues