project-chip/connectedhomeip #25711: [BUG] Ota-provider implementation doesn't validate VID, PID and ProtocolSupported fields before issuing a successful QueryImageResponse

fessehaeve, 2023-03-16:
### Reproduction steps

Right now chip-ota-provider-app doesn't validate VID, PID and ProtocolSupported values in the QueryImage command. Therefore, the device proceeds with receiving the image just because the software version has a higher number.

This is bad since Certification labs might use the wrong file for OTA updates and it all seems to work until after the device rejects the received image late in the process of bootloading (which might have additional criteria for accepting the received image).

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

ToT

### Platform

darwin, nrf

fessehaeve, 2023-08-16:
@bzbarsky-apple 

```
===== VendorID Field
This field SHALL contain the assigned Vendor ID of the Node invoking this command, as it would appear in that Node's <<ref_BasicInfoCluster_VendorID,Basic Information Cluster VendorID>> attribute.
```

This VendorID field from the AnnounceOTAProvider command, is it of the provider itself or of the destination Node?

bzbarsky-apple, 2023-08-28:
@fessehaeve It's the vendor ID of the node implementing the client side of the OTA Provider cluster.  That is, the vendor id of the node that wants to check whether an updated image is available for it.