project-chip/connectedhomeip #25719: [BUG]

NovakiDouglas, 2023-03-16:
### Reproduction steps

when I run the command below I get the error listed
./scripts/build/build_examples.py --target android-arm64-chip-tool build

➜  connectedhomeip git:(master) ✗ ./scripts/build/build_examples.py --target android-arm64-chip-tool build
2023-03-16 12:30:35 INFO    Building targets: android-arm64-chip-tool
2023-03-16 12:30:35 INFO    Preparing builder 'android-arm64-chip-tool'
2023-03-16 12:30:35 INFO    Generating /Users/novaki/Documents/connectedhomeip/out/android-arm64-chip-tool
2023-03-16 12:30:35 INFO    Setting up Android deps through Gradle
2023-03-16 12:30:38 INFO    > Task :buildSrc:compileJava NO-SOURCE
2023-03-16 12:30:38 INFO    > Task :buildSrc:compileGroovy UP-TO-DATE
2023-03-16 12:30:38 INFO    > Task :buildSrc:processResources NO-SOURCE
2023-03-16 12:30:38 INFO    > Task :buildSrc:classes UP-TO-DATE
2023-03-16 12:30:38 INFO    > Task :buildSrc:jar UP-TO-DATE
2023-03-16 12:30:38 INFO    > Task :buildSrc:assemble UP-TO-DATE
2023-03-16 12:30:38 INFO    > Task :buildSrc:compileTestJava NO-SOURCE
2023-03-16 12:30:38 INFO    > Task :buildSrc:compileTestGroovy NO-SOURCE
2023-03-16 12:30:38 INFO    > Task :buildSrc:processTestResources NO-SOURCE
2023-03-16 12:30:38 INFO    > Task :buildSrc:testClasses UP-TO-DATE
2023-03-16 12:30:38 INFO    > Task :buildSrc:test NO-SOURCE
2023-03-16 12:30:38 INFO    > Task :buildSrc:check UP-TO-DATE
2023-03-16 12:30:38 INFO    > Task :buildSrc:build UP-TO-DATE
2023-03-16 12:30:38 INFO    > Task :copyArtifacts UP-TO-DATE
2023-03-16 12:30:38 INFO    > Task :setUpAndroidDeps
2023-03-16 12:30:38 INFO    
2023-03-16 12:30:38 INFO    BUILD SUCCESSFUL in 2s
2023-03-16 12:30:38 INFO    4 actionable tasks: 1 executed, 3 up-to-date
2023-03-16 12:30:38 INFO    Command ['python3', 'third_party/android_deps/set_up_android_deps.py'] completed
2023-03-16 12:30:38 INFO    Setting up Java deps
2023-03-16 12:30:46 INFO    Command ['third_party/java_deps/set_up_java_deps.sh'] completed
2023-03-16 12:30:46 INFO    Building /Users/novaki/Documents/connectedhomeip/out/android-arm64-chip-tool
2023-03-16 12:30:46 INFO    Building JNI android-arm64-chip-tool
2023-03-16 12:30:46 INFO    ninja: Entering directory `/Users/novaki/Documents/connectedhomeip/out/android-arm64-chip-tool'
2023-03-16 12:30:47 INFO    [0/1] Regenerating ninja files
2023-03-16 12:30:50 INFO    [1/458] stamp obj/third_party/jsoncpp/jsoncpp.stamp
2023-03-16 12:30:50 INFO    [2/458] cp ../../third_party/java_deps/artifacts/jsr305-3.0.2.jar lib/third_party/java_deps/jsr305-3.0.2.jar
2023-03-16 12:30:50 INFO    [3/458] cp ../../third_party/java_deps/artifacts/protobuf-java-3.22.0.jar lib/third_party/java_deps/protobuf-java-3.22.0.jar
2023-03-16 12:30:50 INFO    [4/458] stamp obj/third_party/java_deps/annotation.stamp
2023-03-16 12:30:50 INFO    [5/458] stamp obj/third_party/java_deps/protobuf-java.stamp
➜  connectedhomeip git:(master) ✗ ./scripts/build/build_examples.py --target android-arm64-chip-tool build
2023-03-16 12:32:57 INFO    Building targets: android-arm64-chip-tool
2023-03-16 12:32:57 INFO    Preparing builder 'android-arm64-chip-tool'
2023-03-16 12:32:57 INFO    Generating /Users/novaki/Documents/connectedhomeip/out/android-arm64-chip-tool
2023-03-16 12:32:57 INFO    Setting up Android deps through Gradle
2023-03-16 12:32:58 INFO    > Task :buildSrc:compileJava NO-SOURCE
2023-03-16 12:32:58 INFO    > Task :buildSrc:compileGroovy UP-TO-DATE
2023-03-16 12:32:58 INFO    > Task :buildSrc:processResources NO-SOURCE
2023-03-16 12:32:58 INFO    > Task :buildSrc:classes UP-TO-DATE
2023-03-16 12:32:58 INFO    > Task :buildSrc:jar UP-TO-DATE
2023-03-16 12:32:58 INFO    > Task :buildSrc:assemble UP-TO-DATE
2023-03-16 12:32:58 INFO    > Task :buildSrc:compileTestJava NO-SOURCE
2023-03-16 12:32:58 INFO    > Task :buildSrc:compileTestGroovy NO-SOURCE
2023-03-16 12:32:58 INFO    > Task :buildSrc:processTestResources NO-SOURCE
2023-03-16 12:32:58 INFO    > Task :buildSrc:testClasses UP-TO-DATE
2023-03-16 12:32:58 INFO    > Task :buildSrc:test NO-SOURCE
2023-03-16 12:32:58 INFO    > Task :buildSrc:check UP-TO-DATE
2023-03-16 12:32:58 INFO    > Task :buildSrc:build UP-TO-DATE
2023-03-16 12:32:58 INFO    > Task :copyArtifacts UP-TO-DATE
2023-03-16 12:32:58 INFO    > Task :setUpAndroidDeps
2023-03-16 12:32:58 INFO    
2023-03-16 12:32:58 INFO    BUILD SUCCESSFUL in 734ms
2023-03-16 12:32:58 INFO    4 actionable tasks: 1 executed, 3 up-to-date
2023-03-16 12:32:58 INFO    Command ['python3', 'third_party/android_deps/set_up_android_deps.py'] completed
2023-03-16 12:32:58 INFO    Setting up Java deps
2023-03-16 12:33:06 INFO    Command ['third_party/java_deps/set_up_java_deps.sh'] completed
2023-03-16 12:33:06 INFO    Building /Users/novaki/Documents/connectedhomeip/out/android-arm64-chip-tool
2023-03-16 12:33:06 INFO    Building JNI android-arm64-chip-tool
2023-03-16 12:33:06 INFO    ninja: Entering directory `/Users/novaki/Documents/connectedhomeip/out/android-arm64-chip-tool'
2023-03-16 12:33:06 INFO    [0/1] Regenerating ninja files
[... 39 lines cut ...]
  File "/Users/novaki/Documents/connectedhomeip/./scripts/build/build_examples.py", line 211, in cmd_build
    context.obj.Build()
  File "/Users/novaki/Documents/connectedhomeip/scripts/build/build/__init__.py", line 71, in Build
    builder.build()
  File "/Users/novaki/Documents/connectedhomeip/scripts/build/builders/builder.py", line 99, in build
    self._build()
  File "/Users/novaki/Documents/connectedhomeip/scripts/build/builders/android.py", line 428, in _build
    self._Execute(
  File "/Users/novaki/Documents/connectedhomeip/scripts/build/builders/builder.py", line 104, in _Execute
    self._runner.Run(cmdarray, title=title)
  File "/Users/novaki/Documents/connectedhomeip/scripts/build/runner/shell.py", line 73, in Run
    raise Exception('Command %r failed: %d' % (cmd, code))
Exception: Command ['ninja', '-C', '/Users/novaki/Documents/connectedhomeip/out/android-arm64-chip-tool'] failed: 1

### Bug prevalence

impossible to compile

### GitHub hash of the SDK that was being used

061f80a946d320c4b677a5a86352bcbe117080af

### Platform

android

bzbarsky-apple, 2023-03-16:
From that log:
```
 2023-03-16 12:33:07 INFO kotlinc: command not found
```

@NovakiDouglas did you follow the instructions in `docs/guides/android_building.md` that say:

> The build also requires `kotlinc` to be in your `$PATH`.

?

NovakiDouglas, 2023-03-16:
yes I followed all the instructions with the same versions
2x on linux and 2x on mac

two more notes:
- could you provide an apk already compiled for testing
- JAVA_HOME has the wrong path in the latest versions of android studio they modified
image below shows how it is currently and how it is in the new model

[image]

NovakiDouglas, 2023-03-16:
@bzbarsky-apple if you can provide me with an app built for testing

bzbarsky-apple, 2023-03-17:
>  yes I followed all the instructions with the same versions

Ok, so is `kotlinc` in your `PATH`?

Past that, you'd want to talk to one of the maintainers of the java/android bits.

NovakiDouglas, 2023-03-17:
no my PATH does not have kotlinc
could you help me with this problem? or at least send me an APK for testing

bzbarsky-apple, 2023-03-17:
> no my PATH does not have kotlinc

Well, you need to fix that.  I wish I could tell you how, but I have never tried building any of the Java or Android bits, sorry.

mmt-10, 2023-03-17:
If your environment is ubuntu, you may be able to install kotlin as below command.
`sudo snap install --classic kotlin`

NovakiDouglas, 2023-03-17:
after installing kotlin on mac i was able to generate the app

closed 2023-03-17 by NovakiDouglas

swap9391, 2023-06-06:
getting error for **set_up_java_deps**.sh file

/Android/connectedhomeip$ ./scripts/build/build_examples.py --target android-arm64-chip-tool build
2023-06-06 19:10:09 INFO    Building targets: android-arm64-chip-tool
2023-06-06 19:10:09 INFO    Preparing builder 'android-arm64-chip-tool'
2023-06-06 19:10:09 INFO    Generating /home/user/Android/connectedhomeip/out/android-arm64-chip-tool
2023-06-06 19:10:09 INFO    Setting up Android deps through Gradle
2023-06-06 19:10:10 INFO    > Task :buildSrc:compileJava NO-SOURCE
2023-06-06 19:10:12 INFO    > Task :buildSrc:compileGroovy
2023-06-06 19:10:12 INFO    > Task :buildSrc:processResources NO-SOURCE
2023-06-06 19:10:12 INFO    > Task :buildSrc:classes
2023-06-06 19:10:12 INFO    > Task :buildSrc:jar
2023-06-06 19:10:12 INFO    > Task :buildSrc:assemble
2023-06-06 19:10:12 INFO    > Task :buildSrc:compileTestJava NO-SOURCE
2023-06-06 19:10:12 INFO    > Task :buildSrc:compileTestGroovy NO-SOURCE
2023-06-06 19:10:12 INFO    > Task :buildSrc:processTestResources NO-SOURCE
2023-06-06 19:10:12 INFO    > Task :buildSrc:testClasses UP-TO-DATE
2023-06-06 19:10:12 INFO    > Task :buildSrc:test NO-SOURCE
2023-06-06 19:10:12 INFO    > Task :buildSrc:check UP-TO-DATE
2023-06-06 19:10:12 INFO    > Task :buildSrc:build
2023-06-06 19:10:12 INFO    > Task :copyArtifacts
2023-06-06 19:10:12 INFO    > Task :setUpAndroidDeps
2023-06-06 19:10:12 INFO    
2023-06-06 19:10:12 INFO    BUILD SUCCESSFUL in 2s
2023-06-06 19:10:12 INFO    4 actionable tasks: 4 executed
2023-06-06 19:10:13 INFO    Command ['python3', 'third_party/android_deps/set_up_android_deps.py'] completed
2023-06-06 19:10:13 INFO    Setting up Java deps
2023-06-06 19:10:13 WARNING third_party/java_deps/set_up_java_deps.sh: line 20: curl: command not found
2023-06-06 19:10:13 WARNING third_party/java_deps/set_up_java_deps.sh: line 21: curl: command not found
2023-06-06 19:10:13 WARNING third_party/java_deps/set_up_java_deps.sh: line 22: curl: command not found
2023-06-06 19:10:13 WARNING third_party/java_deps/set_up_java_deps.sh: line 23: curl: command not found
2023-06-06 19:10:13 WARNING third_party/java_deps/set_up_java_deps.sh: line 24: curl: command not found
2023-06-06 19:10:13 WARNING third_party/java_deps/set_up_java_deps.sh: line 25: curl: command not found
2023-06-06 19:10:13 WARNING third_party/java_deps/set_up_java_deps.sh: line 26: curl: command not found
2023-06-06 19:10:13 WARNING third_party/java_deps/set_up_java_deps.sh: line 27: curl: command not found
Traceback (most recent call last):
  File "/home/user/Android/connectedhomeip/./scripts/build/build_examples.py", line 230, in <module>
    main(auto_envvar_prefix='CHIP')
  File "/home/user/Android/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/user/Android/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/user/Android/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    rv.append(sub_ctx.command.invoke(sub_ctx))
  File "/home/user/Android/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/user/Android/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/user/Android/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/user/Android/connectedhomeip/./scripts/build/build_examples.py", line 220, in cmd_build
    context.obj.Build()
  File "/home/user/Android/connectedhomeip/scripts/build/build/__init__.py", line 67, in Build
    self.Generate()
  File "/home/user/Android/connectedhomeip/scripts/build/build/__init__.py", line 62, in Generate
    builder.generate()
  File "/home/user/Android/connectedhomeip/scripts/build/builders/android.py", line 321, in generate
    self._Execute(
  File "/home/user/Android/connectedhomeip/scripts/build/builders/builder.py", line 104, in _Execute
    self._runner.Run(cmdarray, title=title)
  File "/home/user/Android/connectedhomeip/scripts/build/runner/shell.py", line 73, in Run
    raise Exception('Command %r failed: %d' % (cmd, code))
Exception: Command ['third_party/java_deps/set_up_java_deps.sh'] failed: 127

mmt-10, 2023-06-06:
I think you should install `curl`.

**curl: command not found**

> 2023-06-06 19:10:13 WARNING third_party/java_deps/set_up_java_deps.sh: line 20: curl: command not found
> 2023-06-06 19:10:13 WARNING third_party/java_deps/set_up_java_deps.sh: line 21: curl: command [...]

swap9391, 2023-06-07:
@mmt-10 yes it worked 

$ sudo apt update
$ sudo apt upgrade
$ sudo apt install curl