project-chip/connectedhomeip #25722: [1.0] The Linux chip-tool and Android's chip-tool generate NOC method are different? (RCAC, ICAC)

resghst, 2023-03-17:
### Reproduction steps

Because I want to use the same fabric control device, I try to exchange RCAC and ICAC between the Linux chip-tool and Android chip-tool, but the two CA settings are different.
The Linux chip-tool maybe use  ICAC flow and Android's chip-tool use without optional ICAC (Sepc 1.0 Page.300)

The Linux chip-tool setting has the following: (/tmp/chip_tool_config.XXXX.ini)
1. ExampleCAIntermediateCert0
2. ExampleCARootCert0
3. ExampleOpCredsCAKey0
4. ExampleOpCredsICAKey0

Android chip-tool setting has the following:
1. AndroidCARootCert1
2. AndroidDeviceControllerKey

Thus, how to transfer the Linux chip-tool config and Android chip-tool config for the same fabric control a device? (This transfer chip-tool config is bidirectional.)

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

5407db2de6000f5c953020375d4156774479c9ec

### Platform

android, raspi

### Type

Spec Compliance Issue, Platform Issue

bzbarsky-apple, 2023-03-17:
>  The Linux chip-tool maybe use ICAC flow and Android's chip-tool use without optional ICAC (Sepc 1.0 Page.300)

Yes.

> how to transfer the Linux chip-tool config and Android chip-tool config for the same fabric control a device?

These are both test tools; neither one is really meant for actual long-term control of a device, much less sharing a single fabric across the two of them.

closed 2023-03-17 by bzbarsky-apple

WladMl, 2023-08-07:
Hello @resghst 
Have you managed to solve the issue?

resghst, 2023-08-11:
Hi @WladMl 
This question tells us Matter provides two methods to generate NOC (RCAC, ICAC), and the Linux chip-tool and the Android chip-tool use different methods.
　
Assume you want to use the same fabric control device from different controllers
you need to ensure the device's NOC type before exchanging NOC between two devices.