project-chip/connectedhomeip #25734: [BUG] Chip-cert-bins docker script failing on TOT

raju-apple, 2023-03-17:
### Reproduction steps

Trying to build the cert-bins docker image and its failing : 

Steps used : 

docker buildx build --build-arg COMMITHASH=bbb8e935b250e5dc291021cb220b192a50cc98cf --platform linux/arm64 --tag connectedhomeip/chip-cert-bins:bbb8e935b250e5dc291021cb220b192a50cc98cf --push .

Probably a zap issue ?

Error Snippet is here : 

```
#0 0.820 🚀 Executing: git log -1 --format={"hash": "%H","timestamp": %ct}
#0 0.825 🔍 Git commit: ac6dd831c2634315b5c847e37cc9b44bb7c22e5e from Mon Mar 06 2023 20:44:30 GMT+0000 (Coordinated Universal Time)
#0 0.848 🔍 Current src hash: tt1kGIKwkwmZc8maXjP1W3u9tu4=
#0 0.850 🔍 Current src-shared hash: g/WbG5bVfcehj7QTAknSkxvoR5c=
#0 0.851 🔍 Previous src hash: tt1kGIKwkwmZc8maXjP1W3u9tu4=
#0 0.851 🔍 Previous src-shared hash: g/WbG5bVfcehj7QTAknSkxvoR5c=
#0 0.851 👍 There were no changes to front-end code, so we don't have to rebuild the SPA.
#0 0.851 🚀 Executing: npx tsc --build ./tsconfig.json
#0 2.786 🚀 Executing: npx copyfiles ./src-electron/**/*.sql ./src-electron/icons/* ./dist/
#0 2.877 🚀 Executing: npx node --unhandled-rejections=strict /opt/zap-v2023.03.06-nightly/dist/src-electron/main-process/main.js generate -z /root/connectedhomeip/src/app/zap-templates/zcl/zcl.json -g /root/connectedhomeip/src/app/zap-templates/matter-idl.json -i /root/connectedhomeip/scripts/examples/../../examples/placeholder/linux/apps/app1/config.zap -o /root/connectedhomeip/scripts/examples/../../zzz_generated/placeholder/app1/zap-generated --tempState
#0 3.135 Expected node versions: v14.x.x,v16.x.x
#0 3.136 Provided node version: v12.22.12
#0 3.136 WARNING: you are using different node version than recommended.
#0 3.141 🔧 Using temporary state directory: /tmp/zapvn0hytrg/zap.o4VDoC
#0 3.141 🤖 ZAP generation started: 
#0 3.141     🔍 input files: /root/connectedhomeip/scripts/examples/../../examples/placeholder/linux/apps/app1/config.zap
#0 3.141     🔍 output pattern: /root/connectedhomeip/scripts/examples/../../zzz_generated/placeholder/app1/zap-generated
#0 3.141     🔍 using templates: /root/connectedhomeip/src/app/zap-templates/matter-idl.json
#0 3.141     🔍 using zcl data: /root/connectedhomeip/src/app/zap-templates/zcl/zcl.json
#0 3.141     🔍 zap version: ver. undefined, featureLevel 94, commit: ac6dd831c2634315b5c847e37cc9b44bb7c22e5e from 2023-03-06T20:44:30.000Z, mode: source, exe: node
#0 3.763 🕐 Setup time: 622ms 
#0 3.763 👉 using input file: /root/connectedhomeip/scripts/examples/../../examples/placeholder/linux/apps/app1/config.zap
#0 3.855 👉 using output destination: /root/connectedhomeip/scripts/examples/../../zzz_generated/placeholder/app1/zap-generated
#0 3.857 🕐 File loading time: 94ms
#0 4.140 🤖 Generating files:
#0 4.140     ✍  /root/connectedhomeip/zzz_generated/placeholder/app1/zap-generated/Clusters.matter
#0 4.140 🕐 Generation time: 283ms 
#0 4.170 😎 All done: 3s 350ms.
#0 4.175 Searching for zcl file from /root/connectedhomeip/scripts/examples/../../examples/placeholder/linux/apps/app1/config.zap
#0 4.175 Unable to check version using %r. Assuming version is ok. ['node', 'src-script/zap-start.js']
#0 4.175 Files generated in: /root/connectedhomeip/scripts/examples/../../zzz_generated/placeholder/app1/zap-generated
#0 4.231 Searching for zcl file from /root/connectedhomeip/scripts/examples/../../examples/placeholder/linux/apps/app1/config.zap
#0 4.231 Error: /root/connectedhomeip/scripts/examples/../../examples/placeholder/linux/../templates/templates.json does not exists or is not a file.
------
Dockerfile:275
--------------------
 273 |         esac
 274 |     
 275 | >>> RUN scripts/examples/gn_build_test_example.sh app1
 276 |     
 277 |     RUN source scripts/activate.sh && scripts/build_python.sh -m platform -d true -i no
--------------------
ERROR: failed to solve: process "/bin/bash -c scripts/examples/gn_build_test_example.sh app1" did not complete successfully: exit code: 1
raju@Rajus-MacBook-Pro chip-cert-bins %
``` 
[Docker_chip_cert_bins_error.rtf.zip](https://github.com/project-chip/connectedhomeip/files/11003893/Docker_chip_cert_bins_error.rtf.zip)

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

bbb8e935b250e5dc291021cb220b192a50cc98cf

### Platform

raspi

### Platform Version(s)

Linux/arm64

### Anything else?

Trying to build the cert-bins docker image and its failing : 

Steps used : 

docker buildx build --build-arg COMMITHASH=bbb8e935b250e5dc291021cb220b192a50cc98cf --platform linux/arm64 --tag connectedhomeip/chip-cert-bins:bbb8e935b250e5dc291021cb220b192a50cc98cf --push .

Error Snippet : 

```
#0 0.820 🚀 Executing: git log -1 --format={"hash": "%H","timestamp": %ct}
#0 0.825 🔍 Git commit: ac6dd831c2634315b5c847e37cc9b44bb7c22e5e from Mon Mar 06 2023 20:44:30 GMT+0000 (Coordinated Universal Time)
#0 0.848 🔍 Current src hash: tt1kGIKwkwmZc8maXjP1W3u9tu4=
#0 0.850 🔍 Current src-shared hash: g/WbG5bVfcehj7QTAknSkxvoR5c=
#0 0.851 🔍 Previous src hash: tt1kGIKwkwmZc8maXjP1W3u9tu4=
#0 0.851 🔍 Previous src-shared hash: g/WbG5bVfcehj7QTAknSkxvoR5c=
#0 0.851 👍 There were no changes to front-end code, so we don't have to rebuild the SPA.
#0 0.851 🚀 Executing: npx tsc --build ./tsconfig.json
#0 2.786 🚀 Executing: npx copyfiles ./src-electron/**/*.sql ./src-electron/icons/* ./dist/
#0 2.877 🚀 Executing: npx node --unhandled-rejections=strict /opt/zap-v2023.03.06-nightly/dist/src-electron/main-process/main.js generate -z /root/connectedhomeip/src/app/zap-templates/zcl/zcl.json -g /root/connectedhomeip/src/app/zap-templates/matter-idl.json -i /root/connectedhomeip/scripts/examples/../../examples/placeholder/linux/apps/app1/config.zap -o /root/connectedhomeip/scripts/examples/../../zzz_generated/placeholder/app1/zap-generated --tempState
#0 3.135 Expected node versions: v14.x.x,v16.x.x
#0 3.136 Provided node version: v12.22.12
#0 3.136 WARNING: you are using different node version than recommended.
#0 3.141 🔧 Using temporary state directory: /tmp/zapvn0hytrg/zap.o4VDoC
#0 3.141 🤖 ZAP generation started: 
#0 3.141     🔍 input files: /root/connectedhomeip/scripts/examples/../../examples/placeholder/linux/apps/app1/config.zap
#0 3.141     🔍 output pattern: /root/connectedhomeip/scripts/examples/../../zzz_generated/placeholder/app1/zap-generated
#0 3.141     🔍 using templates: /root/connectedhomeip/src/app/zap-templates/matter-idl.json
#0 3.141     🔍 using zcl data: /root/connectedhomeip/src/app/zap-templates/zcl/zcl.json
#0 3.141     🔍 zap version: ver. undefined, featureLevel 94, commit: ac6dd831c2634315b5c847e37cc9b44bb7c22e5e from 2023-03-06T20:44:30.000Z, mode: source, exe: node
#0 3.763 🕐 Setup time: 622ms 
#0 3.763 👉 using input file: /root/connectedhomeip/scripts/examples/../../examples/placeholder/linux/apps/app1/config.zap
#0 3.855 👉 using output destination: /root/connectedhomeip/scripts/examples/../../zzz_generated/placeholder/app1/zap-generated
#0 3.857 🕐 File loading time: 94ms
#0 4.140 🤖 Generating files:
#0 4.140     ✍  /root/connectedhomeip/zzz_generated/placeholder/app1/zap-generated/Clusters.matter
#0 4.140 🕐 Generation time: 283ms 
#0 4.170 😎 All done: 3s 350ms.
#0 4.175 Searching for zcl file from /root/connectedhomeip/scripts/examples/../../examples/placeholder/linux/apps/app1/config.zap
#0 4.175 Unable to check version using %r. Assuming version is ok. ['node', 'src-script/zap-start.js']
#0 4.175 Files generated in: /root/connectedhomeip/scripts/examples/../../zzz_generated/placeholder/app1/zap-generated
#0 4.231 Searching for zcl file from /root/connectedhomeip/scripts/examples/../../examples/placeholder/linux/apps/app1/config.zap
#0 4.231 Error: /root/connectedhomeip/scripts/examples/../../examples/placeholder/linux/../templates/templates.json does not exists or is not a file.
------
Dockerfile:275
--------------------
 273 |         esac
 274 |     
 275 | >>> RUN scripts/examples/gn_build_test_example.sh app1
 276 |     
 277 |     RUN source scripts/activate.sh && scripts/build_python.sh -m platform -d true -i no
--------------------
ERROR: failed to solve: process "/bin/bash -c scripts/examples/gn_build_test_example.sh app1" did not complete successfully: exit code: 1
raju@Rajus-MacBook-Pro chip-cert-bins % 

```

raju-apple, 2023-03-17:
@andreilitvin can you take a look please ? does the ZAP need to be updated?

bzbarsky-apple, 2023-03-20:
Presumably something about #25378 and the changes it made?  @krypton36

bzbarsky-apple, 2023-03-20:
Does applying:
```
diff --git a/scripts/examples/gn_build_test_example.sh b/scripts/examples/gn_build_test_example.sh
index 52b56c2428..66ed9c3dd6 100755
--- a/scripts/examples/gn_build_test_example.sh
+++ b/scripts/examples/gn_build_test_example.sh
@@ -40,7 +40,7 @@ function runZAP() {
     "$CHIP_ROOT"/scripts/tools/zap/generate.py "$ZAP_INPUT_FILE" -o "$ZAP_OUTPUT_DIR"
 
     # Generates the specific files for the given zap configuration
-    TARGET_APP=$APP_DIR "$CHIP_ROOT"/scripts/tools/zap/generate.py "$ZAP_INPUT_FILE" -t "$INPUT_DIR"/../templates/templates.json -o "$ZAP_OUTPUT_DIR"
+    TARGET_APP=$APP_DIR "$CHIP_ROOT"/scripts/tools/zap/generate.py "$ZAP_INPUT_FILE" -t "$INPUT_DIR"/apps/$APP_DIR/templates/templates.json -o "$ZAP_OUTPUT_DIR"
 }
 
 function runGN() {
```
help?

raju-apple, 2023-03-20:
Thanks @krypton36 , will wait for your PR to land and then try to re-build on TOT

PR #25754 (2023-03-21): Add simulated app to python builder and update cert docker image
Summary:
- Fixes #25734
- Adds the simulated app the python builder
- Updated the Cert Docker file to build the simulated app with python builder.

Testing:
- Validated that python builder builds the simulated app with the docker command.
closed 2023-03-21 by bzbarsky-apple