project-chip/connectedhomeip #25743: [BUG] Matter Google SDK says Not a Matter-certified device

dhdanh, 2023-03-20:
### Reproduction steps

Steps
1. Build the FW with Matter cert test VID - **0xFFF1**
2. The Google Home app able to Commission the device successfully
3. Build the FW with my company VID - **0x136E**
4. The Google Home app says error "Not a Matter-certified device"
5. The Apple Home app is able to commission the device successfully
6. How do I build the FW with the company VID - **0x136E**? Please see below

`In order to use vendor VID/PID instead of standard SDK (0xFFF1/0x8001) we need to replace chip::Credentials::Examples::ExampleDACProvider with our own implementation

This means we need to create our own certificates (PAA/PAI/DAC) and CD

PAA/PAI/DAC - https://github.com/project-chip/connectedhomeip/blob/72f0471509f9bcba783940055975bd438b7d6e68/credentials/test/gen-test-attestation-certs.sh 

CD - https://github.com/project-chip/connectedhomeip/blob/master/credentials/test/gen-test-cds.sh 

CD should be signed with connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-Signing-Key.pem`

### Bug prevalence

all the time

### GitHub hash of the SDK that was being used

c1810b6d95cb2a946fc0ba71c6e63e88b15554cd

### Platform

esp32

umbraHare, 2023-03-20:
https://developers.home.google.com/matter/integration/create?hl=en

Spacefish, 2023-03-23:
I don't get it why they would artificially limit their Matter Platform to certified devices. I mean, isn't having the most versatile / most supported devices in your platform the name of the game / the thing that gets your platform adopted?