project-chip/connectedhomeip #25752: Need support on Binding with chip tool

rajeshkrishnan2k, 2023-03-20:
Hi,
Can someone give some examples for binding like what data we need to pass. Also how should we bind if we have light and light switch apps?

rajeshkrishnan2k, 2023-03-21:
Yes i ran the following command : $ ./chip-tool binding write binding '[{"Cluster":"0x6","node":"0x462","endpoint":"0x1","fabricIndex":1}]' "0x46D" 1

after running this i ran into this error in the end device. 
 Binding: Failed to create session for unicast binding to device 0000000000000462: 3

Should i change any ACL before trying to bind or is there something else i am missing. 

Thanks.

bzbarsky-apple, 2023-03-21:
That said, "3" is CHIP_ERROR_INCORRECT_STATE.  The most likely reason for that is that the binding manager was not initialized with a CASE session manager.   Is the server in question doing `BindingManager::GetInstance().Init()`?  If so, what is it passing in?

rajeshkrishnan2k, 2023-03-22:
@bzbarsky-apple  node 462 is present on fabric.

cjandhyala, 2023-03-22:
@rajeshkrishnan2k Assuming you have chip-tool as the commissioner and you have on LB (Light bulb) and LS (Light switch), LS is the DUT and you want to control LB with LS.

1. Provision both LB (node1) and LS (node2)
2. using chip-tool , write the binding entry in node2. i.e. ./chip-tool binding write binding '[{"node" : 1 , "cluster" : "0x0006" , "endpoint" : <endpt> }]' node2 <endpt>
3.  using chip-tool, write ACL in node1 , ./chip-tool accesscontrol write acl '[{"privilege": 3, "authMode": 2, "subjects": [node2], "targets": [{"cluster": 6, "endpoint": 1, "deviceType": null}]}]' node1 0
4. Now you should be able to use node2 to send unicast command
5. If you are testing multicast on the binding entries, you will need to first create grp key, then map it to a grpI,  add grp and use that grp in the binding entry (in step2 above). Provide the ACL accordingly. 

If you are testing with example app on linux platform, build the all-cluster-app with chip_build_libshell=true build flag. Provision this as your node2. Once the device is provisioned, you hit enter at the shell prompt, you will get shell and you can send 'switch on' or off commands.

closed 2023-03-24 by rajeshkrishnan2k

Numeez, 2024-01-09:
@cjandhyala  Are you sure the endpoint in the step 2 should be set to nothing or it is just a typo

bzbarsky-apple, 2024-01-10:
@Numeez That looks like a typo.