project-chip/connectedhomeip #25761: Battery Level not showing up in Apple Homekit

absoluteabutaj, 2023-03-21:
Hello there,

I'm trying to show the battery level of the end device in Apple Homekit. 

Device Information:
Hardware: nRF52840 DevKit
SDK: NCS 2.2.0
Sample: Window Covering

I've added the power source cluster to the window covering application (zap) and reporting,the battery level (mV) and

absoluteabutaj, 2023-03-22:
@bzbarsky-apple - I did added the power source configuration cluster as well, however it's still not showing up in the homekit.

Power Source Cluster -> Server.

bzbarsky-apple, 2023-03-22:
@absoluteabutaj In that case, you should file a feedback assistant ticket on Home as follows:

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

absoluteabutaj, 2023-03-23:
@bzbarsky-apple - Just in case to understand, I've the off-the-self Matter product and when I commission in the Apple fabric it shows the battery percentage.

Is it something not right with the implementation I've?

bzbarsky-apple, 2023-03-23:
> Is it something not right with the implementation I've?

Probably, but it's hard to say without more information than you've provided so far.  Things like exact values of the relevant attributes in the Power Source Configuration and Power Source clusters, the exact endpoint configuration, the device type ids involved, etc.

absoluteabutaj, 2023-03-23:
@bzbarsky-apple - Please find the additional information regarding the clusters, attributes and it's default values in the attached PDF.
[power-source-configuration-cluster-defaults.pdf](https://github.com/project-chip/connectedhomeip/files/11050088/power-source-configuration-cluster-defaults.pdf)

Device Type - 514 (0x0202)

bzbarsky-apple, 2023-03-23:
> Device Type - 514 (0x0202)

On which endpoint?

Which endpoint is the Power Source Configuration cluster on?

absoluteabutaj, 2023-03-23:
Both the Power Source Configuration and Power Source are in endpoint 0.

End point 1 is window covering (0x0202).

bzbarsky-apple, 2023-03-24:
@absoluteabutaj OK.  In that case, I'm out of ideas about what could plausibly go wrong here.  Please file the Feedback Assistant ticket.

s07641069, 2023-04-14:
@absoluteabutaj On my side I able to see battery status after correctly added Power Source Configuration and Power Source clusters.
To verify on your side, just run all automated certification tests for Power Source Configuration and Power Source. This was helpful for me.

absoluteabutaj, 2023-04-15:
@s07641069 - Could you point me to the guide on how I can run the Automated Certification tests for Power Source Configuration?

s07641069, 2023-04-18:
@absoluteabutaj 
./chip-tool tests Test_TC_PS_1_1 --nodeId  $NODE_ID
Power Source Configuration and Power Source clusters tests: Test_TC_PS_1_1, Test_TC_PS_2_1, Test_TC_PSCFG_1_1,  Test_TC_PSCFG_2_1, TC-PSCFG-2.2

absoluteabutaj, 2023-04-19:
@s07641069 - Thanks.

woody-apple, 2023-06-23:
Closing, please reopen if this is still an issue.

closed 2023-06-23 by woody-apple

arrow53, 2023-11-19:
@s07641069 for EP1 what do you have 0xFFFC `FeatureMap` set to? I have it set to 0, but get this error
```
***** Test Failure: value expects the field with value 1 to be set but it is not.
```

I also see this failure on 0x000A `ActiveWiredFaults`

```
***** Test Failure: status value mismatch: expected 0 but got 1
```

What values do you have for these attributes? Or, can you provide a hint if you've seen such errors when you run the tests.