project-chip/connectedhomeip #25777: [1.0] how to remove an device while it is offline?

MissionaryHGQ, 2023-03-22:
### Reproduction steps

If we want to remove a commissioned device, we can call the api "MTRBaseClusterOperationalCredentials  removeFabricWithParams". But in advanced, you should get the device from api "MTRDevicecontroller getBaseDevice:(uint64_t)deviceID queue:(dispatch_queue_t)queue completionHandler:(MTRDeviceConnectionCallback)completionHandler".
So the question is that, once the device is offline, you won't get the device from the api,
and if you don't call any api, you just remove the device from your own server,   you will suffer a new question in a situation described below:
1、you developer home app(named urHome.app),  your app commissioned a matter device.
2、you share the device through paring code to Apple Home.app.
3、the device is offline, then you still remove the app.
4、the device is online now, you use Apple Home.app share the device to your app, or you just use you app to start a new
commission with the device, finally you got the error on your develop console:
 [CTL] Failed to perform commissioning step 13, DevicePairingDelegate Commissioning complete. NodeId 131 Status ../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/controller/CHIPDeviceController.cpp:1311: CHIP Error 0x0000007E: Trying to add a NOC for a fabric that already exists;
commission fail, error: The device is already a member of this fabric.

how to solve this problem, can anyone give a tip?

### Bug prevalence

always

### GitHub hash of the SDK that was being used

iOS 16.2 system Matter.framework

### Platform

darwin

### Platform Version(s)

1.0

### Type

Platform Issue

bzbarsky-apple, 2023-03-22:
> how to solve this problem, can anyone give a tip?

If you "remove a device" when you can't talk to it, obviously you can't remove your fabric's NOC from the device.

If you then try to commission it into your fabric again, it will fail as above.

The only options are:

1. Detect when you get the relevant error during commissioning, remove the relevant fabric from the device at that point (since you are now communicating with the device), and try commissioning again.
2. When you start commissioning, after you establish PASE check whether the device is already on your fabric and either stop the commissioning attempt or remove it from your fabric and proceed with the new commissioning, depending on what your goal is.

closed 2023-03-22 by bzbarsky-apple

MissionaryHGQ, 2023-04-25:
in ios16.5beta1, I can pair the device again after I remove the device without calling  API "removeFabricWithParams", but in 16.5beta2, it will always failed. I don't know what different between two system versions.

bzbarsky-apple, 2023-04-25:
>  after I remove the device without calling API "removeFabricWithParams"

What exact steps does "remove the device" mean here?

bzbarsky-apple, 2023-04-25:
@MissionaryHGQ What I strongly recommend is filing a feedback assistant ticket with as much information as you can provide: sysdiagnoses from the working and failing case, what your code interacting with the framework looks like, if you are able to share it, etc.  Please put the feedback assistant ticket number here so I can follow up on it.

MrLiuYunPing, 2023-12-13:
> If you "remove a device" when you can't talk to it, obviously you can't remove your fabric's NOC from the device.
>
> If you then try to commission it into your fabric again, it will fail as above. [...]

How to remove fabric when pairing? As you know,the device is deleted from the server, we don't know the nodeid,and we can only know the product id and vendor id of the device when commissioning from the qr code or manual code. But, we cannot identify which device is it even when we do not reality delete the device by just hiding it. 

> If you "remove a device" when you can't talk to it, obviously you can't remove your fabric's NOC from the device.
>
> If you then try to commission it into your fabric again, it will fail as above. [...]

How to remove fabric when pairing? As you know,the device is deleted from the server, we don't know the nodeid,and we can only know the product id and vendor id of the device when commissioning from the qr code or manual code. But, we cannot identify which device is it even when we do not reality delete the device by just hiding it.

bzbarsky-apple, 2023-12-13:
> How to remove fabric when pairing?

Using the RemoveFabric command.

> But, we cannot identify which device is it even when we do not reality delete the device by just hiding it.

When you connect to the device via PASE, you can:

1. Detect whether it's already in your fabric
2. If it is, read the node id from the device.

MrLiuYunPing, 2023-12-14:
> Using the RemoveFabric command.
>
>
> When you connect to the device via PASE, you can:
>
> 1. Detect whether it's already in your fabric
> 2. If it is, read the node id from the device.

Thank you very much! But, I can detect whether the device is in my fabric by the error code returned when pairing, how can I read the nodeid? Is there an API can be invoked?  If exists, what's the name of the class of method? As you know the MTRBaseDevice should be initialized with a controller and nodeid ,if I cannot get the nodeid,how can I remove the fabric?

bzbarsky-apple, 2023-12-15:
> how can I read the nodeid?

You can read the "Fabrics" attribute from the "Operational Credentials" cluster, with fabric filtering turned off.  Find the entry in the list that has the right RootPublicKey and FabricID, then get the NodeID from that entry.

> MTRBaseDevice should be initialized with a controller and nodeid ,if I cannot get the nodeid,how can I remove the fabric?

If you're doing this during _commissioning_, you get the MTRBaseDevice using https://github.com/project-chip/connectedhomeip/blob/6c6fe0a76b1b4a79f8dbf1102324d7bdf74d9e80/src/darwin/Framework/CHIP/MTRDeviceController.h#L178-L182

Note that the node id used there (which is the node id you plan to assign to the device) is unrelated to the node id the device already has if it was previously commissioned into your fabric.

MrLiuYunPing, 2023-12-18:
> You can read the "Fabrics" attribute from the "Operational Credentials" cluster, with fabric filtering turned off. Find the entry in the list that has the right RootPublicKey and FabricID, then get [...]

Thanks very much! I will try this.

MrLiuYunPing, 2024-01-03:
> You can read the "Fabrics" attribute from the "Operational Credentials" cluster, with fabric filtering turned off. Find the entry in the list that has the right RootPublicKey and FabricID, then get [...]

I have tried, but the MatterSupport API will not call the override func commissionDevice(in home: MatterAddDeviceRequest.Home?, onboardingPayload: String, commissioningID: UUID) async throws method, when the device has been already commissioned in the ios system. How can we solve this problem?

bzbarsky-apple, 2024-01-03:
That part should work.  If it's not working for you, please file a feedback assistant issue per the instructions above with details about _exactly_ what you are doing.