project-chip/connectedhomeip #25778: How to start a commission from sharing device without input wifi account and password.

MissionaryHGQ, 2023-03-22:
### Reproduction steps

1、You commissioned  a matter device using  Apple Home.app, then you click the button "turn on pairing mode" to share the device  and get a pairing code A.
2、You developed a home app named urHome.app,  you call the api "MTRDeviceContrller pairDevice:(uint64_t)deviceID onboardingPayload:(NSString *)onboardingPayload" using the paring code A you got from Apple Home. After a while , you should prompt up an alertView to get the wifi ssid and password to go on commissioning. This procedure is unreasonable as it's shared from a commissioned app.
3、then you call the api "MTRDeviceController pairDevice:(uint64_t)deviceID discriminator:(uint16_t)discriminator   setupPINCode:", to avoid inputing wifi ssid and password. but the question is that,  you use the api "MTRManualSetupPayloadParser populatePayload", you won't get a correct discriminator from the paing code A, so you can't go on commissioning. 
So how to get a correct discriminator, or how to start a commission from sharing device without input wifi ssid and password. can anyone help me?

### Bug prevalence

always

### GitHub hash of the SDK that was being used

iOS 16.2 system Matter.framework

### Platform

darwin

### Platform Version(s)

1.0

bzbarsky-apple, 2023-03-22:
> After a while , you should prompt up an alertView to get the wifi ssid and password to go on commissioning.

There's nothing in Matter.framework itself that puts up such an alertView.  Who is putting it up and why?

> you won't get a correct discriminator from the paing code A

Is this a numeric pairing code (as opposed to a QR code), so you're just getting a short discriminator?

MissionaryHGQ, 2023-03-24:
> There's nothing in Matter.framework itself that puts up such an alertView. Who is putting it up and why?
>
>
> Is this a numeric pairing code (as opposed to a QR code), so you're just getting a short [...]

Yes , this is a 11-digit numeric code produced by  "open paring window", and I found that I can parsed out a normal discriminator (as well as vendorId and productId) from it. The discriminator I got is "15", which seems to be a short discrimunator. 
the alertView is put up by me, because the next commission step requires the wifi ssid and password, and I can't get a correct discriminator from paring code A

bzbarsky-apple, 2023-03-24:
> The discriminator I got is "15", which seems to be a short discrimunator.

15 could be a long discriminator value too.  But a numeric code always has a short discriminator, so you do in fact have a short discriminator.

> because the next commission step requires the wifi ssid and password

Why does it require that if the device is already on-network?

> and I can't get a correct discriminator from paring code A

What is "not correct" about the short discriminator?  The short discriminator will let you discover all devices that match that short discriminator, just like a long discriminator would let you discover all devices that match it.  There might be more things that match the short discriminator, because it has fewer possible values so the chance of collision is higher, but apart from that....

I guess what I don't understand is what you are doing exactly.   You are given an onboarding payload (what you are calling `A` above) for an on-network device.  If you are using the version of `Matter.framework` in 16.2, the simplest thing is to use:
```
    - (BOOL)pairDevice:(uint64_t)deviceID
        onboardingPayload:(NSString *)onboardingPayload
                    error:(NSError * __autoreleasing *)error;
```
and just pass in the `A` that you got.  This will do the discovery, etc.  Then when your `MTRDevicePairingDelegate` is notified `onPairingComplete` without an error, you use:
```
   - (BOOL)commissionDevice:(uint64_t)deviceId
        commissioningParams:(MTRCommissioningParameters *)commissioningParams
                      error:(NSError * __autoreleasing *)error;
```
and just don't specify any network credentials in the `MTRCommissioningParameters`, since the device is already on the network.

At no point in that do you need to parse out discriminators, etc.

MissionaryHGQ, 2023-03-27:
It works for me that "just don't specify any network credentials in the MTRCommissioningParameters",  I can pair the device from a device which is already on the network now. But the other question is that , how to judge "the device is already on the network ", which api can help? I should consider in which situation, I must specify the network  credentials in the MTRCommissioningParameters and which is not.

bzbarsky-apple, 2023-03-27:
@MissionaryHGQ If you use MatterSupport, the device will already be on the network.

MissionaryHGQ, 2023-03-27:
do you mean that the Matter.framework have not api to confirm the device (which  the device paring  from a mannual code.) is a new one  or  already on the work？

bzbarsky-apple, 2023-03-27:
@MissionaryHGQ Once you have set up the PASE session, there is `sessionTransportType` available on the `MTRBaseDevice`, starting in iOS 16.4.

But again: the only current way to commission devices that are not already on the network is to use MatterSupport.framework to bring them onto the network, open a commissioning window, and get a setup payload, and then use Matter.framework to commission them.

MissionaryHGQ, 2023-03-28:
I have studied MatterSupport.framework before for many weeks, but sufferd many unknown error.
such as , "HMErrorDomain error 2" after "commissioned complete successfully",  Xcode throw out an error "device.setup.manager Failed to perform Matter device setup setup: Error Domain=HMErrorDomain Code=2 "(null)) .
and the Matter Extension has not invoked any funtion durning the whole process.
So finally I couldn't even pair successful.
Finally,  I used Matter.frawork and it worked!.
 So is there any demo or document about MatterSupport.framework for the time being?

MissionaryHGQ, 2023-03-29:
> @MissionaryHGQ Once you have set up the PASE session, there is `sessionTransportType` available on the `MTRBaseDevice`, starting in iOS 16.4.
>
> But again: the only current way to commission devices [...]

I am nowing using MatterSupport.framework to pair new device now.  The pair procedure is very smoothly. And from my device log, I see logs such as "received sigmal3 msg"， ”commissioning completed successfully“, "received command to open commissioning window", "commissioning window is now open". 

It seems that the commissioning is successful, but finally my iPhone prompt an error alertView "Unable to Pair, Home couldn't connect to this accessary". And the MatterExtension didn't receive any logs I write in. Is there any special setup about MatterExtension or any other works I should do , can you help me ? I have  confused about it for several months.

bzbarsky-apple, 2023-03-29:
@MissionaryHGQ I don't know what's going on there.  If you get "commissioning window is now open" I would expect you to get a setup payload that you can then use with Matter.framework to commission the device into your fabric.  Are you getting a payload?

MissionaryHGQ, 2023-03-29:
not, I can't get a payload.  "open commissioning window" is automatically invoked after I called the MatterSupported API "MatterAddDeviceRequest()", And I can't get the setup payload because It didn't have any callback during the commissioning.  The application extension class "MatterAddDeviceExtensionRequestHandler" didn't give me any callback too. I didn't know  in which time the commissioning is completed as not receive any callback during the commissioning.

MissionaryHGQ, 2023-03-30:
I am just wondering that whether I have used the application extension "MatterExtension" correctly.
I just add an matter extension target, then config the bunleID and do noting.
is there any special setup about it ?
as the matter extension didn't show any invoke log,  so I wonder about i t.

bzbarsky-apple, 2023-03-30:
@MissionaryHGQ I'm hoping there will be better documentation for MatterSupport in the near future, but in the meantime if you attach your code that uses MatterSupport here I might be able to ask people to look at it and see if anything jumps out at them.

MissionaryHGQ, 2023-03-31:
I have commited  a feedback attached my code to  feedbackassistant.apple.com and the feedback id is "FB12086187",  nobody response to me yet. 
I think it may be the question of communication  between the  MatterSupport.frawework's  application extention with my main application,  but I don't how to use it correctly.

MissionaryHGQ, 2023-03-31:
my code is simple as below:
` func performMatterRequest(
        ecosystemName: String = "Magenta Fabric",
        homeName: String = "My Home", payload: MTRSetupPayload? = nil) async {
            
            let magentaHome = MatterAddDeviceRequest.Home(displayName: homeName)
            let topology = MatterAddDeviceRequest.Topology(
                ecosystemName: ecosystemName,
                homes: [magentaHome])
            let request = MatterAddDeviceRequest(
                topology: topology,
                setupPayload: payload)
            do {
                try await request.perform()
            } catch {
                self.error = error
                logger.error("\(error.localizedDescription)")
                showingAlert = true
            }
        }`

error log:
 [device.setup.manager] [E3523A78-042A-4A85-9746-87F80652EE94] Failed to perform Matter device setup setup: Error Domain=HMErrorDomain Code=23 "(null)"
2023-03-31 18:39:50.803882+0800 LuffyPocTestProject[5577:769393] [app] The operation couldn’t be completed. (HMErrorDomain error 23.)
2023-03-31 18:40:12.283989+0800 LuffyPocTestProject[5577:769402] [xpc.server.proxy] [D62DACD9-FBAF-4E27-9A5B-22CF45B11852] Failed to obtain remote object proxy for performDeviceSetupUsingRequest: Error Domain=NSCocoaErrorDomain Code=4097 "connection to service named com.apple.matter.support.xpc" UserInfo={NSDebugDescription=connection to service named com.apple.matter.support.xpc}
2023-03-31 18:40:12.284221+0800 LuffyPocTestProject[5577:769402] [device.setup.manager] [CDDB17EA-6FCB-4419-B94C-0370DEFA5AE7] Failed to perform Matter device setup setup: Error Domain=NSCocoaErrorDomain Code=4097 "connection to service named com.apple.matter.support.xpc" UserInfo={NSDebugDescription=connection to service named com.apple.matter.support.xpc}
2023-03-31 18:40:12.287937+0800 LuffyPocTestProject[5577:769700] [app] Couldn’t communicate with a helper application.

muralivm, 2023-04-10:
@MissionaryHGQ: Which matter accessory are you using to commission and are you using Apple matter controller for commissioning?

edWin-m, 2023-04-24:
@bzbarsky-apple could you please expand a bit on the commissioning flow you hinted at below? 

> But again: the only current way to commission devices that are not already on the network is to use MatterSupport.framework to bring them onto the network, open a commissioning window, and get a setup payload, and then use Matter.framework to commission them.

We are trying to build our app for production and we just found out that we need to use MatterSupport for commissioning. 
When we used the methods noted [here](https://developer.apple.com/documentation/mattersupport), the device was commissioned successfully, however, like what @MissionaryHGQ  said, there aren't any callbacks returning the payload or the device info in our case.

bzbarsky-apple, 2023-04-24:
@edWin-m @MissionaryHGQ Are you subclassing `MatterAddDeviceExtensionRequestHandler` in your application as described in https://developer.apple.com/documentation/mattersupport/matteradddeviceextensionrequesthandler ?

MissionaryHGQ, 2023-04-25:
> @edWin-m @MissionaryHGQ Are you subclassing `MatterAddDeviceExtensionRequestHandler` in your application as described in https://developer.apple.com/documentation/mattersupport/matteradddeviceextensionrequesthandler ?

sure, I did.

edWin-m, 2023-04-25:
@bzbarsky-apple same here

bzbarsky-apple, 2023-04-25:
@MissionaryHGQ @edWin-m And your subclass's `commissionDevice` is not being called?

edWin-m, 2023-04-25:
@bzbarsky-apple we have a slight issue with logging and breakpoints in `MatterAddDeviceExtensionRequestHandler`. Somehow they are not effective. Thus we can't tell for sure that the 'commissionDevice' is being called.  
However, some changes we made in the subclass's `rooms(in home: MatterAddDeviceRequest.Home?)` were picked up by the UI.  Furthermore,  from the device logs, we can tell that the commissioning window was opened successfully.

edWin-m, 2023-04-25:
@bzbarsky-apple It's being called.

bzbarsky-apple, 2023-04-25:
@edWin-m OK, so if your `commissionDevice` is being called, then it's handled a setup payload that you can use to commission the device into your fabric, no?

edWin-m, 2023-04-26:
@bzbarsky-apple We got the onboarding payload. Now we are working on the commissioning of the devices into our fabric. Thanks for the assistance.

MissionaryHGQ, 2023-04-26:
> 

did your commission with mattersupport.framework work well before iOS16.5?

edWin-m, 2023-04-26:
> did your commission with mattersupport.framework work well before iOS16.5?

@MissionaryHGQ We never tested it on the previous versions.  We just started with iOS16.5.  Is commissioning failing on the previous versions?

MissionaryHGQ, 2023-04-26:
> > did your commission with mattersupport.framework work well before iOS16.5?
> 
> @MissionaryHGQ We never tested it on the previous versions. We just started with iOS16.5. Is commissioning failing on the previous versions?

yes, commission before iOS16.5 is always unstable in my case. you may try and share me your result.

edWin-m, 2023-04-27:
@MissionaryHGQ We have tested this on iOS16.4, and we haven't encountered anything out of the ordinary. Commissioning on both versions appears to have the same success rate.

MissionaryHGQ, 2023-04-27:
> @MissionaryHGQ We have tested this on iOS16.4, and we haven't encountered anything out of the ordinary. Commissioning on both versions appears to have the same success rate.

you may repeat trying to remove and pair device again,  the commision may become unstable.
btw,  which beta version of iOS16.5 are you using?

woody-apple, 2023-06-23:
That said, for Matter Support or Home issues, please file a feedback assistant ticket on Home as follows:

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

closed 2023-06-23 by woody-apple

Nishant0585, 2023-08-05:
@bzbarsky-apple 
Hi 
I have onboarded matter accessory using matterSupport and Commissioned it on iPhone using Matter iOS SDK API's
Now Again I want to pen pairing window to other admin from my app?
no clue how to open it 
trying https://developer.apple.com/documentation/matter/mtrbasedevice/4226882-opencommissioningwindow?changes=_6_7_5
openCommissioningWindow(withDiscriminator:duration:queue:completion:)
But how to call this method

or How to get my paired Accessory , using Matter iOS SDK

bzbarsky-apple, 2023-08-05:
>  But how to call this method

Well, you need an MTRBaseDevice, right?  You presumably have a controller for the relevant fabric and know the accessory's node ID, right?  In that case, https://developer.apple.com/documentation/matter/mtrbasedevice/4120565-devicewithnodeid?changes=_6_7_5&language=objc to create the device (`[MTRBaseDevice deviceWithNodeID:whatever controller:something]`) and then call the method.

Nishant0585, 2023-08-05:
@bzbarsky-apple 
Ok, I'll try this but, to to open CommissioningWindow
we required below params, setupPasscode, discriminator, from where is can fetch ?

func openCommissioningWindow(
    withSetupPasscode setupPasscode: 
    discriminator: 
    duration: 
    queue: 
    completion: 
)

Nishant0585, 2023-08-05:
@bzbarsky-apple 
 got through with this ([MTRBaseDevice deviceWithNodeID:whatever controller:something])
I am getting object of MTRBaseDevice
But when try to openCommissioningWindow with random setupPasscode & discriminator
getting Transaction time out...

Nishant0585, 2023-08-07:
@bzbarsky-apple 
trying to read all attributes of the connected device get error (passing nil in all param of wildcard*)
https://developer.apple.com/documentation/matter/mtrbasedevice/4120567-readattributes?changes=_6_7_5

Error is
Mdns: Resolve failure (KDNSServiceErr_timeout)
chip error 0*000000032: Timeout

bzbarsky-apple, 2023-08-07:
> we required below params, setupPasscode, discriminator, from where is can fetch ?

You get to pick them.  Random generation is fine, as long as it follows the rules for these things in the spec.  `MTRSetupPayload` has a `generateRandomSetupPasscode` class method for doing that for the passcode, since the rules are not trivial.  Discriminator can be any value 0-4095.

>  Mdns: Resolve failure (KDNSServiceErr_timeout)

Did you use the right node id?  What does `dns-sd -B _matter._tcp` show before you try to do the operational resolve?  What node ID are you passing to `deviceWithNodeID:`?

Nishant0585, 2023-08-07:
@bzbarsky-apple 
I am passing same device  which I have used in setup commissioning session

with that node id I am creating MTRBaseDevice

on that trying to call open pairing window 
or read attributes
but it say Mdns: Resolve failure (KDNSServiceErr_timeout)

Nishant0585, 2023-08-07:
12:54:43.100  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
12:54:43.101  Add        3  15 local.               _matter._tcp.        93E7D599F41DC1DB-00000000749E482F
12:54:43.101  Add        2  15 local.               _matter._tcp.        E13D03B5DD7EB17D-0000000000000001

bzbarsky-apple, 2023-08-07:
> 12:54:43.101 Add 3 15 local. _matter._tcp. 93E7D599F41DC1DB-00000000749E482F
> 12:54:43.101 Add 2 15 local. _matter._tcp. E13D03B5DD7EB17D-0000000000000001

OK, and what was the numeric value of the node ID you passed to `openCommissioningWindow` in that exact situation?

At some point here, the right answer is for you to file a feedback assistant ticket, capture the relevant logs on the phone and attach them there, and let me know what the feedback assistant issue id is....  Because I can't just keep guessing at what you are doing, especially if you won't tell me.

Nishant0585, 2023-08-07:
> OK, and what was the numeric value of the node ID you passed to `openCommissioningWindow` in that exact situation?
>
> At some point here, the right answer is for you to file a feedback assistant [...]

@bzbarsky-apple 
NodeId used is 1

Nishant0585, 2023-08-07:
> eedback assistant ticket
@bzbarsky-apple 
Created new Ticket, added code also
https://github.com/project-chip/connectedhomeip/issues/28568

bzbarsky-apple, 2023-08-08:
@Nishant0585 I'll follow up there, but I was talking about a feedback assistant ticket, not a github issue....

qawitherev, 2023-08-21:
> @bzbarsky-apple It's being called.

how do you get the methods in RequestHandler being called, does it have to do anything with `beginRequest(with context: NEExtensionContext)`?

qawitherev, 2023-08-24:
> @edWin-m @MissionaryHGQ I have followed https://developer.apple.com/documentation/mattersupport/matteradddeviceextensionrequesthandler But it seems like didn't work for me. The matter extension cann't be launched. And the subclass's function not be called. Is matter extension the right way to subclass 'MatterAddDeviceExtensionRequestHandler' ?
> 
> It will be appreciate if you could share about your experience about subclassing the RequestHandler. Thank you very much.

@eling22 
Yes. Matter extension is the right way to subclass the request handler. This request handler is launched whenever MatterAddDeviceRequest.perform() is executed. However, since it is an extension (not the app), you cannot see the debug print in Xcode. You have to use [Logger](https://developer.apple.com/documentation/os/logger) and see the debug print in Console app on your Mac.

LucaFagan, 2023-09-13:
@qawitherev i have the same situation: RequestHandles extenstion is not called. Have you found a solution ?

LucaFagan, 2023-09-13:
> > @bzbarsky-apple It's being called.
> 
> @edWin-m how do you get the methods in RequestHandler being called, does it have to do anything with `beginRequest(with context: NEExtensionContext)`?
> 
> nvm, got it, we can't see the log using print as it is an extension, not the app itself 😷

I had the same problem, and solved in this way:

- in your app etitlements set: Matter Allow Setup Payload = YES
- in your extension use: os_log(.default, "your message")   don't forget import OSLog
- use console app to see log messages

Nishant0585, 2023-10-27:
Hi @bzbarsky-apple 
I am using matter support framework 
I want device already on the network, should not display in more option,
MatterAddDeviceRequest.addrequest

how to utilize showDeviceCriteria
what value will be passed when I set showDeviceCriteria  .not(.......) ?

bzbarsky-apple, 2023-10-28:
@Nishant0585 I don't know offhand, sorry....

bzbarsky-apple, 2023-10-30:
@Nishant0585 I have been trying to find the answer to your question, but I realized I am not quite sure what you are asking.  What do you mean by "only those in more option which is not paired(like apple home)"?

huchuankan, 2024-03-14:
Matter Commission  demo video ： https://www.bilibili.com/video/BV1Nx421y7sa/
use mattersupport ， Commission without ssid and pwd