project-chip/connectedhomeip #25779: [1.0.0.2] TC-ADMIN-1.9 get a wrong output

strangermans, 2023-03-22:
### Reproduction steps

1. ./chip-tool pairing onnetwork 1 20202021
2. ./chip-tool pairing open-commissioning-window 1 1 200 1000 3841
3. avahi-browse -rt _matterc._udp
4. ./chip-tool payload generate-manualcode --discriminator 0xF00 --setup-pin-code 20202022 --version 0 --vendor-id 0xFFF1 --product-id 0x8001 --commissioning-mode 0
5. ./chip-tool pairing  code  2  34970212338  --timeout 40  --commissioner-name beta
6. ./chip-tool pairing  code  2  35489460785   --commissioner-name beta

### Bug prevalence

Whenever i do this

### GitHub hash of the SDK that was being used

1.0.0.2

### Platform

other

### Type

Test Improvement

### Anything else?

when i run the TC-CADMIN-1.9, I get a wrong code which called "Incorrect state", but test result expected "Connection aborted", so i want to know the reason.
[1.9.txt](https://github.com/project-chip/connectedhomeip/files/11037479/1.9.txt)

kendallgoto, 2025-07-24:
should be closed by https://github.com/project-chip/matter-test-scripts/issues/459 based on my testing - though the 1s sleep introduced in https://github.com/project-chip/matter-test-scripts/issues/459 has some variability depending on the DUT and may need to be extended in some cases.

kendallgoto, 2025-08-01:
@murda-sigma you can see this was resolved upstream by allowing both error codes instead of sleeping https://github.com/project-chip/connectedhomeip/pull/36546

but this doesn't apply for the v1.4 test cases so would have to be ported down. I think your workaround with the sleep would be acceptable though.