project-chip/connectedhomeip #25784: [Documentation] Miss example on how Matter client establish Matter server cluster information

Henrik9000, 2023-03-22:
### Documentation issues

The Matter Client Example at https://github.com/project-chip/connectedhomeip/tree/master/examples/chip-tool would benefit from information on how to establish Matter server cluster information.

AFAIK, one way to do it is to 
`chip-tool descriptor read parts-list <destination-id> 0`
to find the number of endpoints, and iterate over endpoints using
`chip-tool descriptor read server-list <destination-id> <endpoint-id>`
to obtain a description of each endpoint.

But I am uncertain if there are better ways. I think this approach seems a bit clumsy.

IMHO, the optimal approach would be one command to obtain a complete model of the server, returning an easy-to-parse format.

### Anything else?

A link touching the same subject:
https://community.silabs.com/s/question/0D58Y000093gGqdSAE/how-to-get-the-endpoint-list-of-matter-devices-in-chiptool

bzbarsky-apple, 2023-03-22:
> But I am uncertain if there are better ways.

You could do a wildcard read for all server-list attributes across all endpoints.  So:

```
chip-tool descriptor read-by-id 0x0001 <node-id> 0xFFFF
```

> the optimal approach would be one command to obtain a complete model of the server, returning an easy-to-parse format.

Using tracing for chip-tool might get you the easy-to-parse-format bit, perhaps.

Henrik9000, 2023-03-23:
Thanks @bzbarsky-apple, seems to work fine:
```
chip-tool descriptor read-by-id 0x0001 28 0xFFFF
[1679482564.264340][16594:16594] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
...
[1679482566.848133][16594:16599] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_001D Attribute 0x0000_0001 DataVersion: 7305299
[1679482566.848204][16594:16599] CHIP:TOO:   ServerList: 17 entries
[1679482566.848232][16594:16599] CHIP:TOO:     [1]: 3
[1679482566.848253][16594:16599] CHIP:TOO:     [2]: 29
[1679482566.848273][16594:16599] CHIP:TOO:     [3]: 31
[1679482566.848292][16594:16599] CHIP:TOO:     [4]: 40
[1679482566.848312][16594:16599] CHIP:TOO:     [5]: 42
[1679482566.848336][16594:16599] CHIP:TOO:     [6]: 43
[1679482566.848358][16594:16599] CHIP:TOO:     [7]: 44
[1679482566.848378][16594:16599] CHIP:TOO:     [8]: 47
[1679482566.848398][16594:16599] CHIP:TOO:     [9]: 48
[1679482566.848418][16594:16599] CHIP:TOO:     [10]: 49
[1679482566.848437][16594:16599] CHIP:TOO:     [11]: 50
[1679482566.848456][16594:16599] CHIP:TOO:     [12]: 51
[1679482566.848480][16594:16599] CHIP:TOO:     [13]: 52
[1679482566.848502][16594:16599] CHIP:TOO:     [14]: 53
[1679482566.848522][16594:16599] CHIP:TOO:     [15]: 60
[1679482566.848541][16594:16599] CHIP:TOO:     [16]: 62
[1679482566.848561][16594:16599] CHIP:TOO:     [17]: 63
[1679482566.848720][16594:16599] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_001D Attribute 0x0000_0001 DataVersion: 3866947121
[1679482566.848750][16594:16599] CHIP:TOO:   ServerList: 2 entries
[1679482566.848772][16594:16599] CHIP:TOO:     [1]: 29
[1679482566.848791][16594:16599] CHIP:TOO:     [2]: 1026
[1679482566.848915][16594:16599] CHIP:TOO: Endpoint: 2 Cluster: 0x0000_001D Attribute 0x0000_0001 DataVersion: 1622942900
[1679482566.848941][16594:16599] CHIP:TOO:   ServerList: 2 entries
[1679482566.848963][16594:16599] CHIP:TOO:     [1]: 29
[1679482566.848983][16594:16599] CHIP:TOO:     [2]: 1029
[1679482566.849080][16594:16599] CHIP:TOO: Endpoint: 3 Cluster: 0x0000_001D Attribute 0x0000_0001 DataVersion: 1788893351
[1679482566.849107][16594:16599] CHIP:TOO:   ServerList: 2 entries
[1679482566.849128][16594:16599] CHIP:TOO:     [1]: 29
[1679482566.849148][16594:16599] CHIP:TOO:     [2]: 1027
```
With tracing enabled like
```chip-tool descriptor read server-list 28 0xFFFF --trace_file mytrace.txt```
I'm getting messages in the form of 
```
[1679482752.721212]   SecureMsg.PreparedMsgSent
    json        {"destination": "UDP:[fd69:f4b9:c81d:1:f923:d4e9:e21b:8c38%wlan0]:5540", "payload_size": 26, "payload_hex": "04000000e28ca0009104e9a06cd524870310321b0000dfd48b62", "buffer_ptr": 281472896394544}
```
I now need to figure out how to decipher those.

Henrik9000, 2023-03-24:
Okay, so the TLVReader method would be come in handy:
https://github.com/project-chip/connectedhomeip/blob/master/src/lib/core/TLVReader.h

Henrik9000, 2023-03-24:
I will close the issue now. However, I obviously believe documentation should be improved on this point.

closed 2023-03-24 by Henrik9000

nmtoan91, 2023-12-11:
@bzbarsky-apple ,
Is there any Python library for decoding the TLV-encoded string?

bzbarsky-apple, 2023-12-11:
@nmtoan91 You mean apart from the Matter SDK?  The TLV encoding is defined by the Matter spec....