project-chip/connectedhomeip #25792: [ESP32] All-Clusters-App example does not build

TestUserForIOT, 2023-03-23:
### Reproduction steps

Dear Ladies and Gentleman,  

I am pretty new to the Matter project. To gain some experience with Matter, I tried to build the All-Clusters-App example for the ESP32C3 Chip to test the Solution with available Hardware.  

When I try to build the example within the esp-idf environment, I get the error that the compiler is checking the version of the zap-cli framework by calling “zap-cli --version”. The compiler than stops with the Error message “No such option: --version Did you mean --verbose”. And in fact, there is no such option for the zap-cli framework.  

I added the complete compiler output below. Thanks for your help.
Best wishes
Jannik  

# My System setup
Connectedhomeip Branch: Master Release: #[a6dc62c](https://github.com/project-chip/connectedhomeip/commit/a6dc62c6582de394d3c628f4386defd71b09bee1)
Ubuntu Version: 22.04 LTS
Python Version: 3.10.6
Pip Version: 23.0.1
Idf Version: 4.4, py3.10 environment
Ninja Version: 1.10.1
ZAP CLI Version: 0.10.0

# Compiler output
`~/connectedhomeip/examples/all-clusters-app/esp32$ idf.py build`
`Executing action: all (aliases: build)`
`Running cmake in directory ~/connectedhomeip/examples/all-clusters-app/esp32/build`
`Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32c3 -DCCACHE_ENABLE=0` `~/connectedhomeip/examples/all-clusters-app/esp32"...`
`-- Found Git: /usr/bin/git (found version "2.34.1")`
`-- The C compiler identification is GNU 8.4.0`
`-- The CXX compiler identification is GNU 8.4.0`
`-- The ASM compiler identification is GNU`
`-- Found assembler: ~/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc`
`-- Detecting C compiler ABI info`
`-- Detecting C compiler ABI info - done`
`-- Check for working C compiler: ~/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc - skipped`
`-- Detecting C compile features`
`-- Detecting C compile features - done`
`-- Detecting CXX compiler ABI info`
`-- Detecting CXX compiler ABI info - done`
`-- Check for working CXX compiler: ~/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf`-`g++ - skipped`
`-- Detecting CXX compile features`
`-- Detecting CXX compile features - done`
`-- Building ESP-IDF components for target esp32c3`
`Skipping optional dependency: espressif/mdns`
`Processing 2 dependencies:`
`[1/2] espressif/led_strip (1.1.0-alpha)`
`[2/2] idf (4.4.3)`
`-- Found Python3: ~/.espressif/python_env/idf4.4_py3.10_env/bin/python3.10 (found version "3.10.6") found components: Interpreter`
`Skipping optional dependency: espressif/mdns`
`-- Project sdkconfig file ~/connectedhomeip/examples/all-clusters-app/esp32/sdkconfig`
`Loading defaults file ~/connectedhomeip/examples/all-clusters-app/esp32/sdkconfig.defaults...`
`-- Found Python3: ~/.espressif/python_env/idf4.4_py3.10_env/bin/python3.10 (found version "3.10.6") found components: Interpreter`
`-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS`
`-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success`
`-- App "chip-all-clusters-app" version: v1.0`
`-- Adding linker script ~/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/esp_system/ld/memory.ld`
`-- Adding linker script ~/esp-idf/components/esp_system/ld/esp32c3/sections.ld.in`
`-- Adding linker script ~/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.ld`
`-- Adding linker script ~/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld`
`-- Adding linker script ~/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld`
`-- Adding linker script ~/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld`
`-- Adding linker script ~/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.version.ld`
`-- Adding linker script ~/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib-time.ld`
`-- Adding linker script ~/esp-idf/components/esp_rom/esp32c3/ld/esp32c3.rom.eco3.ld`
`-- Adding linker script ~/esp-idf/components/soc/esp32c3/ld/esp32c3.peripherals.ld`
`-- Components: QRCode app_trace app_update asio bootloader bootloader_support bt cbor chip cmock coap console cxx driver efuse esp-tls esp32_mbedtls esp32c3 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump espressif__led_strip esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table protobuf-c protocomm pthread riscv sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb unity usb vfs wear_levelling wifi_provisioning wpa_supplicant`
`-- Component paths: ~/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/examples/common/QRCode ~/esp-idf/components/app_trace ~/esp-idf/components/app_update ~/esp-idf/components/asio ~/esp-idf/components/bootloader ~/esp-idf/components/bootloader_support ~/esp-idf/components/bt ~/esp-idf/components/cbor ``~/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/config/esp32/components/chip ~/esp-idf/components/cmock ~/esp-idf/components/coap ~/esp-idf/components/console ~/esp-idf/components/cxx ~/esp-idf/components/driver ~/esp-idf/components/efuse ~/esp-idf/components/esp-tls ~/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/config/esp32/components/esp32_mbedtls ~/esp-idf/components/esp32c3 ~/esp-idf/components/esp_adc_cal ~/esp-idf/components/esp_common ~/esp-idf/components/esp_eth ~/esp-idf/components/esp_event ~/esp-idf/components/esp_gdbstub ~/esp-idf/components/esp_hid ~/esp-idf/components/esp_http_client ~/esp-idf/components/esp_http_server ~/esp-idf/components/esp_https_ota ~/esp-idf/components/esp_https_server ~/esp-idf/components/esp_hw_support ~/esp-idf/components/esp_ipc ~/esp-idf/components/esp_lcd ~/esp-idf/components/esp_local_ctrl ~/esp-idf/components/esp_netif ~/esp-idf/components/esp_phy ~/esp-idf/components/esp_pm ~/esp-idf/components/esp_ringbuf ~/esp-idf/components/esp_rom ~/esp-idf/components/esp_serial_slave_link ~/esp-idf/components/esp_system ~/esp-idf/components/esp_timer ~/esp-idf/components/esp_websocket_client ~/esp-idf/components/esp_wifi ``~/esp-idf/components/espcoredump ~/connectedhomeip/examples/all-clusters-app/esp32/managed_components/espressif__led_strip ~/esp-idf/components/esptool_py ~/esp-idf/components/expat ~/esp-idf/components/fatfs ~/esp-idf/components/freemodbus ~/esp-idf/components/freertos ~/esp-idf/components/hal ~/esp-idf/components/heap ~/esp-idf/components/idf_test ~/esp-idf/components/ieee802154 ~/esp-idf/components/jsmn ~/esp-idf/components/json ~/esp-idf/components/libsodium ~/esp-idf/components/log ~/esp-idf/components/lwip ~/connectedhomeip/examples/all-clusters-app/esp32/main ~/esp-idf/components/mbedtls ~/esp-idf/components/mdns ~/esp-idf/components/mqtt ~/esp-idf/components/newlib ~/esp-idf/components/nghttp ~/esp-idf/components/nvs_flash ~/esp-idf/components/openssl ~/esp-idf/components/openthread ~/esp-idf/components/partition_table ~/esp-idf/components/protobuf-c ~/esp-idf/components/protocomm ~/esp-idf/components/pthread ~/esp-idf/components/riscv ~/esp-idf/components/sdmmc ~/esp-idf/components/soc ~/esp-idf/components/spi_flash ~/esp-idf/components/spiffs ~/esp-idf/components/tcp_transport ~/esp-idf/components/tcpip_adapter ~/esp-idf/components/tinyusb ~/esp-idf/components/unity ~/esp-idf/components/usb ~/esp-idf/components/vfs ~/esp-idf/components/wear_levelling ~/esp-idf/components/wifi_provisioning ~/esp-idf/components/wpa_supplicant`
`-- Configuring done`
`-- Generating done`
`-- Build files have been written to: ~/connectedhomeip/examples/all-clusters-app/esp32/build`
`Running ninja in directory ~/connectedhomeip/examples/all-clusters-app/esp32/build`
[... 34 lines cut ...]
`subprocess.CalledProcessError: Command '['zap-cli', '--version']' returned non-zero exit status 2.`
`[21/1231] Generating ../../gen/app-codegen/cpp-app/app/PluginApplicationCallbacks.h, ../../gen/app-codegen/cpp-app/app/callback-stub.cpp`
`2023-03-23 05:39:21 INFO    Parsing idl from ~/connectedhomeip/examples/all-clusters-app/all-clusters-common/all-clusters-app.matter`
`2023-03-23 05:39:22 INFO    Running code generator cpp-app`
[... cut ...]
`2023-03-23 05:39:22 INFO    Writing new data to: ~/connectedhomeip/examples/all-clusters-app/esp32/build/gen/app-codegen/cpp-app/app/callback-stub.cpp`
`2023-03-23 05:39:22 INFO    Done`
`[22/1231] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/bignum.c.obj`
`ninja: build stopped: subcommand failed.`
`ninja failed with exit code 1`
`~/connectedhomeip/examples/all-clusters-app/esp32$`

### Platform

esp32

### Platform Version(s)

All-cluster-app Version 1.0

### Type

Unit tested

TestUserForIOT, 2023-03-23:
When I type in zap-cli, I get the following output:

`~/connectedhomeip/examples/all-clusters-app/esp32$ zap-cli`

`Usage: zap-cli [OPTIONS] COMMAND [ARGS]...`

 ` ZAP CLI v0.10.0 - A simple commandline tool for OWASP ZAP.`

`Options:`
`  --boring            Remove color from console output.`
`  -v, --verbose       Add more verbose debugging output.`
`  --zap-path TEXT     Path to the ZAP daemon. Defaults to /zap or the value of
                      the environment variable ZAP_PATH.`
 ` -p, --port INTEGER  Port of the ZAP proxy. Defaults to 8090 or the value of
                      the environment variable ZAP_PORT.`
 ` --zap-url TEXT      The URL of the ZAP proxy. Defaults to http://127.0.0.1
                      or the value of the environment variable ZAP_URL.`
`  --api-key TEXT      The API key for using the ZAP API if required. Defaults
                      to the value of the environment variable ZAP_API_KEY.`
 ` --log-path TEXT     Path to the directory in which to save the ZAP output
                      log file. Defaults to the value of the environment
                      variable ZAP_LOG_PATH and uses the value of --zap-path
                      if it is not set.`
 ` --help              Show this message and exit.`

`Commands:`
`  active-scan  Run an Active Scan.`
 ` ajax-spider  Run the AJAX Spider against a URL.`
  `alerts       Show alerts at the given alert level.`
 ` context      Manage contexts for the current session.`
 ` exclude      Exclude a pattern from all scanners.`
 ` open-url     Open a URL using the ZAP proxy.`
`  policies     Enable or list a set of policies.`
 ` quick-scan   Run a quick scan.`
 ` report       Generate XML, MD or HTML report.`
 ` scanners     Enable, disable, or list a set of scanners.`
`  scripts      Manage scripts.`
`  session      Manage sessions.`
`  shutdown     Shutdown the ZAP daemon.`
`  spider       Run the spider against a URL.`
 ` start        Start the ZAP daemon.`
`  status       Check if ZAP is running.`

bzbarsky-apple, 2023-03-23:
> ZAP CLI v0.10.0 - A simple commandline tool for OWASP ZAP.

@TestUserForIOT That is not the right "ZAP CLI".  It has nothing to do with the ZAP Matter uses.

If you have bootstrapped, and followed the build instructions regarding `activate.sh`, the right zap-cli should be in your PATH if you are using an x86-compatible chip for the host OS for the compile.

If you are on an ARM processor, you need to use ZAP from source; there is no zap-cli prebuilt ARM binary yet.