project-chip/connectedhomeip #25793: [TC-WAKEONLAN-4.1]  Mac Address returned is Empty for wake-on-lan-mac-address attribute.

Rajashreekalmane, 2023-03-23:
**Problem**
When running [TC_MC_4.1], Step 1, the Mac Address returned is Empty for wake-on-lan-mac-address attribute for the wakeonlan  cluster.
Expect to see the mac address of the device, in this case, the Linux device running chip-tv-app.

[image]

**Command used to build the TV-app**
`built the tv-app in linux using scripts/examples/gn_build_example.sh      examples/tv-app/linux/      examples/tv-app/linux/out/tv-app   chip_inet_config_enable_ipv4=false  chip_enable_wake_on_lan=true   `  

**GitHub hash of the SDK that was being used**
https://github.com/project-chip/connectedhomeip/commit/2767b518fe59769720e848a9971f34838770da59 - Master

**PFA Log below**
[image]

[TC-WAKEONLAN-4.1.txt](https://github.com/project-chip/connectedhomeip/files/11047681/TC-WAKEONLAN-4.1.txt)

bzbarsky-apple, 2023-03-23:
The Linux version of `WakeOnLanManager::HandleGetMacAddress` is fairly broken (e.g. encodes char strings, not byte strings), but apart from that it seems to assume that `/sys/class/net/eth0/address` will exist, which it might well not on the hardware you are running on... and it has no error handling around it not existing.

chrisdecenzo, 2023-03-27:
We can fix this to work on the RPI dummy linux tv-app DUT but each real device will likely need to tweak the implementation to use the correct interface.

bzbarsky-apple, 2023-03-27:
Sounds like the relevant part of the implementation should be in some sort of app-implemented provider instead of being hardcoded.....

cjandhyala, 2023-03-27:
@chrisdecenzo as long as we have it fixed on RPI in the tv-app, QA can verify this TC on reference platform. For actual cert testing, since this is a manual test, we"ll have the ATL verify the actual MAC address.

PR #25884 (2023-04-01): Address WoL issue 25793
Fixes #25793 

Read Platform's default interface MAC address in tv-info dummy implementation
This can be changed when ported to other hardware
closed 2023-04-01 by andy31415