project-chip/connectedhomeip #25800: we wonder if we can just make android or iOS app act as an OTA provider?

MrLiuYunPing, 2023-03-23:
### Feature description

...
As far as i know, when the firmware of the matter device need to be updated, the only way is to query from the other devices which had implemented the OTASoftwareUpdateRequestor Cluster (acted as a Matter hub). Is there any way to OTA the device without another device? Can it be possible if we can only use our iOS or android app to do this?

### Platform

android, darwin

bzbarsky-apple, 2023-03-23:
Yes, this should be doable.  On iOS, Matter.framework has facilities for specifying your OTA provider delegate and whatnot, for example.

closed 2023-03-23 by bzbarsky-apple

MrLiuYunPing, 2023-05-26:
Yes， thanks! It's a good job! But, android not supported. I have done in android as iOS, but when I use controller to act as a node to provider the OTA and set OTA delegate, in the OTAProvider SetDelegate method, emberAfFindClusterServerEndpointIndex always return Invalid endpoint. So, I don't know where is wrong? Can you help me?

bzbarsky-apple, 2023-05-26:
Sounds like the Android bits you are using need to hook up actually being used as an OTA provider.  Someone who understand that Android stuff would need to do that.

yunhanw-google, 2023-07-28:
related to https://github.com/project-chip/connectedhomeip/issues/26868