project-chip/connectedhomeip #25833: [BUG] Failed to establish CASE connection with ESP32 M5

yufengwangca, 2023-03-25:
### Reproduction steps

I (1023002) chip[SC]: Received Sigma1 msg
I (1023012) chip[SC]: CASE matched destination ID: fabricIndex 1, NodeID 0x000000001C69524A
E (1023892) chip[DL]: Failed to post event to CHIP Platform event queue
E (1023892) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
I (1023912) chip[EM]: <<< [E:32197r S:0 M:15571910 (Ack:390702)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:31 (SecureChannel:CASE_Sigma2)
E (1023912) chip[DL]: Failed to post event to CHIP Platform event queue
I (1023922) chip[IN]: (U) Sending msg 15571910 to IP address 'UDP:[FE80::F9B2:E9D:B37F:FA18%st1]:35508'
E (1023922) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
I (1023942) chip[SC]: Sent Sigma2 msg
E (1023952) chip[DL]: Failed to post event to CHIP Platform event queue
E (1023952) chip[DL]: Long dispatch time: 993 ms, for event type 3
E (1023952) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
I (1024102) chip[EM]: >>> [E:32197r S:0 M:390703 (Ack:15571910)] (U) Msg RX from 0:452D86714E61B06B [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1024112) chip[EM]: >>> [E:32197r S:0 M:390704 (Ack:15571910)] (U) Msg RX from 0:452D86714E61B06B [0000] --- Type 0000:32 (SecureChannel:CASE_Sigma3)
I (1024132) chip[EM]: <<< [E:32197r S:0 M:15571911 (Ack:390704)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1024142) chip[IN]: (U) Sending msg 15571911 to IP address 'UDP:[FE80::F9B2:E9D:B37F:FA18%st1]:35508'
I (1024152) chip[SC]: Received Sigma3 msg
E (1024162) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
E (1025252) chip[DL]: Failed to post event to CHIP Platform event queue
E (1025252) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (1025272) chip[DL]: Failed to post event to CHIP Platform event queue
E (1025272) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (1025392) chip[DL]: Failed to post event to CHIP Platform event queue
E (1025392) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (1025412) chip[DL]: Failed to post event to CHIP Platform event queue
E (1025412) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (1025432) chip[DL]: Failed to post event to CHIP Platform event queue
E (1025432) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (1025452) chip[DL]: Failed to post event to CHIP Platform event queue
E (1025452) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (1025472) chip[DL]: Failed to post event to CHIP Platform event queue
E (1025472) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (1025492) chip[DL]: Failed to post event to CHIP Platform event queue
E (1025492) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (1025952) chip[DL]: Failed to post event to CHIP Platform event queue
E (1025952) chip[DL]: Failed to schedule work: 1000000
E (1025952) chip[DL]: Long dispatch time: 1787 ms, for event type 2
I (1058912) chip[EM]: >>> [E:32198r S:0 M:390705] (U) Msg RX from 0:11908B34A242E68D [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (1058922) chip[EM]: <<< [E:32198r S:0 M:15571912 (Ack:390705)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1058932) chip[IN]: (U) Sending msg 15571912 to IP address 'UDP:[FE80::F9B2:E9D:B37F:FA18%st1]:35508'
I (1065502) chip[EM]: >>> [E:32199r S:0 M:390706] (U) Msg RX from 0:3DD5AA3CEF6E61C8 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (1065512) chip[EM]: <<< [E:32199r S:0 M:15571913 (Ack:390706)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1065522) chip[IN]: (U) Sending msg 15571913 to IP address 'UDP:[FE80::F9B2:E9D:B37F:FA18%st1]:35508'
I (1071982) chip[EM]: >>> [E:32200r S:0 M:390707] (U) Msg RX from 0:8C93D8483FDF0215 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (1071982) chip[EM]: <<< [E:32200r S:0 M:15571914 (Ack:390707)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1072002) chip[IN]: (U) Sending msg 15571914 to IP address 'UDP:[FE80::F9B2:E9D:B37F:FA18%st1]:35508'
I (1078572) chip[EM]: >>> [E:32201r S:0 M:390708] (U) Msg RX from 0:491D33DCB7FBB006 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (1078582) chip[EM]: <<< [E:32201r S:0 M:15571915 (Ack:390708)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1078592) chip[IN]: (U) Sending msg 15571915 to IP address 'UDP:[FE80::F9B2:E9D:B37F:FA18%st1]:35508'
I (1085052) chip[EM]: >>> [E:32202r S:0 M:390709] (U) Msg RX from 0:9DB9BF1514C3F06D [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (1085062) chip[EM]: <<< [E:32202r S:0 M:15571916 (Ack:390709)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1085072) chip[IN]: (U) Sending msg 15571916 to IP address 'UDP:[FE80::F9B2:E9D:B37F:FA18%st1]:35508'
I (1091682) chip[EM]: >>> [E:32203r S:0 M:390710] (U) Msg RX from 0:3DD3D3B9E1E60F00 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (1091692) chip[EM]: <<< [E:32203r S:0 M:15571917 (Ack:390710)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1091702) chip[IN]: (U) Sending msg 15571917 to IP address 'UDP:[FE80::F9B2:E9D:B37F:FA18%st1]:35508'
I (1098252) chip[EM]: >>> [E:32204r S:0 M:390711] (U) Msg RX from 0:B5E44B700D24B711 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (1098262) chip[EM]: <<< [E:32204r S:0 M:15571918 (Ack:390711)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1098272) chip[IN]: (U) Sending msg 15571918 to IP address 'UDP:[FE80::F9B2:E9D:B37F:FA18%st1]:35508'
I (1112372) chip[EM]: >>> [E:32205r S:0 M:390712] (U) Msg RX from 0:2459157E7F35951E [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (1112382) chip[EM]: <<< [E:32205r S:0 M:15571919 (Ack:390712)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1112392) chip[IN]: (U) Sending msg 15571919 to IP address 'UDP:[FE80::F9B2:E9D:B37F:FA18%st1]:35508'
I (1118992) chip[EM]: >>> [E:32206r S:0 M:390713] (U) Msg RX from 0:BB08EC36374C4200 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (1119002) chip[EM]: <<< [E:32206r S:0 M:15571920 (Ack:390713)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1119012) chip[IN]: (U) Sending msg 15571920 to IP address 'UDP:[FE80::F9B2:E9D:B37F:FA18%st1]:35508'
I (1125702) chip[EM]: >>> [E:32207r S:0 M:390714] (U) Msg RX from 0:65DF370C474A0F08 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (1125712) chip[EM]: <<< [E:32207r S:0 M:15571921 (Ack:390714)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1125722) chip[IN]: (U) Sending msg 15571921 to IP address 'UDP:[FE80::F9B2:E9D:B37F:FA18%st1]:35508'
I (1132242) chip[EM]: >>> [E:32208r S:0 M:390715] (U) Msg RX from 0:A69F6CA9E6ED6CB7 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
[... 29 lines cut ...]
I (1257162) chip[EM]: >>> [E:32218r S:0 M:390725] (U) Msg RX from 0:C399BF9FE22C1104 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (1257162) chip[EM]: <<< [E:32218r S:0 M:15571932 (Ack:390725)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1257182) chip[IN]: (U) Sending msg 15571932 to IP address 'UDP:[FE80::F9B2:E9D:B37F:FA18%st1]:35508'
I (1263722) chip[EM]: >>> [E:32219r S:0 M:390726] (U) Msg RX from 0:5962A37B5BBC51DE [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (1263732) chip[EM]: <<< [E:32219r S:0 M:15571933 (Ack:390726)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1263742) chip[IN]: (U) Sending msg 15571933 to IP address 'UDP:[FE80::F9B2:E9D:B37F:FA18%st1]:35508'
I (1270472) chip[EM]: >>> [E:32220r S:0 M:390727] (U) Msg RX from 0:DCF25512AA0E5F86 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (1270482) chip[EM]: <<< [E:32220r S:0 M:15571934 (Ack:390727)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1270492) chip[IN]: (U) Sending msg 15571934 to IP address 'UDP:[FE80::F9B2:E9D:B37F:FA18%st1]:35508'

### Bug prevalence

almost 100%

### GitHub hash of the SDK that was being used

c6cdc9e79ac22385cddb90bc260e2386993d4600

### Platform

esp32

### Anything else?

I don't have tcpdump utility either in my pixel phone or esp32 device to capture pcap log,  from the log, it looks the device failed to queue certain sigma message to the outgoing buffer

yufengwangca, 2023-03-25:
[esp32_case_fail.log](https://github.com/project-chip/connectedhomeip/files/11069694/esp32_case_fail.log)

wqx6, 2023-03-27:
The reason for the event queue overflow should be same as #23180. Increasing event queue size can help and the packet buffer pool will not be used up since #25679 has been merged.

bzbarsky-apple, 2023-03-27:
I thought https://github.com/project-chip/connectedhomeip/pull/23957 was supposed to have fixed https://github.com/project-chip/connectedhomeip/issues/23180

But it looks like we are not actually using those filters to filter mdns traffic?  https://github.com/project-chip/connectedhomeip/pull/23957 mentions a follow-up PR for esp32, but I guess that never happened?  @tcarmelveilleux

tcarmelveilleux, 2023-04-03:
@bzbarsky-apple

> I thought https://github.com/project-chip/connectedhomeip/pull/23957 was supposed to have fixed https://github.com/project-chip/connectedhomeip/issues/23180
>
> But it looks like we are not actually [...]

Yes, it never got enabled in the repos, since I ran out of time to integrate prior to having some of my time redirected to a different project that leaves me less time to do SDK business.