project-chip/connectedhomeip #25836: [Feature] Connect with the Real devices

Agatha2333, 2023-03-26:
### Feature description

...
Currently, I am unable to use the chip-tool with my eve matter window sensor. But I could use chip-tool to connect and control with the lighting-app running on Linux. I wonder if the chip-tool could work with the real-device.

Thank you for your consideration, and please let me know if you need any additional information from me.

### Platform

all

bzbarsky-apple, 2023-03-27:
> Currently, I am unable to use the chip-tool with my eve matter window sensor.

Can you please attach (not paste) a log showing the failure?  chip-tool should work fine with "real devices"....

bzbarsky-apple, 2023-03-31:
> run the command chip-tool pairing code 0x654321 MT:1214-605-2432

That's not right; the numeric code is not the QR code, and "MT:..." indicates QR code.  You'd want:
```
chip-tool pairing code 0x654321 12146052432
```

But this will only work if the device is already on the IP network.  Similar for `pairing code ... MT: U1ZA3QMV17N79A5.U10`: that's failing for you because the device is not discoverable on the IP network.

You probably want to be using `chip-tool pairing code-wifi` or `chip-tool pairing code-thread` depending on whether you're trying to place the device on a wifi or thread network.

bzbarsky-apple, 2023-04-03:
Both of those are failing to verify device attestation for the device.  Code 101 for device attestation means `kPaaNotFound`.

Which means that your chip-tool did not know about the product attestation authority that signed the device attestation for this device.

By default, chip-tool only uses a test PAA.  If you want to allow other PAAs, you need to decide which ones to trust and put the relevant certificates into a directory that you then pass to `--paa-trust-store-path` when doing `pairing code`.  Or if you want to just trust them all you can use `--paa-trust-store-path credentials/development/paa-root-certs` in the Matter SDK tree.

bzbarsky-apple, 2023-04-14:
@Agatha2333 you should be able to get PAAs from the DCL; see https://github.com/zigbee-alliance/distributed-compliance-ledger/blob/master/docs/quickStartGuide.adoc perhaps?

For the specific one here, you would have to dig through payload logs to see what the public key involved was...

bzbarsky-apple, 2023-04-17:
They may not operate their own PAA.  They might be operating a PAI under some existing PAA.

bzbarsky-apple, 2023-04-17:
Or more to the point, what is the Issuer field of the PAI certificate it's sending?