project-chip/connectedhomeip #25841: Matter device cannot be group controlled

Akihiro-gif, 2023-03-27:
### Reproduction steps

I'm trying to do group control on matter device.
I have tested the connectedhomeip/examples/lighting-app/esp32 example using the CHIPTool v1.0-branch.
I implemented the code below.

1. Add Group permission to ACL using writeAttributeACLWithValue.
2. Set the group key using keySetWriteWithParams.
3. Add a map using writeAttributeGroupKeyMapWithValue.
4. Added to a group using addGroupWithParams.

But I can't do group control.
I would appreciate it if someone could show me how to implement group control in CHIPTool.

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

58cd5fb

### Platform

darwin

bzbarsky-apple, 2023-03-27:
@Akihiro-gif Are you using the Darwin CHIPTool app?  That does not support group control, and neither does the `Matter.framework` it's using.

Akihiro-gif, 2023-03-28:
Yes, I am trying to implement a group control function in the Darwin CHIPTool app. Doesn't it also support matter.framework?

bzbarsky-apple, 2023-03-28:
@Akihiro-gif Darwin CHIPTool uses Matter.framework.  Matter.framework currently does not support group communication.

Akihiro-gif, 2023-03-28:
Thank you. Does that mean that you still can't control multiple devices using Matter.framework as it is?

bzbarsky-apple, 2023-03-28:
You can control multiple devices, but not using group messages.

Akihiro-gif, 2023-03-29:
Thank you.
I thought it could be done by Matter.framework,because it seems to be controlled by the command line like # 20979.
When will group control become possible?

bzbarsky-apple, 2023-03-29:
> because it seems to be controlled by the command line like # 20979.

Nothing in that issue involves Matter.framework...

>  When will group control become possible?

@Akihiro-gif When someone implements it...  There is no concrete time line for that right now, and no one is working on it that I'm aware of.  There was work in progress in https://github.com/project-chip/connectedhomeip/pull/18375 that is not being actively pursued right now

Akihiro-gif, 2023-03-30:
Thank you.
I'll wait until it's implemented.
Is there any other way to implement automation between devices besides group control? (For example, turn LightApp on when you unlock LockApp, etc.)
If so, please let me know. Is this possible with the Scene and Binding clusters?

bzbarsky-apple, 2023-03-30:
> Is there any other way to implement automation between devices besides group control?

Having something track the automation and do unicast communication with the relevant devices?

> when you unlock LockApp, etc.

This can't be done with groups on its own, right?  Either you're the one unlocking the lock, and that cannot be done via group communication per spec (because a door lock device is not allowed to implement the Groups cluster), or someone else is unlocking the lock and then you have to observe the unlock and communicate with the light(s).

Akihiro-gif, 2023-03-31:
Yes, I understand. Thank you. I'll try to find another way.

uascent, 2023-04-15:
Matter Support Framework is unstable, also run into some unknow error,  no more logs, very frustration, can't do anything.

uascent, 2023-04-15:
Special under version < IOS 16.4, HomKit Error Code: 2 , and 23(OperationCancel), anybody know the reason?

woody-apple, 2023-06-23:
@uascent:

> Special under version < IOS 16.4, HomKit Error Code: 2 , and 23(OperationCancel), anybody know the reason?

For Matter Support or Home issues, please file a feedback assistant ticket on Home as follows:

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

closed 2023-06-23 by woody-apple

huchuankan, 2024-03-14:
Matter Commission  demo video ： https://www.bilibili.com/video/BV1Nx421y7sa/