project-chip/connectedhomeip #25858: [BUG] [Android] [CHIPTool] crash after scanning QR code

mmt-10, 2023-03-28:
### Reproduction steps

1. start CHIPTool on Android
2. tap `PROVISION CHIP DEVICE WITH WI-FI`
3. scan QR code
  Then, CHIPTool app crashed and CHIPTool is minimized (may be background).
  CHIPTool does not show SSID and Password form screen.
  CHIPTool switchs to foreground and retry above 2 and 3.
  Then, it can input SSID and Password.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit def9802275c65ce678217a55574ba70e1367d1a4 (HEAD -> master, origin/master, origin/HEAD)

### Platform

android

### Anything else?

Andorid : Pixel 7 pro
cluster : ESP32-C3-devKitM-1
cluster app : lighting-app (v1.0.0.2)

lazeintech, 2023-06-20:
Same problem here with either Scan Qr or Provisioning Wifi

PoltoS, 2023-11-13:
Have you solved it? Scanning itself works, but both WiFi and Thread provisioning lead to a crash

yunhanw-google, 2024-01-18:
this should be resolved by https://github.com/project-chip/connectedhomeip/pull/25958
please retest it using the latest code, I would close it, feel free to reopen or create a new one if you still see this crash regarding QR code action frame with crash dump.

thanks