project-chip/connectedhomeip #25874: [BUG] [ESP32C3][lock-app] kernel panic

mmt-10, 2023-03-29:
### Reproduction steps

1. cd ${MATTER_SDK}
2. source scripts/activate.sh
3. cd ${IDF_PATH}
4. source export.sh
5. cd ${MATTER_SDK}/examples/lock-app/esp32
6. idf.py set-target esp32c3
7. idf.py build
8. idf.py erase-flash flash
9. idf.py monitor
 then, kernel panic occurs.
[esp32c3.log](https://github.com/project-chip/connectedhomeip/files/11097014/esp32c3.log)

 

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit def9802275c65ce678217a55574ba70e1367d1a4 (HEAD -> master, origin/master, origin/HEAD)

### Platform

esp32

### Anything else?

## Environment
Host OS : Ubuntu 22.04
ESP-IDF: v4.4.4
Target Board : ESP32-C3-devKitM-1

bzbarsky-apple, 2023-03-29:
This is the key part from the log:
```
E (1466) chip[DL]: Chip stack locking error at './third_party/connectedhomeip/src/app/util/attribute-storage.cpp:568'. Code is unsafe/racy
```
So something is trying to read or write attributes while not holding the stack lock.  What is the actual stack to the assert?  That would tell you immediately what the erroneous callsite is @mmt-10 

@dhrishi @shubhamdp

bzbarsky-apple, 2023-03-29:
Got esp32 stuff to build; stack looks like this in `idf.py monitor`:
```
0x400e89e6: chip::app::Clusters::DoorLock::Attributes::LockState::Get(unsigned short, chip::app::DataModel::Nullable<chip::app::Clusters::DoorLock::DlLockState>&) at /Users/bzbarsky/connectedhomeip/examples/lock-app/esp32/third_party/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/attributes/Accessors.cpp:11466

0x400eeb88: BoltLockManager::InitLockState() at /Users/bzbarsky/connectedhomeip/examples/lock-app/esp32/third_party/connectedhomeip/examples/platform/esp32/lock/BoltLockManager.cpp:765

0x400e820c: AppTask::Init() at /Users/bzbarsky/connectedhomeip/examples/lock-app/esp32/main/AppTask.cpp:89
```

Looking at `examples/platform/esp32/lock/BoltLockManager.cpp `, `InitLockState` _definitely_ expects to be called with the Matter lock held.

dhrishi, 2023-03-30:
@jadhavrohit924 Please check

PR #25895 (2023-04-03): [ESP32] Fixed lock app crash
Problem:

- Fixes #25874 and #24913.

Changes:

- See above.

Testing:

- Tested commissioning and cluster control.
closed 2023-04-03 by andy31415