project-chip/connectedhomeip #25878: [BUG] Error in Commissioning Process (ESP32 - Chip Tool CLI)

leandropg, 2023-03-29:
### Reproduction steps

I am working to commission a **ESP32** using the lighting app example over Wifi ([ESP32 Lighting App](https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/esp32)). I have compiled the code and put the lighting example into the device (ESP32 or ESP32-S3) and I trigger the chip tool with the next command:

```
cd
cd connectedhomeip
sudo ./out/ct/chip-tool pairing ble-wifi 445566 <MY_WIFI_SSID> <MY_WIFI_PASSWOD> 20202021 3840
```

The commissioning process start to work, but the chip tool stuck in the **'FindOperational'** state... I saw in the logs that the device achieve connected to Wifi and obtain its IPv6 address... But the **Chip-Tool** can't connect with the node again... At final **Chip-Tool** abort the process, but the device stays connected with Wifi without finish the commissioning process. In this point I need reflash the device again. I attached the logs captured in the process in both sides (Device and Chip-Tool). I have replaced my Wifi credentials in the logs by <MY_WIFI_SSID> and <MY_WIFI_PASSWORD>.

[ESP32-Chip-Tool-Logs.txt](https://github.com/project-chip/connectedhomeip/files/11098628/ESP32-Chip-Tool-Logs.txt)
[ESP32-Device-Logs.txt](https://github.com/project-chip/connectedhomeip/files/11098629/ESP32-Device-Logs.txt)

I have compiled the **Chip-Tool** in a Linux machine using **Ubuntu 22.04**.

Steps to compile **Chip-Tool**
```
git clone https://github.com/project-chip/connectedhomeip.git
cd connectedhomeip
source scripts/activate.sh
./scripts/examples/gn_build_example.sh examples/chip-tool out/ct
```

Steps to flash the Lighting App into ESP32
```
sudo apt install cmake
-----------------------------------
git clone -b v4.4.3 --recursive https://github.com/espressif/esp-idf.git
cd esp-idf
./install.sh
-----------------------------------
cd
cd esp-idf
source export.sh
cd
cd connectedhomeip
source scripts/activate.sh
-----------------------------------
cd
cd connectedhomeip/examples/lighting-app/esp32
idf.py set-target esp32
idf.py build
sudo usermod -a -G dialout $USER
sudo reboot
------------------------------------
cd connectedhomeip/examples/lighting-app/esp32
idf.py -p /dev/ttyUSB0 erase_flash
idf.py -p /dev/ttyUSB0 flash monitor
```

### Bug prevalence

In each commissioning process

### GitHub hash of the SDK that was being used

def9802275c65ce678217a55574ba70e1367d1a4

### Platform

esp32

bzbarsky-apple, 2023-03-29:
From the device log:
```
I (66058) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fe80:0000:0000:0000:f208:d1ff:fed0:4514
...
I (66088) chip[DL]: IPv4 address changed on WiFi station interface: 192.168.0.35/255.255.255.0 gateway 192.168.0.1
...
I (68058) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fdaa:bbcc:ddee:0000:f208:d1ff:fed0:4514
```
as the network comes up.

From the chip-tool log, the first CASE attempt looks like this:
```
1680077120.897642][12006:12008] CHIP:DIS: UDP:[fe80::f208:d1ff:fed0:4514%enp0s3]:5540: new best score: 3
[1680077120.897671][12006:12008] CHIP:DIS: UDP:192.168.0.35%enp0s3:5540: score has not improved: 2
...
[1680077120.898745][12006:12008] CHIP:IN: (U) Sending msg 238284817 to IP address 'UDP:[fe80::f208:d1ff:fed0:4514%enp0s3]:5540'
[1680077120.899171][12006:12008] CHIP:SC: Sent Sigma1 msg
```
and then we never get a response or ack from the device, so that times out.  Presumably because by this point the device is using `fdaa:bbcc:ddee:0000:f208:d1ff:fed0:4514`, not `fe80:0000:0000:0000:f208:d1ff:fed0:4514`, for its IP.

So then we try CASE again:
```
[1680077128.386009][12006:12008] CHIP:DIS: UDP:[fe80::f208:d1ff:fed0:4514%enp0s3]:5540: new best score: 3
1680077128.386193][12006:12008] CHIP:DIS: UDP:[fdaa:bbcc:ddee:0:f208:d1ff:fed0:4514%enp0s3]:5540: new best score: 6
[1680077128.386247][12006:12008] CHIP:DIS: UDP:192.168.0.35%enp0s3:5540: score has not improved: 2
...
[1680077128.386825][12006:12008] CHIP:IN: (U) Sending msg 238284818 to IP address 'UDP:[fdaa:bbcc:ddee:0:f208:d1ff:fed0:4514]:5540'
[1680077128.386962][12006:12008] CHIP:SC: Sent Sigma1 msg
```

and the same thing happens: no responses, no acks.  Device side log does not show it receiving those Sigma1 messages, even though now we are using the "new" IP.

So we try a third time:
```
[1680077170.871333][12006:12008] CHIP:DIS: UDP:[fe80::f208:d1ff:fed0:4514%enp0s3]:5540: new best score: 3
[1680077170.871448][12006:12008] CHIP:DIS: UDP:[fdaa:bbcc:ddee:0:f208:d1ff:fed0:4514%enp0s3]:5540: new best score: 6
[1680077170.871606][12006:12008] CHIP:DIS: UDP:192.168.0.35%enp0s3:5540: score has not improved: 2
...
[1680077170.872588][12006:12008] CHIP:IN: (U) Sending msg 238284819 to IP address 'UDP:[fdaa:bbcc:ddee:0:f208:d1ff:fed0:4514]:5540'
[1680077170.872835][12006:12008] CHIP:SC: Sent Sigma1 msg
```

and the same thing happens.

@leandropg Please check whether something in your network configuration is causing IPv6 packets to be dropped?

leandropg, 2023-03-29:
Thanks @bzbarsky-apple ... In fact is only a basic Wifi router with both Wifi bands: 2.4 GHz and 5 GHz... It has enabled assign IPv6 address too... It's configured to 2.4GHz (802.11g+n) and 5GHz (802.11a+n+ac)... The test must use the 2.4GHz band... I reviewed the router configuration and I don't see any strange configuration 🤔

bzbarsky-apple, 2023-03-29:
Is the Linux machine chip-tool is running on connected via 2.4 GHz or 5 GHz?  I assume both bands use the same SSID?

leandropg, 2023-04-11:
Hi @bzbarsky-apple I can't make the test because the router doesn't permit disable the 2.4GHz band... the 2.4GHz and 5GHz can be turn on or turn off together... :( Maybe I need install my own router to make this test... This is the ISP router... Is there routers with problems handling IPv6?

Additionally, I try to ping the IPv6 address of the ESP32 connected in the failed commissioning... I see the IPv6 address in the router... it was assigned by the router... Using the same linux pc with chip-tool... I run the ping command... But the node is unreachable... I can't reach a node that have a IPv6 address 😥

**ping -6 fdaa:bbcc:ddee:0000:260a:c4ff:fe5a:1f30**
```
PING fdaa:bbcc:ddee:0000:260a:c4ff:fe5a:1f30(fdaa:bbcc:ddee:0:260a:c4ff:fe5a:1f30) 56 data bytes
From fdaa:bbcc:ddee::1 icmp_seq=1 Destination unreachable: Unknown code 5
From fdaa:bbcc:ddee::1 icmp_seq=2 Destination unreachable: Unknown code 5
From fdaa:bbcc:ddee::1 icmp_seq=3 Destination unreachable: Unknown code 5
From fdaa:bbcc:ddee::1 icmp_seq=4 Destination unreachable: Unknown code 5
From fdaa:bbcc:ddee::1 icmp_seq=5 Destination unreachable: Unknown code 5
From fdaa:bbcc:ddee::1 icmp_seq=6 Destination unreachable: Unknown code 5
From fdaa:bbcc:ddee::1 icmp_seq=7 Destination unreachable: Unknown code 5
From fdaa:bbcc:ddee::1 icmp_seq=8 Destination unreachable: Unknown code 5
From fdaa:bbcc:ddee::1 icmp_seq=9 Destination unreachable: Unknown code 5
From fdaa:bbcc:ddee::1 icmp_seq=10 Destination unreachable: Unknown code 5
From fdaa:bbcc:ddee:0:720:9c14:2e33:9722 icmp_seq=11 Destination unreachable: Address unreachable
From fdaa:bbcc:ddee:0:720:9c14:2e33:9722 icmp_seq=12 Destination unreachable: Address unreachable
From fdaa:bbcc:ddee:0:720:9c14:2e33:9722 icmp_seq=13 Destination unreachable: Address unreachable
From fdaa:bbcc:ddee::1 icmp_seq=14 Destination unreachable: Unknown code 5
From fdaa:bbcc:ddee::1 icmp_seq=15 Destination unreachable: Unknown code 5
From fdaa:bbcc:ddee::1 icmp_seq=16 Destination unreachable: Unknown code 5
```

Is possible that the router is drop

bzbarsky-apple, 2023-04-11:
> Is there routers with problems handling IPv6?

Yes, that can also happen...

leandropg, 2023-04-12:
Hi @bzbarsky-apple ... Finally I repeated the test using a Wifi Hostpot instead my Wifi Router... and chip-tool can commission the device 😑The problem was my Wifi Router... For some reason it doesn't handle the IPv6 packets even when the IPv6 functionality is enabled... Maybe I must looking a tool that verify if the Wifi Router is ready to IPv6 and all the Matter commands... The problem weren't the Matter nodes or Chip-Tool... It was the Wifi Router... @bzbarsky-apple Do you know some tool that verify that a Wifi router is compatible with Matter? Thanks in advance

bzbarsky-apple, 2023-04-12:
@leandropg I don't know of a tool, sorry.

leandropg, 2023-04-13:
Thanks @bzbarsky-apple for your help. Considering that the problem was the router. I close this