project-chip/connectedhomeip #25910: [BUG][Android][Plugfest #4] Crash CHIPTool when read invalid String attribute

joonhaengHeo, 2023-03-31:
### Reproduction steps

1. Commissioning device using Android CHIPTool
2. Read attributes string type.
3. Crash when parsing String hex.

08-14 09:30:05.986 F 15822    15927    e.chip.chiptoo:                                                                     java_vm_ext.cc:577] JNI DETECTED ERROR IN APPLICATION: input is not valid Modified UTF-8: illegal continuation byte 0 
08-14 09:30:05.986 F 15822    15927    e.chip.chiptoo:                                                                     java_vm_ext.cc:577]     string: tfm_bas� 
08-14 09:30:05.986 F 15822    15927    e.chip.chiptoo:                                                                     java_vm_ext.cc:577]     input: 0x74 0x66 0x6d 0x5f 0x62 0x61 0x73 0xd5 
08-14 09:30:05.986 F 15822    15927    e.chip.chiptoo:                                                                     java_vm_ext.cc:577]     in call to NewStringUTF 

### Bug prevalence

read attribute with specific device.

### GitHub hash of the SDK that was being used

ce0ddc035143117d9193eb5b59647a7dd957aea0

### Platform

android

### Platform Version(s)

master

### Anything else?

During testing in Plugfest 4, while analyzing the issue, the same crash occurs in Android CHIPTool, so an issue is registered.

andy31415, 2023-04-03:
@joonhaengHeo what property did you try to set to `0x74 0x66 0x6d 0x5f 0x62 0x61 0x73 0xd5` ? It seems this tries to assign a non-utf8 string to a string. 

I am trying to figure out if matter spec require UTF8 or not but cannot see any obvious text where the data type string is defined ... but utf8 seems often reasonable when a separate "octet string" exists.

andy31415, 2023-04-03:
Trying to figure out if the fix should be instead an exception rather than allowing invalid utf8.

joonhaengHeo, 2023-04-04:
@andy31415 
As explained above, this issue is from Plugfest.

When trying to read a specific attribute, it was confirmed that `0x74 0x66 0x6d 0x5f 0x62 0x61 0x73 0xd5` values were read, and a crash occurred due to this value.
(0x34 - Software Diagnostics, 0x00 - ThreadMetrics)

Detailed log
`08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 		 
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 		AttributeReportIB = 
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 		{ 
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 			AttributeDataIB = 
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 			{ 
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 				DataVersion = 0x43495aa5, 
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 				AttributePathIB = 
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 				{ 
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 					Endpoint = 0x0, 
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 					Cluster = 0x34, 
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 					Attribute = 0x0000_0000, 
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 					ListIndex = Null, 
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 				} 
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 					 
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 				Data =  
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 				{ 
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 					0x0 = 11,  
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 					0x1 = "tfm_bas�" (8 chars),  
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 					0x3 = 879,  
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 					0x4 = 16777344,  
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 				}, 
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 			}, 
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 			 
08-14 09:30:05.977 D 15822    15927    DMG                                                                                     : 		}, `

joonhaengHeo, 2023-04-04:
My goal is also not to crash.
However, since this value is also read normally in chip-tool, I modified it in a way that can be read.

Do you think it is right to generate an error in your opinion?

PR #25936 (2023-04-12): [Android] Fix to avoid NewStringUTF crash
Fix #25910 

Crash is occurring when a value is passed instead of an ASCII value.
(Tested values are "0x74, 0x66, 0x6d, 0x5f, 0x62, 0x61, 0x73, 0xd5"). - It is occurring at 0xd5.
To solve this problem, I would like to prevent Crash by changing the method of generating java string.

1. Make new API in JniReference for changing jstring.
2. Using this API in ZAP files.