project-chip/connectedhomeip #25932: [BUG] DNS-SD operational discovery timeouts in network with many services?

plan44, 2023-03-31:
### Reproduction steps

## Matter Setup
- AppleTV as controller, Ethernet connection to LAN
- matter bridge app on Linux as accessory, Ethernet-only connection to LAN

## Summary
The Bug is that this setup works 100% in a LAN with ~50 published DNS-SD services total (all types, not just matter), but **completely fails operating in a LAN with ~120 published DNS-SD services**.

## Step 1: Commissioning in the LAN with ~120 DNS-SD services:
- AppleTV and bridge app connected to LAN
- Commission the bridge via an iPhone/iPad in the same LAN, everything works
- Bridge/Devices appear in Home.app
- Homekit passes operation to AppleTV, matter stack in AppleTV takes control, tries to **re-discover bridge via DNS-SD, 100% fails** like:

    ```
    Discovery	error	19:47:45.090502+0200	homed	Mdns: Resolve failure (kDNSServiceErr_Timeout)
    Discovery	error	19:47:45.090661+0200	homed	OperationalSessionSetup[1:0000000042169F10]: operational discovery failed: ../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:96: CHIP Error 0x00000032: Timeout
    DataManagement	error	19:47:45.090863+0200	homed	Failed to establish CASE for re-subscription with error '../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:96: CHIP Error 0x00000032: Timeout'
    ```

    which then leads to

    ```
    hmmtr.accessory.server	error	19:47:45.431924+0200	homed	[975063280/1] Bridged accessory for endpoint 3 is unreachable
    hmmtr.accessory.server	error	19:47:45.432064+0200	homed	[975063280/1] Read cache operation aborted for characteristic since the bridged accessory is unreachable: <HAPCharacteristic>
    ```

    which means all bridged devices immediately get status "not responding" and do never get operational.

## Step 2: Move the setup to LAN with fewer DNS-SD services (~50)
- changing **nothing** in the setup, not even power down, just swap ethernet connection into a smaller test LAN with ~50 DNS-SD services.
- wait a little
- **bridged devices get fully operational**, and remaining so.

## Step 3..n: Switch back an forth
- Without doing anything but plugging ethernet jacks to switch LANs, the setup can be made work and fail back and forth.

### Bug prevalence

100% reproducible

### GitHub hash of the SDK that was being used

9e371e6f191c15ac446e0cc2270a420458018cda

### Platform

darwin

### Platform Version(s)

16.3, 16.4, 16.5 beta

### Anything else?

## Notes:
- the relevant `_matter._tcp` services get visible reliably in **both** ~120 and ~50 services LANs likewise when observing with `avahi-browse -a -t` from linux based bridge itself as well as from another linux based system in the same LAN.
- the relevant `_matter._tcp` service is also reliably visible in **both** ~120 and ~50 services LANs likewise when observing with Lily Ballard's `Discovery` native macOS app (means: using macOS native DNS-SD).
- Accessory/bridge side tested with 9e371e6f191c15ac446e0cc2270a420458018cda and v1.0.0
- Controller (AppleTV) side tested with tvOS 16.3, 16.4 and 16.5 beta 1.

## Conclusion:
- As avahi and discovery app always see and resolve all services for ~50 and ~120 services cases, the publishing part (bridge accessory) cannot be at fault. **It must be the DNS-SD browsing side in the controller not seeing the published services.**
- Assuming same DNS-SD stack in macOS as in tvOS (@bzbarsky-apple: true?), this seems more likely a problem in the matter layers on top of darwin platform DNS-SD **so that's why I post the issue here**. 
- As the error is "timeout", there might be hard-coded assumptions somewhere how long at most service discovery or address resolution might take, which might be value somehow dependent on the total number of services, leading to "giving up too early".

bzbarsky-apple, 2023-03-31:
> Assuming same DNS-SD stack in macOS as in tvOS (@bzbarsky-apple: true?),

I don't know, but @Abhayakara might.

> this seems more likely a problem in the matter layers on top of darwin platform DNS-SD

Assuming that they are doing the same things.  I expect `Discovery` does a browse, and certainly `avahi-browse` does, but operational session setup just does a resolve.

Per the log above (and note that the relevant SHA for the tvOS bits is _not_ the SHA you cited), you are geting a timeout from here in `src/lib/address_resolve/AddressResolve_DefaultImpl.cpp`:

```
NodeLookupAction NodeLookupHandle::NextAction(System::Clock::Timestamp now)
....
    if (elapsed >= mRequest.GetMaxLookupTime())
    {
        return NodeLookupAction::Error(CHIP_ERROR_TIMEOUT);
    }
```

The max lookup time for an operational lookup in the SDK right now is 45 seconds.  After that we fail out.  This is not MacOS-specific.

Abhayakara, 2023-04-01:
I've chatted about this with Boris privately, but just to summarize, I think this approach isn't really thinking about user experience very carefully. Two observations I made:

1. The timeout is actually the result of the network behaving badly. I've looked at the packet trace, and we've had some email discussions and a meeting about this, and the upshot appears to be that the router is not reliably delivering multicast packets. This is a problem, but one aspect of the problem is actually liaising with the router manufacturer to get the bug fixed (or at least understand why it's doing this). In the case I looked at where there were pcap traces, the timeout happened only because of this multicast packet loss, not because of any issue on the mDNS registrar or with the mDNS queries on the pairing host.
2. The user experience when the network is behaving badly should be as good as possible. If the app has all the information it needs to complete the pairing, it should be fine for the user to put the phone in their pocket or move on to the next device. That is to say, timeouts are simply a bad programming model. The right thing to do is just persist until the transaction finishes, giving the user the information they need to understand what is going on and what the expected outcome is. So I think a first step in that direction is to not have the API force the app to time out. Of course the actual behavior of the app is up to the app developer, but an API that prevents the app developer from delivering the best possible UX when the network is bad is not a well-designed API.

bzbarsky-apple, 2023-04-01:
@Abhayakara In this case the "pairing" completes fine.  It's just attempts to actually use the device afterward that don't work particularly well in the case describe above.  This is unrelated to the situation with the packet traces you were looking at, as far as I know.

Abhayakara, 2023-04-01:
Hm, okay, but I don’t think this changes anything. We need to understand why this is happening. And the app should decide when to give up, not the api. 

Can we get some packet traces for this case?

Abhayakara, 2023-04-01:
Sorry, I’m realizing that the matter bridge app may be the first thing for me to understand. I hadn’t heard about this before. I just read it as “bridge” which probably isn’t the same thing. What is the bridge app supposed to do?

plan44, 2023-04-01:
> @Abhayakara : What is the bridge app supposed to do?

I meant to refer to the `connectedhomeip` example in `examples/bridge-app`. For the controller, it is just a special kind of accessory, with a root end point representing the bridge and child end points representing *bridged devices*.

I consider it very unlikely that the choice of accessory example is relevant in the context of this problem (I's just that I work with bridges so that where I can observe this **controller side** discovery problem right now)

plan44, 2023-04-01:
> @Abhayakara  1. The timeout is actually the result of the network behaving badly. I've looked at the packet trace, and we've had some email discussions and a meeting about this, and the upshot appears to be that the router is not reliably delivering multicast packets. This is a problem, but one aspect of the problem is actually liaising with the router manufacturer to get the bug fixed (or at least understand why it's doing this). In the case I looked at where there were pcap traces, the timeout happened only because of this multicast packet loss, not because of any issue on the mDNS registrar or with the mDNS queries on the pairing host.

Just saying, I'm *acutely* aware of these problems with routers, "smart" switches and multicast, because DNS-SD is a centerpiece in the devices I work with (and have to support in customer environments). So I can assure you that my lab network does *not* have these problems. Browsing and resolving names works fine and with no noticeable delay in this network.

It must be something more specific to the matter resolver.

Abhayakara, 2023-04-01:
Ok. What is the topology? Is the "matter bridge" bridging between two vlans?

I can think of a couple of issues that could happen here.  First, going from 50 to 120 devices is more than doubling the number of things that need to be announced via mDNS. To see the matter resolver suddenly not be able to resolve anything is a bit weird, but not completely impossible.

It would really help to see packet traces of the working scenario and the non-working scenario, from the perspective of the device on which the resolver is running.

plan44, 2023-04-02:
> Ok. What is the topology? Is the "matter bridge" bridging between two vlans?

The "matter bridge" is not bridging anything in TCP or Ethernet network terms. It's just a matter node that represents/maps non-matter devices into matter (see *Matter-1.0-Core-Specification, 9.12.Bridge for non-Matter devices*, pages 475ff). There is no difference between a "matter bridge" and any other accessory type regarding discovery and network connectivity.

There are no VLANs involved, just a single network segment with devices connected through "dumb" switches (GS108E). There's a router providing DHCPv4 + v6, v6 RA, and internet access. The router is not involved in forwarding any LAN traffic (except to the internet, of course).

> @Abhayakara It would really help to see packet traces of the working scenario and the non-working scenario, from the perspective of the device on which the resolver is running.

I wish I could deliver that - but the controller (device on which resolver is running) is an AppleTV, where all I can do is provide console logs. Let me know if you need something more specific than the parts I already quoted above.

I can only state that DNS-SD browsing and resolving works fine in the same ~120 service network *from all other network nodes I have access to* (macOS computers, Linux boxes).

> @bzbarsky-apple The max lookup time for an operational lookup in the SDK right now is 45 seconds. After that we fail out. This is not MacOS-specific.

Exactly because that (enormous) timeout seems to be reached in the AppleTV, while all other resolvers in the same network (including macOS laptop) can resolve the matter accessory in fractions of a second using platform APIs/tool, I assume it must be something more subtle and also specific to the matter resolver.

> @Abhayakara I can think of a couple of issues that could happen here. First, going from 50 to 120 devices is more than doubling the number of things that need to be announced via mDNS. To see the matter resolver suddenly not be able to resolve anything is a bit weird, but not completely impossible.

The net with ~120 *services* (not: *devices*! - there are only ~30 wired + ~10 wireless network devices in that network segment) used to work with matter until recently, when a few more devices were added. Thus, the threshold services count that seems to make AppleTV resolver suddenly fail 100% is probably near ~120 rather than ~50.

The ~120 services network is an operational real-world lab network, where the possibilities of adding/removing things is limited for me. I can observe it, but not tear it apart ;-) The ~50 services network is a toy network I set up as a first step trying to narrow down this issue.

I am aware that the data I can provide at this point is not much, so I'm open to your suggestions what to try and what to collect for getting more insight what could be wrong.

I am not aware of DNS-SD scaling badly in moderately sized networks (40 IP devices is not really a lot, many smarthome user's network I've seen have more). The only systematic reason for DNS-SD failing I know of are multicast forwarding issues caused by "smart" switches and routers with broken IGMP snooping implementations. My lab network is free of these (for that very reason).

Abhayakara, 2023-04-02:
Thanks for the detailed explanation—this definitely helps.

One thing to say is that I am just worried that mDNS isn't scaling for some reason because that's a part of the system I'm responsible for, and what you are seeing sounds bad, and so I tend to catastrophize about it a bit, but that doesn't mean my speculation is correct. Based on what you are saying I think that's not actually what's happening. I am now wondering if we aren't just running into a crash or some other bug. It might help to reproduce the success situation, take note of the time, reproduce the failure situation, take not of the time, and then take a system diagnostic. I've send you instructions for sharing that separately—definitely don't post it here.

plan44, 2023-04-03:
Thanks @Abhayakara for the logging instructions (got them), I will re-test as soon as I can.

Unfortunately, at the moment I ran into a totally unrelated issue that blocks testing (I wanted to do a clean start, removed the accessory and re-adding now fails for some crypto reason, ECDSA hash signature check suddenly failing in the accessory). I need to debug this first in order to be able to set up a commissioned accessory to test DNS-SD operational discovery against at all...

plan44, 2023-04-05:
@Abhayakara JFYI: after a day of debugging: it could be a red herring - still I do not understand the root cause, but it seems *likely* now it is only indirectly dsn-sd activity related (a weirdness with the avahi setup on the test platform I haven't figured out yet).
Also, I'm not back at the clean test setup I had on Friday, because I cannot re-add that particular accessory in Home.app yet due to the unrelated ECDSA hash check problem.
As soon as I have some more insight, I'll update this issue.

Anyway, thanks a lot for helping, and sorry for having caused you worries...

Abhayakara, 2023-04-05:
No worries, I'm happy to help and appreciate both the update and the information so far!

plan44, 2023-04-11:
@Abhayakara Still testing details after easter vacation, no real conclusions yet why the situation is not really stable (less so with many services, but not completely stable even in the small network, I found).

While testing, I noticed differences between what the specs say and how the (avahi-based) implementation currently works:

- Specs say (Matter 1.0 Core Specs, pg 82) the nodes should not publish IPv4 address records, but IPv6 only. The spec's recommendation to do that on platforms that still need to publish the IPv4 host address for other purposes (like mine) is to use a separate host name for matter, publishing IPv6 records only.
- The current avahi implementation does not do that (I need to look further into that to see why, because technically it should be possible with Avahi). It just attempts to monopolize the host name by forcing it to be the MAC-based matter-style name for all services (not good for other non-matter services), and then publishes IPv6 *and* IPv4 addresses for matter services.
- In consequence, the hostname was occasionally changing, because other parts of the system disagreed about the host name and sometimes did revert it back to the original, non-matter platform name.

Do you think that one of these could confuse controller-side discovery:
- IPv4 address records also published for matter nodes?
- hostname changing from time to time?
- hostname not conforming to the matter host name construction scheme (Matter 1.0 core specs 4.3.1.1, pg 84) of 12-char MAC address?

Abhayakara, 2023-04-11:
I could see this causing problems just because of the name conflict issue. Matter absolutely relies on names not changing.

plan44, 2023-04-11:
Ok, I will explore further and make sure the host name does not change any more.

However, there was never a *conflict*, just the hostname (*not* the service names, these were always correct and stable) did occasionally change because the matter stack and another daemon on the platform both did update it. Still, the information published was never inconsistent.

Abhayakara, 2023-04-11:
Hm. Then why did it change the name?  Anyway, if the library that is discovering the device doesn’t track updates in the SRV record, which is an easy mistake to make, then it won’t see the new information because the hostname changed.

Abhayakara, 2023-04-11:
I think the correct behavior is for matter to have its own domain name for address records that is independent of the hostname that’s used by the host for non-matter purposes. This isn’t normally an issue because matter devices are normally not multi-purpose. 

As for discovering, what I mean is that something wants to talk to the device. That thing needs to track the srv record. If it’s not doing so, it’s not going to work. If that’s the apple device, then it’s worth asking whether it is doing this correctly for matter devices.

Abhayakara, 2023-04-11:
@bzbarsky-apple can you see if this is being done correctly?

plan44, 2023-04-12:
> It sounds like the Linux advertising code doesn't, do that?

It doesn't. It just creates a client and registers the matter services, referring to the default host, and then just calls `avahi_client_set_host_name()` to force the matter hostname upon anything using avahi on that host.

> Avahi command-line does support publishing multiple hostnames for the same IP address [...]
> I would assume the avahi C API lets you do that too [...]

Yes, that's what I was experimenting with yesterday. There is `avahi_entry_group_add_address()` which can establish additional hostname/address pairs. The problem here is obtaining the IP(s) to publish in a clean way from within `platform/linux/DnssdImpl.cpp`.

Related to that, I'm back at question about what addresses to publish at all. The current matter SDK seems to enable (and thus publish) IPv4 pretty everywhere, despite the specs saying matter does not and should not use it.

So when trying to improve `platform/linux/DnssdImpl.cpp`, what should I use as a strategy to obtain the address(es)?

The `chip::Dnssd::DnssdService` struct as set-up by the higher levels demands `Inet::IPAddressType::kAny` (Discovery_ImplPlatform.cpp:499), i.e. all IPv4 and IPv6 addresses. Even one level up, in app/server/Dnssd.cpp:170 and 189, I see `.EnableIpV4(true)` (not relevant technically for the avahi implementation, but expressing some intent contrary to the specs).

Following the specs however I'd guess only IPv6 link local would be relevant, and everything else should not be published for efficiency?

plan44, 2023-04-12:
For now I am considering the case of my "already on wired ethernet" device only, which only advertises in `.local`.
 
Am I wrong to think that things advertised in `.local` would *need* to be located in the same network segment, hence always be reachable via link-local address, without any routing?

bzbarsky-apple, 2023-04-12:
@plan44 In practice, Matter SDK is at "you must do IPv6, and can also do IPv4 if you want to".  That applies to advertising, resolution, etc.

For a Thread node, obviously only IPv6 will work, so everything involved generally needs to support IPv6 to enable interop with Thread.

plan44, 2023-04-18:
@bzbarsky-apple, @Abhayakara  JFYI: I'm working on a patch for the Linux/Avahi implementation to make it work the same way the Darwin implementation does (separate domain name instead of renaming entire platform's). Almost finished, but…

…unfortunately I have totally unrelated other things on my table I have to finish first (as much as I would prefer working on matter) 😞 

And then there is the weird ECDSA hash signature check for `initiatorPublicKey` suddenly failing in `protocols/secure_channel/CASESession.cpp:1468`, which I haven't analyzed enough to post as an issue. It might be something I messed up myself, but maybe also not. Either way, it does not help with testing DNS-SD reliability because it currently prevents adding new accessories to home.

I hope I will find the needed time to sort out these loose ends next week.

fxm2022, 2023-04-21:
@plan44 Could you try to upgrade your HomePod and IOS to 16.5beta2. It may help your ECDSA issue.

plan44, 2023-05-04:
@fxm2022 Thanks! I had a chance to re-test with latest tvOS 16.5 beta (20L5559a) and in fact the ECDSA issue is gone 😄 

@bzbarsky-apple Further investigation into the Avahi based implementation revealed that it does not properly handle publishing more than one service. The code in `src/platform/Linux/DnssdImpl.cpp` just tries to add a service to an already commited (published) service group, which is not supported by avahi at all.

This did not cause problems for some time, as long as only one service had to be published. Now that the new Home architecture (right?) commissions the node into two fabrics, both need to be published via DNS-SD but the second one fails.

I am about to fix all this, having understood the implementation by now, but given the number of moving targets (the original DNS-SD weirdness in my larger network is not yet completely understood) and my non-matter related workload these days, it takes longer to make a presentable PR than anticipated.

If anyone else with more time wants to start working on this, please let me know, I can share the draft code when needed.

bzbarsky-apple, 2023-05-04:
> just tries to add a service to an already commited (published) service group, which is not supported by avahi at all.

Fun and exciting....

Please feel free to create an issue and/or draft PR just so we don't lose track of this, @plan44 !  And thank you for figuring it out.

plan44, 2023-05-05:
> Fun and exciting....
> 
> Please feel free to create an issue and/or draft PR just so we don't lose track of this, @plan44 ! And thank you for figuring it out.

@bzbarsky-apple  Here's a draft PR: #26397

For sure this improves the situation in my testing.

I will update (and possibly close) if this issue turns out to be caused by the fact that before the avahi fix, multiple fabrics could not be published. I have no conclusive idea yet why the outcome could be so different depending on the number of dns-sd services, as I initially observed, though.

plan44, 2023-07-21:
After sucessfully running continuous field tests for a few weeks now in the same environments as described in the original issue, **but with #26397 applied**, and not noticing any DNS-SD weirdness any more, I conclude:

- the basic issue was most likely the **randomly alternating host name** due to faulty matter side Avahi implementation (fixed with #26397).
- the apparently direct dependency between number of devices and actual problems occurring was probably just a matter of how much hostname confusion the combined DNS-SD participants with their caches could handle.
  With only few devices, things could recover quickly enough for practical purposes, with many (and many of them using the faulty Avahi implementation, in the original test!) the overall reliability of DNS-SD just dropped beyond what error handling in the participating nodes could work around using retries.

So I consider this problem solved and will close the issue now. Thanks to all for helping!

closed 2023-07-21 by plan44