project-chip/connectedhomeip #25934: [BUG] Handling large uint64 values in Android and Flutter with CHIP library

TheWalkingDead1024, 2023-04-01:
### Reproduction steps

I am currently working on a project that involves the CHIP library and requires handling uint64 data types in Android and Flutter environments. I have encountered issues when using JSON encoding and decoding due to the lack of support for BigInt in Android and Flutter.

When dealing with uint64 values that exceed the int64 range, these large values are represented as doubles, leading to inaccurate data. This causes issues with data accuracy when encoding and decoding JSON in Android and Flutter, as the large uint64 values are converted to doubles.

To ensure data accuracy and proper handling of large uint64 values, I kindly request guidance on how to work with CHIP library in Android and Flutter environments, considering the limitations of JSON support for BigInt in these platforms.

Suggestions or workarounds to handle large uint64 values accurately would be greatly appreciated, as this would improve the overall compatibility and reliability of the CHIP library when working with such values in Android and Flutter.

Thank you for your time and consideration.

### Bug prevalence

whenever I do this

### GitHub hash of the SDK that was being used

1.0

### Platform

android

bzbarsky-apple, 2023-04-02:
@TheWalkingDead1024 Is the JSON encoding in question happening inside or outside the Matter library?

If outside, it's the responsibility of whatever code is doing the encoding to encode things in ways that are not lossy, no?  This applies to octet strings (which need some work to represent in JSON), 64-bit integer values (both signed and unsigned), etc.

TheWalkingDead1024, 2023-08-22:
> @TheWalkingDead1024 Is the JSON encoding in question happening inside or outside the Matter library?
>
> If outside, it's the responsibility of whatever code is doing the encoding to encode things in [...]

Thank you for your response, @bzbarsky-apple.

You're right, the issue does indeed occur outside of the Matter library. However, the primary challenge we face is that certain platforms, like Flutter and Android, do not support unsigned 64-bit integers. As a result, when we need to represent uint64 values that exceed the int64 range on these platforms, we resort to using BigInt. To achieve this, we first represent the unsigned 64-bit integer from the Matter library as a signed 64-bit integer on these platforms, and then initialize it as BigInt. Similarly, when setting data from these platforms to the Matter library, we first convert from BigInt to a signed 64-bit integer, and then proceed with the setting.

This requires us to write such conversion code in multiple places, adding to the complexity of the code and potential for errors. It would greatly simplify our workflow and increase efficiency if the Matter library could provide support for this scenario at its core.

I understand that this might require some modifications to the library, but I believe it would be immensely beneficial for many developers, especially those working on the Flutter and Android platforms. Thank you once again for your time and consideration, and I hope we can find a more elegant solution.

Alami-Amine, 2025-01-04:
@TheWalkingDead1024 Is this feature still needed? You could draft a Pull Request with a possible solution and then it can be reviewed. Otherwise it seems this issue is stale, will close for now (could be reopened if needed).

closed 2025-01-04 by Alami-Amine