project-chip/connectedhomeip #25938: [BUG] restyle-diff script is not running

joonhaengHeo, 2023-04-03:
### Reproduction steps

After merging #25887, restyle-diff.sh is not running in below log.

usage: git diff [<options>] [<commit> [<commit>]] [--] [<path>...]
2023-04-03 04:16:18 [info     ] Fetching remote file            path=/tmp/restylers-stable.yaml

After erasing `--merge-base`, it was confirmed that the script was executed.

### Bug prevalence

100%

### GitHub hash of the SDK that was being used

7ccd5d40e4540e2a2ef1b24686a8128fe6c6d2ff

### Platform

core

bzbarsky-apple, 2023-04-03:
@joonhaengHeo Is that not running in Matter CI, or locally for you?

If locally, what git version are you using?  It looks like it's a version that does not know about `--merge-base`, but hard to tell, since you didn't actually include much of the log....

bzbarsky-apple, 2023-04-03:
And in particular, the `--merge-base` option was added to git in version 2.30 back about two years ago, but if you're on some sort of LTS, you might be using something older than that...

joonhaengHeo, 2023-04-04:
I used it when a restyle error occurred after code upload. I ran it on Ubuntu 20.04 LTS version. It is confirmed that my environment is the old environment. Now that I know the reason for the change, I will close the issue.

closed 2023-04-04 by joonhaengHeo

bzbarsky-apple, 2023-04-04:
It's possible we could use the older form, with nested git invocation, instead of `--merge-base`, to get the desired behavior.  If that would be useful, please just open a PR with the proposal?