project-chip/connectedhomeip #25957: [BUG] [Android] When SCAN QR, Android Chip Tool happen to Crash

jeonghwan7, 2023-04-04:
### Reproduction steps

When Scan QR Code, Android Chip-tool happen to crash 

The Log is below
```
04-04 16:28:57.789 E 32290    32290    AndroidRuntime:                                                                                               FATAL EXCEPTION: main 
04-04 16:28:57.789 E 32290    32290    AndroidRuntime:                                                                                               Process: com.google.chip.chiptool, PID: 32290 
04-04 16:28:57.789 E 32290    32290    AndroidRuntime:                                                                                               java.lang.IllegalStateException: Neither the parent Fragment null nor the host Activity null of BarcodeFragment implement Callback. 
04-04 16:28:57.789 E 32290    32290    AndroidRuntime:                                                                                               	at com.google.chip.chiptool.util.FragmentUtil.getHost(FragmentUtil.kt:49) 
04-04 16:28:57.789 E 32290    32290    AndroidRuntime:                                                                                               	at com.google.chip.chiptool.setuppayloadscanner.BarcodeFragment.handleScannedQrCode$lambda$8(BarcodeFragment.kt:215) 
04-04 16:28:57.789 E 32290    32290    AndroidRuntime:                                                                                               	at com.google.chip.chiptool.setuppayloadscanner.BarcodeFragment.lambda$ZQcs77Yp5AxXQmBnFOw77e2FItM(Unknown Source:0) 
04-04 16:28:57.789 E 32290    32290    AndroidRuntime:                                                                                               	at com.google.chip.chiptool.setuppayloadscanner.-$$Lambda$BarcodeFragment$ZQcs77Yp5AxXQmBnFOw77e2FItM.run(Unknown Source:4) 
04-04 16:28:57.789 E 32290    32290    AndroidRuntime:                                                                                               	at android.os.Handler.handleCallback(Handler.java:942) 
04-04 16:28:57.789 E 32290    32290    AndroidRuntime:                                                                                               	at android.os.Handler.dispatchMessage(Handler.java:99) 
04-04 16:28:57.789 E 32290    32290    AndroidRuntime:                                                                                               	at android.os.Looper.loopOnce(Looper.java:226) 
04-04 16:28:57.789 E 32290    32290    AndroidRuntime:                                                                                               	at android.os.Looper.loop(Looper.java:313) 
04-04 16:28:57.789 E 32290    32290    AndroidRuntime:                                                                                               	at android.app.ActivityThread.main(ActivityThread.java:8775) 
04-04 16:28:57.789 E 32290    32290    AndroidRuntime:                                                                                               	at java.lang.reflect.Method.invoke(Native Method) 
04-04 16:28:57.789 E 32290    32290    AndroidRuntime:                                                                                               	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571) 
04-04 16:28:57.789 E 32290    32290    AndroidRuntime:                                                                                               	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067) 

```
This PR is already apply https://github.com/project-chip/connectedhomeip/issues/25140

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

def9802275c65ce678217a55574ba70e1367d1a4

### Platform

android

nxtm555, 2023-04-05:
I also same Crash, using Master branch.
This log is when I scan QR code after push PROVISION CHIP DEVICE WITH WI-FI button.
 

```Log 
2023-04-05 10:47:10.183 30156-30156/com.google.chip.chiptool D/SPL: JNI_OnLoad() called
2023-04-05 10:47:10.211 30156-30194/com.google.chip.chiptool D/Camera2CameraImpl: {Camera@9436d3d[id=0]} Use cases [androidx.camera.core.Preview-7c0f662d-f638-4e28-8f45-1854433619aa21308708, androidx.camera.core.ImageAnalysis-87c901b5-dc2f-49c3-ab72-9349e4e3ffd8161085581] now DETACHED for camera
2023-04-05 10:47:10.211 30156-30194/com.google.chip.chiptool D/UseCaseAttachState: All use case: [] for camera: 0
2023-04-05 10:47:10.212 30156-30194/com.google.chip.chiptool D/Camera2CameraImpl: {Camera@9436d3d[id=0]} Resetting Capture Session
2023-04-05 10:47:10.214 30156-30194/com.google.chip.chiptool D/Camera2CameraImpl: {Camera@9436d3d[id=0]} Releasing session in state OPENED
2023-04-05 10:47:10.215 30156-30194/com.google.chip.chiptool D/UseCaseAttachState: Active and attached use case: [] for camera: 0
2023-04-05 10:47:10.217 30156-30194/com.google.chip.chiptool D/UseCaseAttachState: Active and attached use case: [] for camera: 0
2023-04-05 10:47:10.220 30156-30194/com.google.chip.chiptool D/Camera2CameraImpl: {Camera@9436d3d[id=0]} Closing camera.
2023-04-05 10:47:10.220 30156-30194/com.google.chip.chiptool D/Camera2CameraImpl: {Camera@9436d3d[id=0]} Transitioning camera internal state: OPENED --> CLOSING
2023-04-05 10:47:10.222 30156-30194/com.google.chip.chiptool D/CameraStateRegistry: Recalculating open cameras:
    Camera                                       State                 
    -------------------------------------------------------------------
    Camera@60a4c2c[id=1]                         UNKNOWN               
    Camera@9436d3d[id=0]                         CLOSING               
    -------------------------------------------------------------------
    Open count: 1 (Max allowed: 1)
2023-04-05 10:47:10.222 30156-30194/com.google.chip.chiptool D/CameraStateMachine: New public camera state CameraState{type=CLOSING, error=null} from CLOSING and null
2023-04-05 10:47:10.222 30156-30194/com.google.chip.chiptool D/CameraStateMachine: Publishing new public camera state CameraState{type=CLOSING, error=null}
2023-04-05 10:47:10.223 30156-30194/com.google.chip.chiptool D/Camera2CameraImpl: {Camera@9436d3d[id=0]} Resetting Capture Session
2023-04-05 10:47:10.223 30156-30194/com.google.chip.chiptool D/Camera2CameraImpl: {Camera@9436d3d[id=0]} Releasing session in state CLOSING
2023-04-05 10:47:10.224 30156-30194/com.google.chip.chiptool D/CaptureSession: onSessionFinished()
2023-04-05 10:47:10.236 30156-30156/com.google.chip.chiptool D/SurfaceViewImpl: Surface destroyed.
2023-04-05 10:47:10.236 30156-30156/com.google.chip.chiptool D/SurfaceViewImpl: Surface invalidated androidx.camera.core.SurfaceRequest@e75f4bc
2023-04-05 10:47:10.236 30156-30156/com.google.chip.chiptool D/DeferrableSurface: surface closed,  useCount=1 closed=true androidx.camera.core.SurfaceRequest$2@e32ac45
2023-04-05 10:47:10.239 30156-30156/com.google.chip.chiptool D/DeferrableSurface: surface closed,  useCount=1 closed=true androidx.camera.core.impl.ImmediateSurface@50ba0a8
2023-04-05 10:47:10.251 30156-30172/com.google.chip.chiptool E/BufferQueueProducer: [SurfaceView[com.google.chip.chiptool/com.google.chip.chiptool.CHIPToolActivity]#1(BLAST Consumer)1](id:75cc00000003,api:4,p:1644,c:30156) cancelBuffer: BufferQueue has been abandoned
2023-04-05 10:47:10.252 30156-30156/com.google.chip.chiptool D/StreamStateObserver: Update Preview stream state to IDLE
2023-04-05 10:47:10.254 30156-30172/com.google.chip.chiptool E/BufferQueueProducer: [SurfaceView[com.google.chip.chiptool/com.google.chip.chiptool.CHIPToolActivity]#1(BLAST Consumer)1](id:75cc00000003,api:4,p:1644,c:30156) cancelBuffer: BufferQueue has been abandoned
2023-04-05 10:47:10.261 30156-30156/com.google.chip.chiptool D/AndroidRuntime: Shutting down VM
    
    
    --------- beginning of crash
2023-04-05 10:47:10.262 30156-30156/com.google.chip.chiptool E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.google.chip.chiptool, PID: 30156
    java.lang.IllegalStateException: Neither the parent Fragment null nor the host Activity null of BarcodeFragment implement Callback.
        at com.google.chip.chiptool.util.FragmentUtil.getHost(FragmentUtil.kt:49)
        at com.google.chip.chiptool.setuppayloadscanner.BarcodeFragment.handleScannedQrCode$lambda$8(BarcodeFragment.kt:215)
        at com.google.chip.chiptool.setuppayloadscanner.BarcodeFragment.lambda$ZQcs77Yp5AxXQmBnFOw77e2FItM(Unknown Source:0)
        at com.google.chip.chiptool.setuppayloadscanner.-$$Lambda$BarcodeFragment$ZQcs77Yp5AxXQmBnFOw77e2FItM.run(Unknown Source:4)
        at android.os.Handler.handleCallback(Handler.java:942)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loopOnce(Looper.java:201)
        at android.os.Looper.loop(Looper.java:288)
        at android.app.ActivityThread.main(ActivityThread.java:7884)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
    
    
    --------- beginning of system
2023-04-05 10:47:10.286 30156-30156/? I/Process: Sending signal. PID: 30156 SIG: 9
```

nicelyjust, 2023-04-13:
@nxtm555 pls refer to [this](https://github.com/project-chip/connectedhomeip/pull/25958#issuecomment-1506667567),hope to help you

yunhanw-google, 2024-01-18:
fix merged. thanks

closed 2024-01-18 by yunhanw-google