project-chip/connectedhomeip #25963: [BUG]

lboue, 2023-04-04:
### Reproduction steps

## Reproduction steps:

Build the Python/C library:

```
cd ~/connectedhomeip/
git submodule update --init
source scripts/activate.sh

./scripts/build_python_device.sh --chip_detail_logging true

sudo su # dhclient is called, needs root
source ./out/python_env/bin/activate
```

## Error
`../../src/app/clusters/ota-requestor/DefaultOTARequestor.cpp:30:10: fatal error: zap-generated/CHIPClusters.h: No such file or directory`

### Bug prevalence

always

### GitHub hash of the SDK that was being used

d81d4e367d4ba80eef751675ce29ebfdbba74c84

### Platform

python

### Anything else?

```
FAILED: obj/src/app/clusters/ota-requestor/lighting-common.DefaultOTARequestor.cpp.o
g++ -MMD -MF obj/src/app/clusters/ota-requestor/lighting-common.DefaultOTARequestor.cpp.o.d -Wconversion -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -I/usr/include/gio-unix-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=\<lib/address_resolve/AddressResolve_DefaultImpl.h\> -DCHIP_MINMDNS_USE_EPHEMERAL_UNICAST_PORT=1 -DCHIP_MINMDNS_HIGH_VERBOSITY=0 -DCHIP_MINMDNS_DEFAULT_POLICY=1 -I../../zzz_generated/lighting-app -Igen/examples/lighting-app/lighting-common -Igen/examples/lighting-app/lighting-common/zapgen -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -I../../third_party/inipp/repo/inipp -c ../../src/app/clusters/ota-requestor/DefaultOTARequestor.cpp -o obj/src/app/clusters/ota-requestor/lighting-common.DefaultOTARequestor.cpp.o
../../src/app/clusters/ota-requestor/DefaultOTARequestor.cpp:30:10: fatal error: zap-generated/CHIPClusters.h: No such file or directory
   30 | #include <zap-generated/CHIPClusters.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[2/603] g++ -MMD -MF obj/src/app/clusters/ota-requestor/lighting-common.OTATestEventTriggerDelegate.cpp.o.d -Wconversion -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -I/usr/include/gio-unix-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=\<lib/address_resolve/AddressResolve_DefaultImpl.h\> -DCHIP_MINMDNS_USE_EPHEMERAL_UNICAST_PORT=1 -DCHIP_MINMDNS_HIGH_VERBOSITY=0 -DCHIP_MINMDNS_DEFAULT_POLICY=1 -I../../zzz_generated/lighting-app -Igen/examples/lighting-app/lighting-common -Igen/examples/lighting-app/lighting-common/zapgen -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -I../../third_party/inipp/repo/inipp -c ../../src/app/clusters/ota-requestor/OTATestEventTriggerDelegate.cpp -o obj/src/app/clusters/ota-requestor/lighting-common.OTATestEventTriggerDelegate.cpp.o
[3/603] g++ -MMD -MF obj/src/app/clusters/ota-requestor/lighting-common.ExtendedOTARequestorDriver.cpp.o.d -Wconversion -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -I/usr/include/gio-unix-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=\<lib/address_resolve/AddressResolve_DefaultImpl.h\> -DCHIP_MINMDNS_USE_EPHEMERAL_UNICAST_PORT=1 -DCHIP_MINMDNS_HIGH_VERBOSITY=0 -DCHIP_MINMDNS_DEFAULT_POLICY=1 -I../../zzz_generated/lighting-app -Igen/examples/lighting-app/lighting-common -Igen/examples/lighting-app/lighting-common/zapgen -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -I../../third_party/inipp/repo/inipp -c ../../src/app/clusters/ota-requestor/ExtendedOTARequestorDriver.cpp -o obj/src/app/clusters/ota-requestor/lighting-common.ExtendedOTARequestorDriver.cpp.o
ninja: build stopped: subcommand failed.
```

lazeintech, 2023-06-15:
I got the same problem on Mac Intel

glynd, 2023-06-29:
Same on a Raspberry Pi 4 / Ubuntu