project-chip/connectedhomeip #25986: [BUG] [Android] Building Android chip-tools from scripts fails

tetedoie, 2023-04-05:
### Reproduction steps

Building Android chip-tools from scripts start failing on commit 5927a9adf9022cdb2a281aebab97fb06e8c7e21d #25939 with error `clang++: error: unknown argument: '-fmacro-prefix-map=../../='`
We can see the same error in this Github Action https://github.com/project-chip/connectedhomeip/actions/workflows/full-android.yaml

Following instructions from https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md 
     1 - Setup environment variables
     2 - Run bootstrap script
     3 - Run build script (./scripts/build/build_examples.py --target android-arm64-chip-tool build) (same error for any abi tested)  
     
 Building Android chip-tools from script (on full-android.yaml Github Action and locally) is working on commits before 5927a9adf9022cdb2a281aebab97fb06e8c7e21d

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

5927a9adf9022cdb2a281aebab97fb06e8c7e21d

### Platform

android

### Anything else?

Logs from CI: 
```
INFO    [95/1381] c++ obj/src/access/libaccess.AccessControl.cpp.o
INFO    FAILED: obj/src/access/libaccess.AccessControl.cpp.o 
INFO    /opt/android/android-ndk-r21b/toolchains/llvm/prebuilt/linux-x86_64/bin/armv7a-linux-androideabi24-clang++ -MMD -MF obj/src/access/libaccess.AccessControl.cpp.o.d -Wconversion -march=armv7-a -mfpu=vfpv3-d16 -mfloat-abi=softfp -mthumb -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fvisibility=hidden -fno-exceptions -funwind-tables -fPIC -Wall -Werror -Wstack-usage=8192 -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wimplicit-fallthrough -Wheader-hygiene -Wshorten-64-to-32 -Wformat-type-confusion -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -I../../third_party/mbedtls/repo/include -c ../../src/access/AccessControl.cpp -o obj/src/access/libaccess.AccessControl.cpp.o
INFO    clang++: error: unknown argument: '-fmacro-prefix-map=../../='
INFO    [96/1381] c++ obj/src/access/tests/libaccesstest.TestAccessControl.cpp.o
INFO    FAILED: obj/src/access/tests/libaccesstest.TestAccessControl.cpp.o 
INFO    /opt/android/android-ndk-r21b/toolchains/llvm/prebuilt/linux-x86_64/bin/armv7a-linux-androideabi24-clang++ -MMD -MF obj/src/access/tests/libaccesstest.TestAccessControl.cpp.o.d -Wconversion -march=armv7-a -mfpu=vfpv3-d16 -mfloat-abi=softfp -mthumb -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fvisibility=hidden -fno-exceptions -funwind-tables -fPIC -Wall -Werror -Wstack-usage=8192 -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wimplicit-fallthrough -Wheader-hygiene -Wshorten-64-to-32 -Wformat-type-confusion -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -I../../third_party/mbedtls/repo/include -I../../third_party/nlunit-test/repo/src -c ../../src/access/tests/TestAccessControl.cpp -o obj/src/access/tests/libaccesstest.TestAccessControl.cpp.o
INFO    clang++: error: unknown argument: '-fmacro-prefix-map=../../='
INFO    [97/1381] ACTION //src/controller/data_model:data_model_zapgen_zap_pregen(//build/toolchain/android:android_arm)
INFO    ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "/__w/connectedhomeip/connectedhomeip/./scripts/build/build_examples.py", line 230, in <module>
    main(auto_envvar_prefix='CHIP')
  File "/__w/connectedhomeip/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/__w/connectedhomeip/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/__w/connectedhomeip/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 1688, in invoke
    rv.append(sub_ctx.command.invoke(sub_ctx))
  File "/__w/connectedhomeip/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/__w/connectedhomeip/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/__w/connectedhomeip/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/__w/connectedhomeip/connectedhomeip/./scripts/build/build_examples.py", line 220, in cmd_build
    context.obj.Build()
  File "/__w/connectedhomeip/connectedhomeip/scripts/build/build/__init__.py", line 71, in Build
    builder.build()
  File "/__w/connectedhomeip/connectedhomeip/scripts/build/builders/builder.py", line 99, in build
    self._build()
  File "/__w/connectedhomeip/connectedhomeip/scripts/build/builders/android.py", line 428, in _build
    self._Execute(
  File "/__w/connectedhomeip/connectedhomeip/scripts/build/builders/builder.py", line 104, in _Execute
    self._runner.Run(cmdarray, title=title)
  File "/__w/connectedhomeip/connectedhomeip/scripts/build/runner/shell.py", line 73, in Run
    raise Exception('Command %r failed: %d' % (cmd, code))
Exception: Command ['ninja', '-C', '/__w/connectedhomeip/connectedhomeip/out/android-arm-chip-tool'] failed: 1
```
Local logs are really similar and can be provided as well. 
Fixing the Github Action will probably only require to bump the container version like https://github.com/ksperling-apple/connectedhomeip/blob/4d75d75d11b0ec2563031d8e583b8af3ee40d47f/.github/workflows/smoketest-android.yaml#L41 but how do we fix this issue for local machine ?

bzbarsky-apple, 2023-04-05:
@tetedoie What version of the NDK are you using?  See https://github.com/project-chip/connectedhomeip/pull/25952 which was a prerequisite for https://github.com/project-chip/connectedhomeip/pull/25939

tetedoie, 2023-04-05:
I'm using the NDK version (21.4.7075529) specified in https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md. I can test with r23c (which I assume is 23.2.8568313 ?)

tetedoie, 2023-04-05:
Using NDK 23.2.8568313 fix the issue. I'm gonna close this ticket since a new ticket was opened to update the documentation #25987 (I'll add the additional Github Actions update on the other ticket)

closed 2023-04-05 by tetedoie

nmtoan91, 2023-05-10:
@bzbarsky-apple I'm sorry t bother you again. I use NDK 23.2.8568313 but I still got the same problem.
here are my environment variables:
$ANDROID_NDK_HOME -> /home/nuc17/Android/Sdk/ndk/23.2.8568313
$ANDROID_HOME -> /home/nuc17/Android/Sdk
#JAVA_HOME -> /usr/lib/jvm/java-11-openjdk-amd64

bzbarsky-apple, 2023-05-10:
@nmtoan91 Sorry, I don't know enough about the Android side of things to tell you anything useful.

tetedoie, 2023-05-11:
@nmtoan91 What error do you get exactly? The same `clang++: error: unknown argument: '-fmacro-prefix-map=../../='` ? Did you installed NDK `23.2.8568313`? Does it work with `21.4.7075529` before commit https://github.com/project-chip/connectedhomeip/commit/5927a9adf9022cdb2a281aebab97fb06e8c7e21d ?

nmtoan91, 2023-05-11:
@tetedoie,
Sorry, I solved this error by cloning another one and start building again. Maybe the problem is on cached files. 
By the way, I got another problem when building android in [26497](https://github.com/project-chip/connectedhomeip/issues/26497), hope you can take a look.

tetedoie, 2023-05-11:
I'm looking at your other issue