project-chip/connectedhomeip #25987: [Documentation] android_building.md should be updated after SDK and NDK version changes.

andy31415, 2023-04-05:
### Documentation issues

https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md

needs to describe how to build using the latest NDK/SDK versions.

### Platform

android

tetedoie, 2023-04-05:
We also need to update the container version (to 0.7.0) used by this Github actions https://github.com/project-chip/connectedhomeip/blob/e4faf945816ae45bfe4548b0ab4b572d20b20b41/.github/workflows/full-android.yaml#L40 like in this other Github action https://github.com/ksperling-apple/connectedhomeip/blob/4d75d75d11b0ec2563031d8e583b8af3ee40d47f/.github/workflows/smoketest-android.yaml#L41

ksperling-apple, 2023-04-05:
I included the update to the workflow in this PR that depends on it: https://github.com/project-chip/connectedhomeip/pull/25939

bzbarsky-apple, 2023-04-05:
@ksperling-apple The comment above is about the other Android workflow: `[.github/workflows/full-android.yaml](https://github.com/project-chip/connectedhomeip/blob/e4faf945816ae45bfe4548b0ab4b572d20b20b41/.github/workflows/full-android.yaml#L40)`

ksperling-apple, 2023-04-05:
Looks like the SDK version is 26 now instead of 21 since #25204.

I don't have Android Studio installed to check what the exact revision numbers currently are though. There seem to be extra revision numbers there when installing via Android Studio compared to the NDK downloads page.

tetedoie, 2023-04-06:
> I don't have Android Studio installed to check what the exact revision numbers currently are though. There seem to be extra revision numbers there when installing via Android Studio compared to the NDK downloads page.

If the new NDK version is `r23c` (#25952), it correspond to `23.2.8568313` in android studio (https://github.com/android/ndk/wiki/Unsupported-Downloads#r23c). I tested and it's working (when built from scripts)

tetedoie, 2023-04-07:
It seems that the ndk version must be specified in the android config or the geenerated `.cxx/ndk_locator_record` file is still referencing version `21.4.7075529`.

tetedoie, 2023-04-07:
> Looks like the SDK version is 26 now instead of 21 since #25204 so that's wrong now too.

How do we know the SDK version when built from scripts ? (I'm assuming we are referring here to the min SDK requirement to build the `AndroidPlatform.jar`)

When building from source, the `minSdkVersion` of the `chip-library` module is currently 24.

srdantas, 2023-05-08:
same problem... @tetedoie can you share your solution?

karlp, 2023-06-14:
I'm stuck here too, the android studio SDK is now 17.x as well, so the gradle 7.1.1 builds in the v1.1branch builds all fail with `Unsupported class file major version 61`

karlp, 2023-06-14:
FWIW, I finally got it to build after editing `examples/android/CHIPTool/app/build/outputs/apk/debug/app-debug.apk` to use the 7.0.0 gradle plugin (based on information from https://stackoverflow.com/a/70807648) and also updating `examples/android/CHIPTool/gradle/wrapper/gradle-wrapper.properties` and `third_party/android_deps/gradle/wrapper/gradle-wrapper.properties` to use gradle 7.3.3, which supports java 17.

After all that, the app is still..... well, it's an app, but that's unrelated to getting it to build :)

joonhaengHeo, 2024-01-29:
@yunhanw-google What is the final goal of this open issue?

yunhanw-google, 2025-02-03:
we have upgraded the android build with java 17 and gradle 7.3.3, and kotlin 2.0.
https://github.com/project-chip/connectedhomeip/pull/33837, this issue should be resolved, feel free to reopen a new case if this is still problematic. thanks

closed 2025-02-03 by yunhanw-google