project-chip/connectedhomeip #26003: [BUG] Undefined identifier (pw_env_setup_VIRTUAL_ENV}) during configuration (gn gen)

javierbarellano4v, 2023-04-06:
### Reproduction steps

The usual  steps
1.  Cloned matter
2.  Build zap from source
3.  Install prerequisites for Raspberry PI.
4.  Build matter
5.  [app build] gn gen out/debug

### Bug prevalence

each time during build

### GitHub hash of the SDK that was being used

?

### Platform

raspi

### Anything else?

New build of Matter.

Attempting to build lighting-app on Raspberry Pi

I'm able to run the bootstrap.sh activate.sh scripts beforehand without a problem so I'm a bit perplexed.

(also, this was after building zap from source given Raspberry Pi);

javier@javier-ubuntu-raspberry-pi:~/matter1/connectedhomeip/examples/lighting-app/linux$ gn gen out/debug
+ gn gen out/debug
ERROR at //build_overrides/pigweed_environment.gni:30:41: Undefined identifier
    get_path_info("${_bootstrap_root}/${pw_env_setup_VIRTUAL_ENV}", "abspath")
                                        ^-----------------------
See //third_party/connectedhomeip/third_party/pigweed/repo/pw_toolchain/clang_tools.gni:15:1: whence it was imported.
import("//build_overrides/pigweed_environment.gni")
^-------------------------------------------------
See //third_party/connectedhomeip/third_party/pigweed/repo/pw_toolchain/host_clang/toolchains.gni:17:1: whence it was imported.
import("$dir_pw_toolchain/clang_tools.gni")
^-----------------------------------------
See //third_party/connectedhomeip/third_party/pigweed/repo/pw_unit_test/test.gni:20:1: whence it was imported.
import("$dir_pw_toolchain/host_clang/toolchains.gni")
^---------------------------------------------------
See //third_party/connectedhomeip/third_party/pigweed/repo/pw_build/BUILD.gn:22:1: whence it was imported.
import("$dir_pw_unit_test/test.gni")
^----------------------------------
See //third_party/connectedhomeip/build/config/compiler/BUILD.gn:327:15: which caused the file to be included.
  configs = [ "$dir_pw_build:colorize_output" ]

javierbarellano4v, 2023-04-11:
close. fixed

closed 2023-04-11 by javierbarellano4v

LinJiaDeng, 2023-04-11:
Hi, I would like to know how did you fix it, I face the same problem trying to build an example for the esp32c3.
Thanks in advance

javierbarellano4v, 2023-04-12:
To be honest, it was strange.  

I had to move my raspberry pi to a new location,.

Once I booted it up, all I ran the activate.sh script at top level (source scripts/activate.sh).

Launched into  'gn gen out/debug' and the error did not appear.

I followed up building the lighting-app (examples) and that error did not appear either.

LinJiaDeng, 2023-04-14:
@javierbarellano4v Thank you for your answer!

marcocococo, 2023-05-07:
I had the bug. I think they forgot to change some specifity to the examples, becaues the last line point towards another .gni where the pw_env_setup_VIRTUAL_ENV does not exist. So I worked around that bug doing this :

1. Open in edit mode the file .......\esp-matter\connectedhomeip\connectedhomeip\examples\build_overrides
2. Delete the last line about pw_env_setup_VIRTUAL_ENV, and replace it by this one :
 
pw_env_setup_VIRTUAL_ENV =
    "${_bootstrap_root}/.environment/pigweed-venv"

This should now point to the good path.

javierbarellano4v, 2023-05-08:
@marcocococo Thanks for this information. I greatly appreciate for your clarification.

Goks1730, 2023-10-26:
i am getting the same error 

ee //third_party/connectedhomeip/third_party/pigweed/repo/pw_toolchain/host_clang/toolchains.gni:17:1: whence it was imported.
import("$dir_pw_toolchain/clang_tools.gni")
^-----------------------------------------
See //third_party/connectedhomeip/third_party/pigweed/repo/pw_unit_test/test.gni:20:1: whence it was imported.
import("$dir_pw_toolchain/host_clang/toolchains.gni")
^---------------------------------------------------
See //third_party/connectedhomeip/third_party/pigweed/repo/pw_build/BUILD.gn:22:1: whence it was imported.
import("$dir_pw_unit_test/test.gni")
^----------------------------------

but i am tring on WSL(Ubuntu) and using ModusToolsBox for building and flashing 

if i want to try on pi how can i do that that cloning the Matter SDK requires more space?