project-chip/connectedhomeip #26019: [Platform] M5Stack: No response to Sigma3, Failed to post event to CHIP Platform event queue

kpark-apple, 2023-04-07:
### Reproduction steps

1. Commission M5.
2. Add another fabric over CASE of the first commissioned fabric
3. Open a new CASE session over the new fabric
4. Complete commissioning of the new fabric

### Platform

esp32

### Type

Manually tested with SDK

### (Optional) If manually tested please explain why this is only manually tested

The steps above were exercised.

### Anything else?

On M5 log, there were lots of `Failed to post event to CHIP Platform event queue` messages and one after receiving the Sigma3 message as well. So suspecting resource issue.

```
I (59199) chip[EM]: <<< [E:13002r S:0 M:38815149 (Ack:13084043)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (59219) chip[IN]: (U) Sending msg 38815149 to IP address 'UDP:[FE80::18DD:660:1490:E11A%st1]:51277'
I (59219) chip[SC]: Received Sigma3 msg
E (59229) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
E (60179) chip[DL]: Failed to post event to CHIP Platform event queue
E (60179) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (60199) chip[DL]: Failed to post event to CHIP Platform event queue
E (60199) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
...
I (75569) chip[FS]: Fail-safe timer expired
E (75569) chip[SVR]: Failsafe timer expired
E (75569) chip[SVR]: Commissioning failed (attempt 1): 32
```

[M5_pairing_failed_accessory_log_21A220a.txt](https://github.com/project-chip/connectedhomeip/files/11179856/M5_pairing_failed_accessory_log_21A220a.txt)

bzbarsky-apple, 2023-04-07:
@kpark-apple Which SHA is the M5 using?  Is it something after 77e39a62d88b539d6684ec504a8f6c52eb09604d?

This could also be similar to https://github.com/project-chip/connectedhomeip/issues/25833 (note the lower-memory non-default configuration file that was added in https://github.com/project-chip/connectedhomeip/pull/25935), and is likely the same as https://github.com/project-chip/connectedhomeip/issues/24418 and https://github.com/project-chip/connectedhomeip/issues/23258

shah-kartik, 2023-04-07:
The SHA used here was
0f1fc1b230eca4b4ce2f45c4a8cf1ce238284e96

bzbarsky-apple, 2023-04-07:
OK, so that's after https://github.com/project-chip/connectedhomeip/commit/77e39a62d88b539d6684ec504a8f6c52eb09604d

dhrishi, 2024-10-25:
The queue size has been increased to from 25 to 40: https://github.com/project-chip/connectedhomeip/blob/master/config/esp32/components/chip/Kconfig#L385

closed 2024-10-25 by dhrishi