project-chip/connectedhomeip #26027: [Documentation] Inconsistencies in Device Library Specification

Apollon77, 2023-04-08:
### Documentation issues

* chapter 2.2.1 (Power Source): "Utility" is no defined devcie class type, should it be Node or Simple or what? Or is utility missing in list of class types; same for  chapter 2.3.2 (OTA Requestor),  chapter 2.4.2 (OTA provider) and also 2.5.2 (Aggregator) has "Dynamic Utility"
* chapter 5.3.4 (Pump)/9.1.4 (Heater/Cooling Unit)/10.4.4(Speaker)/10.6.4 (Casting Video client): "Level" -> "Level Control" (Cluster name formally wrong)
* Is there a reason why the Groups Cluster client is for some sensoring devicetypes optional (7.1-7.3), and sometimes only optional if Zigbee (7.4+)? Would it not make more sense to have that unified because makes sense in any case potentially. Same basically on Door Lock. That feels a bit random designed ... :-)

In general the documentation is a bit strange when it comes to Device types ... In examples it talks from Device type "Fridge/Freezer" and also "TV" (10.1.3 on pg 71 of Device Library Specs) in tables/charts as if these would be officially defined devcie types. But I could not find them anywhere.

Also for composed Devices there is only one example at all with Bridged devices as device type but nothing else. The TV example in 10.1.3 on pg 71 of Device Library Specs is also kind of an composed device, but then forthe root it states "TV" as decvice type which is not existing -  so what should be set there?

Or do I miss any specs?

Thank you for bringing up some light into that topic.

### Platform

other

bzbarsky-apple, 2023-04-10:
@Apollon77 Ideally spec issues would get filed in the spec repo, because they are not actionable here...  I understand that you might not have access to that repo, though.

> "Utility" is no defined devcie class type

I'm following up to figure out which part of the spec is wrong there.

> "Level" -> "Level Control"

Also the Pump Controller device type...  Filed https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/6390

> why the Groups Cluster client is for some sensoring devicetypes optional (7.1-7.3), and sometimes only optional if Zigbee (7.4+)?

Probably an oversight, but hard to tell.  It should probably be `[Zigbee]` in all of these....

> Same basically on Door Lock. 

I'm not sure what the Door Lock issue you're referring to is.

> In general the documentation is a bit strange when it comes to Device types 

It's not documentation.  It's a specification.

Examples are just that: examples.  They can talk about device types that could exist but don't yet and how they could be represented using the primitives the specification defines, etc.

> for composed Devices there is only one example at all with Bridged devices as device type but nothing else.

For now, yes.  More are in the process of being specified.

Apollon77, 2023-04-10:
And yes I do not have access to the specs repo because I'm "just" a developer that use them to implement the protocol I'm one of the core team devs of matter.js (https://github.com/project-chip/matter.js). No idea if there are ways by signing NDAs or such to get further access to such parts for open source contributors (most likely not). So if there is a better place for the issue please tell.

Thank you for your information. I understand that the examples are just examples, but it's still a bit confusing. In my eyes an example for composed devices which based on the current real device types would be beneficial additional to the others because right now I need to take some assumptions - or look into Chip-tool code if there is an example (knowing that also Chip-tool only is a reference implementation which could also be out of specs) ;-)

bzbarsky-apple, 2023-04-10:
> No idea if there are ways by signing NDAs or such to get further access to such parts for open source contributors 

Please check with @clapre ?

> In my eyes an example for composed devices which based on the current real device types would be beneficial

The problem is lack of existing composed devices, as you noted....

Apollon77, 2023-04-12:
> Please check with @clapre ?

I wrote him an email

Apollon77, 2023-10-15:
Let's close here ... I have no insights into internal chip workgroup processes or documentation enhancement plans ... sooo ...

closed 2023-10-15 by Apollon77